raspberrypi/picamera2 #239: [HOW-TO] Timestamps on still images

magsol, 2022-07-28:
I'm trying to timestamp still images (will make more sense when I take a series of still images over a long period).

The code I've developed is based on the [example provided for timestamping videos](https://github.com/raspberrypi/picamera2/blob/main/examples/timestamped_video.py):

```python
from picamera2 import Picamera2, MappedArray
from picamera2.outputs import FileOutput
import cv2
import time

picam2 = Picamera2()
picam2.configure(picam2.create_still_configuration())

def apply_timestamp(request):
    ts = time.strftime("%Y-%m-%d %X")
    with MappedArray(request, "main") as m:
        cv2.putText(m.array, ts, (0, 30), 
            cv2.FONT_HERSHEY_SIMPLEX, 1, (0, 255, 0), 2)

picam2.pre_callback = apply_timestamp

picam2.start()
time.sleep(2)
metadata = picam2.capture_file("test.jpg")
print(metadata)
picam2.close()
```

However, I get the following error:

```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/opt/python/envs/picam/lib/python3.9/threading.py", line 980, in _bootstrap_inner
    self.run()
  File "/opt/python/envs/picam/lib/python3.9/threading.py", line 917, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/python/envs/picam/lib/python3.9/site-packages/picamera2/previews/null_preview.py", line 25, in thread_func
    callback(picam2)
  File "/opt/python/envs/picam/lib/python3.9/site-packages/picamera2/previews/null_preview.py", line 72, in handle_request
    completed_request = picam2.process_requests()
  File "/opt/python/envs/picam/lib/python3.9/site-packages/picamera2/picamera2.py", line 877, in process_requests
    self.pre_callback(display_request)
  File "/home/squinn/picam.py", line 11, in apply_timestamp
    with MappedArray(request, "main") as m:
  File "/opt/python/envs/picam/lib/python3.9/site-packages/picamera2/request.py", line 62, in __enter__
    raise RuntimeError(f'MappedArray: Cannot reshape buffer {w}x{h}x3 (stride {stride})')
RuntimeError: MappedArray: Cannot reshape buffer 3280x2464x3 (stride 9888)
```

The change in image resolution from video to still appears to be problematic, but I'm not sure how to solve this. I've tried changing a few of the `putText` parameter values (thickness, scale, and org) to no avail. But I also don't feel like I have an intuition for what's really happening here.

Assistance would be welcome!

davidplowman, 2022-07-28:
Ah, that's a nasty little corner. It doesn't currently let you have a "shared memory" view of an image when the image stride required by the hardware does not work out to be an exact number of pixels.

So I have fixed this for the next release (https://github.com/raspberrypi/picamera2/pull/240/commits/4065c085c57fc42899cd28566d60c301729bea90), but if you wanted a workaround for now, you could start your script like this:
```
picam2 = Picamera2()
config = picam2.create_still_configuration()
picam2.align_configuration(config)
picam2.configure(config)
```
This will change your image size slightly, but at least keeps you going until the next release is available.

magsol, 2022-07-28:
Thank you @davidplowman! I'll implement this fix and report back.

As an aside, I'm curious about that `stride` parameter. I'm coming from a machine learning / AI background so I understand what it's doing in a general context, but I'm unsure where the parameter is set or what role it plays in writing text to the image. I glanced through the Picamera code (admittedly not as thoroughly as I'd like so I probably missed it) and it wasn't clear to me where the value was set or what role it played, aside from throwing this error.

Is there anywhere I could look to learn more about it?

davidplowman, 2022-07-29:
Hi, I'm not quite sure where you find out about "stride" in this sense. We always use it to describe the distance, in bytes, from a pixel in one row of an image to the pixel exactly one row below. The reason why "stride" becomes a problem is that in the pure software world, folks often don't need to think about it. The distance from a pixel to the one below is exactly the "pixel width", and if you have a 3-channel 8-bit image that's 3 * "pixel width" bytes, right?

The difficulty is that lots of hardware that produces images has memory alignment restrictions, the reason being that it's much more efficient to do aligned memory writes. For example it may need to start writing each row of an image to an address that is 64-byte aligned. So if your hardware generates a 224 pixel wide RGB image, you would have to allow for an extra 32 bytes of padding on the end of each row. So the "stride" of the image is not 3 * 224 = 672, but 704 which is the next multiple of 64. Therefore you need to use algorithms that understand "stride" (very many do), but when they don't you might have to make a complete copy of the image carefully skipping all the padding bytes!

In the context of Picamera2, "stride" is determined by layers of software much further down the stack. In actual fact it will be the V4L2 (Video for Linux) device drivers down in the Linux kernel that will be making these decisions and passing the numbers back through libcamera to Picamera2. It's only right down there that the software understands what hardware it is using and what constraints it may have.

Hope that helps a bit..

magsol, 2022-07-29:
This does, thank you. My concept of stride is at a much higher level--image convolutions, which happens well after image acquisition--and the fact that this particular stride implementation is happening at the V4L2 layer pretty much tells me it's out of my hands as a Picamera2 user. I was just confused because the stride mentioned in the error message--9888--really isn't a factor of any of the image dimensions or combinations thereof, so I was very confused then as to what stride meant in this case. But if there's extra padding somewhere in the lower software levels, that would make sense.

closed 2022-08-24 by magsol