raspberrypi/picamera2 #241: [HOW-TO] convert recording to YU420

BrodyStone21, 2022-07-28:
I've been working for a few days now with the [mjpeg_server.py](https://github.com/raspberrypi/picamera2/blob/main/examples/mjpeg_server.py) example, but I cannot, for the life of me, figure out how to convert the stream from yu444p to yu420. 

I'm trying to decode the stream on my Nvidia GPU using mjpeg_cuvid, but it doesn't support yu444p. 

I've tried doing something like `picam2.create_video_configuration(main={"size": (1280, 720), "format": "YUV420"})`

However, I get an error saying `Buffer has wrong number of dimensions (expected 3, got 2)`

I've also tried converting with open-cv, but I can't figure out how to convert the video, then stream it. 

Anyone have any solutions?

davidplowman, 2022-07-29:
Ah, hello again, I see that I've already answered your questions elsewhere. But I think the information here is a bit clearer, so let me try to give a better answer!

The first point is that Python libraries, so far as I have found, don't generally support YUV formats very well, but only RGB ones. This is why you get the error - basically you mustn't change the format to YUV420. Unfortunately the JPEG encoders will then spend time converting the RGB they are given to YUV, which they then encode. This all seems pretty sub-optimal, but that's how things seem to be.

Secondly, the choice of YUV444 is presumably being made by the JPEG encoder that does the conversion from RGB to YUV. In this case we're using the [Python "simplejpeg" module](https://pypi.org/project/simplejpeg/). I notice it does have a parameter  that allows you to select the colour subsampling, and so I've created a PR which allows you to pass your choice to the `JpegEncoder` class. In fact, I've also made YUV420 the default.

The PR in question is [here](https://github.com/raspberrypi/picamera2/pull/242) - would you be in a position to try it and see if it does what you want? In which case I'd then be happy to merge it. Thanks!

BrodyStone21, 2022-07-29:
> Ah, hello again, I see that I've already answered your questions elsewhere. But I think the information here is a bit clearer, so let me try to give a better answer!
>
> The first point is that Python [...]

> 

I left this comment on the PR request, but I'll leave it here for other people searching this. 

This works perfectly! Thank you so much. I literally spent close to four days attempting to get this working. I ended up finally giving up, but you've just saved my server's CPU from so much unnecessary load. 

Old output of ffprobe on the http stream: 
```
Input #0, mpjpeg, from 'http://192.168.1.100:8000/stream.mjpg':
  Duration: N/A, bitrate: N/A
  Stream #0:0: Video: mjpeg (Baseline), yuvj444p(pc, bt470bg/unknown/unknown), 1640x1232 [SAR 1:1 DAR 205:154], 25 tbr, 25 tbn
```
New output of ffprobe on the http stream:
```
Input #0, mpjpeg, from 'http://192.168.1.100:8000/stream.mjpg':
  Duration: N/A, bitrate: N/A
  Stream #0:0: Video: mjpeg (Baseline), yuvj420p(pc, bt470bg/unknown/unknown), 1640x1232 [SAR 1:1 DAR 205:154], 25 tbr, 25 tbn
```

closed 2022-07-29 by BrodyStone21