raspberrypi/picamera2 #244: [BUG] Can't add overlay to DRM preview

callidus, 2022-07-31:
**Describe the bug**
If you try to set an overlay when using DRM (none-X11) preview an error is generated

**To Reproduce**
Run the `overlay_drm.py` example from the examples folder

**Expected behaviour**
Overly should work

**Console Output, Screenshots**
```
python ./overlay_drm.py 
[19:34:00.008408942] [26564]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3700-f30ad033
[19:34:00.046462844] [26565]  INFO RPI raspberrypi.cpp:1368 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
[19:34:00.050357758] [26564]  INFO Camera camera.cpp:1029 configuring streams: (0) 640x480-XBGR8888
[19:34:00.050808017] [26565]  INFO RPI raspberrypi.cpp:759 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
Traceback (most recent call last):
  File "/home/pi/cam_stuff/./overlay_drm.py", line 20, in <module>
    picam2.set_overlay(overlay)
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/picamera2.py", line 1278, in set_overlay
    self._preview.set_overlay(overlay)
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/previews/drm_preview.py", line 76, in set_overlay
    with mmap.mmap(new_fb.planes[0].fd, w * h * 4, mmap.MAP_SHARED, mmap.PROT_WRITE) as mm:
AttributeError: 'pykms.pykms.DumbFramebuffer' object has no attribute 'planes'
```

**Hardware :**
Raspberry Pi 4 with HQ camera module attached via CSI

**Additional context**
Running on the latest (at time of opening this ticket) 32bit Raspberry PI OS Lite image (no desktop).

```
pi@camera1:~/cam_stuff $ pip freeze
certifi==2020.6.20
chardet==4.0.0
colorzero==1.1
distro==1.5.0
gpiozero==1.6.2
idna==2.10
numpy==1.23.1
opencv-python==4.6.0.66
picamera2 @ git+https://github.com/raspberrypi/picamera2.git@0076c7b0b18a5ff836c40c833a3edb8fa1e9d599
pidng==4.0.9
piexif==1.1.3
Pillow==9.2.0
python-apt==2.2.1
python-prctl==1.7
requests==2.25.1
RPi.GPIO==0.7.0
simplejpeg==1.6.4
six==1.16.0
spidev==3.5
ssh-import-id==5.10
urllib3==1.26.5
v4l2-python3==0.3.1
```

let me know if I can provide any extra info, thank you.

callidus, 2022-07-31:
The above PR contains a fix that worked for me, first time contributor so apologies if the formatting or anything else is amiss. Please let me know and I will update as needed

davidplowman, 2022-08-01:
Thanks for reporting this and for the PR - have replied on that thread!

davidplowman, 2022-08-03:
I'm going to close this one as I believe the problem is now fixed on the "next" branch, and we'll be releasing that fairly soon. If you are able to check it that would be great, and please file another issue if the problem reappears. Thanks!

closed 2022-08-03 by davidplowman