raspberrypi/picamera2 #251: [BUG] Start Preview (QTGL) loads 'xcb' module from cv2 path vs QT path

ankitb82, 2022-08-04:
**Describe the bug**
On start_preview(QTGL), QT throws an error stating -
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/usr/local/lib/python3.9/dist-packages/cv2/qt/plugins" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb, eglfs, linuxfb, minimal, minimalegl, offscreen, vnc.

**To Reproduce**
Use code that imports/utilizes opencv2 and picamera2. Opencv-python (with GUI bindings) should be installed.

**Expected behaviour**
Preview windows launches without error

**Console Output, Screenshots**
export QT_DEBUG_PLUGINS=1
...

Got keys from plugin meta data ("vnc")
QFactoryLoader::QFactoryLoader() looking at "/usr/lib/aarch64-linux-gnu/qt5/plugins/platforms/libqxcb.so"
Found metadata in lib /usr/lib/aarch64-linux-gnu/qt5/plugins/platforms/libqxcb.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "xcb"
        ]
    },
    "archreq": 0,
    "className": "QXcbIntegrationPlugin",
    "debug": false,
    "version": 331520
}

Got keys from plugin meta data ("xcb")
QFactoryLoader::QFactoryLoader() checking directory path "/usr/bin/platforms" ...
loaded library "/usr/local/lib/python3.9/dist-packages/cv2/qt/plugins/platforms/libqxcb.so"
QObject::moveToThread: Current thread (0x7f40020840) is not the object's thread (0x7f40223430).
Cannot move to target thread (0x7f40020840)

qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/usr/local/lib/python3.9/dist-packages/cv2/qt/plugins" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb, eglfs, linuxfb, minimal, minimalegl, offscreen, vnc.

**Hardware :**
 Raspberry 4 (4GB) with Arducam 16MP

davidplowman, 2022-08-04:
Hi, thanks for reporting this. I'm not managing to see this problem, could you attach a small self-contained code snippet that will reproduce it? Perhaps also explain whether you're logged directly into the Pi or are (for example) connected via ssh.  Thanks!

davidplowman, 2022-08-04:
Just to add that a quick google search (for "python qt plugins opencv") reveals that OpenCV installs the QT plugins for its own use, and it seems like the ones you have are not compatible with the PyQt5 ones. Does that sound plausible in your case? If the [official installation instructions](https://github.com/raspberrypi/picamera2#installation) on a clean image lead to a failure please let us know, otherwise those various pages thrown up by the search suggest various remedies...

ankitb82, 2022-08-04:
Try this script (Be sure to verify that opencv-python is installed)

---
import argparse
import sys
import time

import cv2

from picamera2 import Picamera2, Preview

picam2 = Picamera2()
picam2.start_preview(Preview.QTGL)
picam2.start()
image = picam2.capture_array()

image = cv2.resize(image, (320, 240))
image = cv2.cvtColor(image, cv2.COLOR_BGR2RGB)

cv2.imshow('test', image)

---

I resolved it initially by going to headless opencv. Which is ofcourse not ideal. 

However I found another workaround and that is to include this -

import os, sys
ci_build_and_not_headless = False
try:
    from cv2.version import ci_build, headless
    ci_and_not_headless = ci_build and not headless
except:
    pass
if sys.platform.startswith("linux") and ci_and_not_headless:
    os.environ.pop("QT_QPA_PLATFORM_PLUGIN_PATH")
if sys.platform.startswith("linux") and ci_and_not_headless:
    os.environ.pop("QT_QPA_FONTDIR")

This however breaks opencv2.imshow or any opencv QT based function from working. 

Ask - would it be possible to limit scope for all picamera2 based QT windows to only refer to plugin loaded from pyqt folder vs any other location?

davidplowman, 2022-08-10:
I honestly have no idea whether it's possible to "limit the scope" of Picamera2's Qt windows in any way - is this something you have seen before and could point me at?

I'm also not able to make anything go wrong on my system which makes it difficult to try anything! Can you explain what you did to get your system into this state where it doesn't work properly?

zoldaten, 2022-08-14:
hi! 
i have the same bug when try to start on Raspbian 64 app_capture2.py:
```
[0:13:13.452996081] [3436]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3700-f30ad033
[0:13:13.495844114] [3445]  INFO RPI raspberrypi.cpp:1368 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media0 and ISP device /dev/media2
[0:13:13.500441611] [3436]  INFO Camera camera.cpp:1029 configuring streams: (0) 800x600-XBGR8888
[0:13:13.500943755] [3445]  INFO RPI raspberrypi.cpp:759 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
QObject::moveToThread: Current thread (0x12473eb0) is not the object's thread (0x132981f0).
Cannot move to target thread (0x12473eb0)

qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/home/pi/.local/lib/python3.9/site-packages/cv2/qt/plugins" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb, eglfs, linuxfb, minimal, minimalegl, offscreen, vnc.
```
I think may be it is qt5 issue because when i use this:
```export QT_QPA_PLATFORM=vnc```
The output changed to:
```
[0:19:06.261125967] [4199]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3700-f30ad033
[0:19:06.300875238] [4206]  INFO RPI raspberrypi.cpp:1368 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media0 and ISP device /dev/media2
[0:19:06.304477303] [4199]  INFO Camera camera.cpp:1029 configuring streams: (0) 800x600-XBGR8888
[0:19:06.304855186] [4206]  INFO RPI raspberrypi.cpp:759 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
Traceback (most recent call last):
  File "/home/pi/picamera2/examples/app_capture2.py", line 36, in <module>
    qpicamera2 = QGlPicamera2(picam2, width=800, height=600, keep_ar=False)
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/previews/q_gl_picamera2.py", line 102, in __init__
    self.init_gl()
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/previews/q_gl_picamera2.py", line 138, in init_gl
    self.create_surface()
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/previews/q_gl_picamera2.py", line 132, in create_surface
    surface = eglCreateWindowSurface(self.egl.display, self.egl.config,
  File "/home/pi/.local/lib/python3.9/site-packages/OpenGL/platform/baseplatform.py", line 415, in __call__
    return self( *args, **named )
  File "/home/pi/.local/lib/python3.9/site-packages/OpenGL/error.py", line 230, in glCheckError
    raise self._errorClass(
OpenGL.raw.EGL._errors.EGLError: EGLError(
	err = EGL_BAD_ALLOC,
	baseOperation = eglCreateWindowSurface,
	cArguments = (
		<OpenGL._opaque.EGLDisplay_pointer object at 0x7f8a1e6ec0>,
		<OpenGL._opaque.EGLConfig_pointer object at 0x7f8a8596c0>,
		c_void_p(1),
		None,
	),
	result = <OpenGL._opaque.EGLSurface_pointer object at 0x7f8254cbc0>
)

```

davidplowman, 2022-08-15:
> I think it is qt5 issue. To quick fix just add:
> 
> ```
> import os
> os.environ.pop("QT_QPA_PLATFORM_PLUGIN_PATH")
> ```
> 
> or
> 
> ```
> import os
> os.environ.update({"QT_QPA_PLATFORM_PLUGIN_PATH": "/usr/lib/aarch64-linux-gnu/qt5/plugins/xcbglintegrations/libqxcb-glx-integration.so"})
> ```
> 
> see here: https://stackoverflow.com/questions/63829991/qt-qpa-plugin-could-not-load-the-qt-platform-plugin-xcb-in-even-though-it

Thanks for the info.

@ankitb82 Can you comment on whether that works for you? If this seems to be a generally good workaround then it might be worth adding it to the documentation. Thanks!

ankitb82, 2022-08-19:
This works the same way the workaround I posted above works - it allows picamera2 UX to function but opencv UX functions don't (imshow, etc).

zoldaten, 2022-08-19:
> This works the same way the workaround I posted above works - it allows picamera2 UX to function but opencv UX functions don't (imshow, etc).

d u see stackoverflow link i posted ? thats your question. rebuild opencv if you want cv funcs.

ankitb82, 2022-08-21:
I did look at the stackoverflow link. But think about it, expecting folks to recompile opencv just to work with picamera2 seems like a stretch right? 

Opencv is such a common library and those window functions are super commonly used (During development and otherwise).

davidplowman, 2022-08-22:
Rewinding this discussion a bit, can you say exactly what image you're running and how you installed PyQt5 and OpenCV? So far as I know, if you start with the standard Raspberry Pi OS image and install according to the instructions (use "sudo apt install -y python3-pyqt5 python3-opencv") then everything just works. Was there a particular reason that required you to do something different? Thanks!

davidplowman, 2022-09-06:
Hi, just wondering if there's anything else to add on this issue?

My feeling is that it's not really a Picamera2 issue, it's down to Qt, OpenCV and Python how these things co-exist. In fact the latest Picamera2 on PyPI by default installs neither Qt nor OpenCV, so I'm thinking it's not something for Picamera2 to fix.

As far as I know, everything works fine for folks using the standard installation instructions. But if there are particular circumstances where Qt and OpenCV start to fight then I'd be very happy to add an "Application Note" to our documentation that explains how to deal with this. Having not encountered this myself I don't think I'm really qualified - would anyone else on this thread be able to supply a paragraph or two that I could use for this purpose?

Thanks!

davidplowman, 2022-09-20:
I think I'm going to close this issue at this point, but if anyone would like to return to it - especially if we feel we know what guidance to give users - then please do open another issue. Thanks!

closed 2022-09-20 by davidplowman

trezero, 2022-09-29:
I've been trying to find anyway to get opencv to work with Picamera2 for the past 2 months with no solution. I did not do anything outside of the normal process for install. It simply never works. I've installed over 20 times from scratch and this issue persists. @davidplowman do you have some specific set of instructions of exact commands and order of running them which gets this working? I have Picamera2 working fine, everything works until I try to use any cv2 command which involves preview at which point I constantly get the common "Could not find the Qt platform plugin "xcb" error over and over again.  I would incredibly grateful for any way past this.

davidplowman, 2022-09-29:
@trezero Hi, I start from a fresh Raspberry Pi OS image. I always download and install the latest using the Raspberry Pi Imager. Once that's booted and applied all its updates, I run
```
sudo apt install -y python3-opencv
sudo apt install -y opencv-data
```
as described in section 2.5.1 of the [manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf). After that everything seems to work for me, please let us know if it doesn't for you. Thanks!

nigel03256, 2022-10-03:
I still have the same issue. I was attempting to port a software for DIC called RealPi2dDIC to PiCamera2 and got the said error "Could not find the Qt platform plugin "xcb". The link as follows for the original code for your suggestion on where it might have gone wrong:
https://github.com/ElsevierSoftwareX/SOFTX-D-20-00054

tvoverbeek, 2022-10-03:
One thing to watch out for is mixing software installed with pip and installed with apt.
Python modules are placed in different locations and depending on the python path the wrong version gets picked up.
xcb (X C bindings) is an x-windows library for easy access (simpler than direct Xlib calls).
Qt might also have been installed in more than one place.

quantumjohnny, 2023-12-28:
I had the very same problem (and I could not remember anymore, that I had installed pip install opencv-python
), but then I uninstalled it and with sudo apt install -y python3-opencv everything worked fine! Thank you!