raspberrypi/picamera2 #252: [OTHER] Comments on app_full.py

sandyol55, 2022-08-04:
I've been eagerly following the updates in the next branch and have been trialling the (comprehensive!) app_full.py example.

Not a bug report as the observations are mainly related to use with a 'non-standard' imx462 sensor running with the imx290 overlay and .json file, but in case its useful.......

Selecting the  AEC/AWB tab and picking an AEC Exposure Mode options of either 'Long' or 'Short' results `an instance of 'std::runtime error'` termination. Presumably because the json tuning file only has Normal and Sport exposure modes defined?
(Exercise to self to tinker with tuning file and create some appropriate Long and Short definitions!)

Still in the AEC/AWB section,  dragging the Exposure Value slider to its minimum setting of -8.00 results in a 22microsecond minimum exposure time being selected, but this gets 'locked' and doesn't increase again if the slider value is increased. The value can be increased by deselecting  AEC and manually increasing the Exposure Time.
(The same test with a V2 camera reduces the exposure time to 95 ( ? from memory) which will then increase again if the slider is increased.

And finally, and applicable more widely, if the AWB mode is deselected and the Red/Blue Gain ComboBoxes are incremented the large 1.00 changes seem a bit coarse. 

Looking forward to the next release.

davidplowman, 2022-08-10:
Thanks for the various comments. We can have a look at some of those things though, as you say, some of it may be down to the tuning file or the sensor (probably driver issues).

@will-v-pi Do you want to have a quick look at the AWB combo box issue? Thanks!

will-v-pi, 2022-08-15:
@davidplowman I've fixed the colour gain boxes so that their single step is now 0.1 in #263, which should allow for finer control.

sandyol55, 2022-08-15:
Added the above updates to app_full.py and also modified my .json tuning file to have short and long exposures and had another run through some of the many example features of the app. 

One thing that cropped up is a quirk with the two 'zero-centred'   slider controls, (Brightness in the Image Tuning section and Exposure Value in the AEC/AWB section).  Using the sliders all values can be reached, but starting from 'centre zero' and using the up/down arrows when either + or- 0.6 is reached the value stops incrementing/decrementing and toggles between +/- 0.6 and 0.5.  This happens not just when the IMX462 sensor is used but also with a V2 camera.

(Apologies for nit-picking on such a 'tour de force' app!!!)

will-v-pi, 2022-08-15:
Thanks, I hadn't noticed that - it was just a bit of racing between the slider and the spin box when the values changed. Should be fixed with #266

closed 2022-08-15 by sandyol55