raspberrypi/picamera2 #255: [HOW-TO]

tennisparty, 2022-08-10:
Can someone give me some advice on how to use two streams from a camera simultaneously.

I've seen section '9.3. Multiple Outputs' in the pi documentation but am struggling to get this to work for my use case.

I am trying to combine the mjpegserver.py example (so I can focus the camera in the field) whilst simultaneously making recordings of 5 minutes.

I've got this code so far but the recording seems to pause the web stream and only takes an initial picture. I've read that there is a low res stream and main stream in the software so could use the lowres stream for the web stream and the main stream for the recordings but can't find an example in the docs.

Any help much appreciated.

Code so far
`#!/usr/bin/python3

# Mostly copied from https://picamera.readthedocs.io/en/release-1.13/recipes2.html
# Run this script, then point a web browser at http:<this-ip-address>:8000
# Note: needs simplejpeg to be installed (pip3 install simplejpeg).

import io
import logging
import socketserver
from http import server
from threading import Condition, Thread

from picamera2 import Picamera2
from picamera2.encoders import JpegEncoder
from picamera2.outputs import FfmpegOutput
from picamera2.outputs import FileOutput
import time

PAGE = """\
<html>
<head>
<title>picamera2 MJPEG streaming demo</title>
</head>
<body>
<h1>Picamera2 MJPEG Streaming Demo</h1>
[image]
</body>
</html>
"""

class StreamingOutput(io.BufferedIOBase):
    def __init__(self):
        self.frame = None
        self.condition = Condition()

    def write(self, buf):
        with self.condition:
            self.frame = buf
            self.condition.notify_all()

class StreamingHandler(server.BaseHTTPRequestHandler):
    def do_GET(self):
        if self.path == '/':
            self.send_response(301)
            self.send_header('Location', '/index.html')
            self.end_headers()
        elif self.path == '/index.html':
            content = PAGE.encode('utf-8')
            self.send_response(200)
            self.send_header('Content-Type', 'text/html')
            self.send_header('Content-Length', len(content))
            self.end_headers()
            self.wfile.write(content)
        elif self.path == '/stream.mjpg':
            self.send_response(200)
            self.send_header('Age', 0)
            self.send_header('Cache-Control', 'no-cache, private')
            self.send_header('Pragma', 'no-cache')
            self.send_header('Content-Type', 'multipart/x-mixed-replace; boundary=FRAME')
            self.end_headers()
            try:
                while True:
                    with output.condition:
                        output.condition.wait()
                        frame = output.frame
                    self.wfile.write(b'--FRAME\r\n')
                    self.send_header('Content-Type', 'image/jpeg')
                    self.send_header('Content-Length', len(frame))
                    self.end_headers()
                    self.wfile.write(frame)
                    self.wfile.write(b'\r\n')
            except Exception as e:
                logging.warning(
                    'Removed streaming client %s: %s',
                    self.client_address, str(e))
        else:
            self.send_error(404)
            self.end_headers()

class StreamingServer(socketserver.ThreadingMixIn, server.HTTPServer):
    allow_reuse_address = True
    daemon_threads = True

picam2 = Picamera2()
picam2.configure(picam2.create_video_configuration(main={"size": (1280, 720)}))
output = StreamingOutput()

# picam2.start_recording(JpegEncoder(), FileOutput(output))

# encoder = JpegEncoder()
# encoder.output = FileOutput(output)
# picam2.start_encoder(encoder)
# picam2.start()

# encoder = JpegEncoder()
# output1 = FileOutput()
# encoder.output = [output]
# picam2.start_encoder(encoder)
# picam2.start()

encoder = JpegEncoder()
output1 = FileOutput(output)
# output2 = FfmpegOutput("fat.mp4")
output2 = FileOutput()
encoder.output = [output1, output2]
picam2.start_encoder(encoder)
picam2.start()
output2.fileoutput = "blooptest.mjpg"
output2.start()
time.sleep(5)
output2.stop()

# output2.fileoutput = "bloopey.h264"
# output2.start()
# time.sleep(5)
# output2.stop()

try:
    address = ('', 8000)
    server = StreamingServer(address, StreamingHandler)
    server.serve_forever()
finally:
    picam2.stop_recording()
`

davidplowman, 2022-08-10:
Hi, thanks for posting the question. I've tried your script and as far as I can tell, it's doing what I would expect. I see the following:

1. It starts the camera, writing to two outputs.
2. output1 is actually the one used by the webserver, and output2 is initially not enabled.
3. Then it starts output2, writing to "blooptest.mjpg", which runs for about 5 seconds.
4. Next, output2 is stopped.
5. Finally it moves on and starts the webserver, which seems to work fine when I connect to it.
6. I can play blooptest.mjpg (for example with `ffplay`) which contains ~5 seconds of video.

Could you maybe explain what you wanted to happen, then hopefully we can figure out what needs to be done differently. I wonder if there's some confusion because the webserver is only starting right at the end of the script, via the `server_forever()` method which (like the name suggests...) never returns.  Thanks!

tennisparty, 2022-08-10:
Thankyou, that's really helpful. I'm aiming to display the stream in the web server whilst recording video to a file at the same time. 

Currently I think this is happening separately i.e. step
3. records file (for 5 seconds)
5. streams to webserver (after step 3 but not at the same time)

When I try and open the mjpg file (or other video extensions) vlc only shows a single frame, are you seeing a full 5 second video?

Many thanks,
Sam

tennisparty, 2022-08-10:
I've changed the end of the code to the following which seems to allow the recording and server to run at the same time. Let me know if this is how you would do it.

I'm still having problems with the bloopey.mjpg file which is a static image and not a video

'''
encoder = JpegEncoder()
output1 = FileOutput(output)
# output2 = FfmpegOutput("fat.mp4")
output2 = FileOutput()
encoder.output = [output1, output2]
picam2.start_encoder(encoder)
picam2.start()
output2.fileoutput = "bloopey.mjpg"

def func1():
    try:
        address = ('', 8000)
        server = StreamingServer(address, StreamingHandler)
        server.serve_forever()
    finally:
        picam2.stop_recording()

def func2():
    output2.start()
    print("sleeping")
    time.sleep(5)
    print("sleep ends")
    output2.stop()

if __name__ == '__main__':
    a = Thread(target = func1)
    b = Thread(target = func2)
    a.start()
    b.start()
    a.join()
    b.join()
'''

davidplowman, 2022-08-10:
I think your code with the threads looks fine, you don't have to run everything in another thread, but there's no harm in it.

Ah, I was going to add that I think the mjpeg file is "fine" just as your last message arraived. The trouble with mjpeg is that it's not a "real" standard, it's just a collection of "what folks do". As such it can be tricky to know exactly how to interpret it. As you've discovered, vlc doesn't like this flavour of mjpeg out of the box, though you can force it to do the right thing with:
```
cvlc --demux=avformat blooptest.mjpg
```
Anyway, glad it's working better!

tennisparty, 2022-08-10:
I was trying to export to mp4 via FfmpgOutput and finding the same with those files. Working fine with ffplay but not VLC. Is there a program you recommend for converting these files to something more standard media players can read?

davidplowman, 2022-08-11:
mp4 files that you create with `FfmpegOutput` should play fine in VLC, in fact with any player. If you had some (simple!) code that creates mp4 files that _don't_ play in VLC I'd be interested to see. It might be worth experimenting with some very simple examples first, such as [this one](https://github.com/raspberrypi/picamera2/blob/main/examples/mp4_capture.py). Note that we do know that VLC _doesn't_ play h.264 files properly whereas (for example) `ffplay` does. (By h.264 file, I mean an h.264 bitstream that hasn't been wrapped in any kind of container format.)

To answer the general point, your go-to program for doing any conversion of video files is going to be ffmpeg, it supports so much and is so comprehensive. Of course that can make it a bit tricky to use correctly sometimes, though there is reams and reams of help online...

tennisparty, 2022-08-11:
Hi, do you know is it possible to initiate two encoders simultaneously? 

I would like to convert a lowres stream to mjpeg for the web stream and at the same time convert the main stream to h264 and save in files. Is this possible? Would it take a lot of processing power?

davidplowman, 2022-08-11:
_Picamera2_ won't run two encoders at once. To be fair I think it could be made to work, but it isn't supported now. 

Nonetheless I think you could still do what you describe if you can compromise on the image resolutions and framerate of the mjpeg stream. Here's what I might try:

1. You'll have to stop using _Picamera2_'s encoder for mjpeg but use it to save your h.264/mp4 file instead.
2. Implement the mjpeg encoder in your own loop (running in its own thread). This loop will (forever):
    1. Call `picam2.capture_array("lores")` to get the image.
    2. Python doesn't like YUV420 (which the lores stream has to be). OpenCV has a [reasonably quick conversion function](https://github.com/raspberrypi/picamera2/blob/main/examples/yuv_to_rgb.py#L16) that you can use to get RGB.
    3. Invoke the simplejpeg encoder to encode this [(just like the `JpegEncoder` class does)](https://github.com/raspberrypi/picamera2/blob/main/picamera2/encoders/jpeg_encoder.py#L37).
    4. Write the output to `self.frame` just as the original code does at the top of this report.
3. Apart from that, I think everything else should mostly work as before.

As I said, you'll get worse mjpeg performance, not least because when _Picamera2_ runs the jpeg encoder it uses all 4 cores. Here we're just stuck with one thread, but on the upside, it means it can't eat your entire CPU! (It might even be possible to make use of the `JpegEncoder` directly so as to regain the multi-threaded behaviour, but I think that's a bit of a stretch initially.)

tennisparty, 2022-08-11:
Thanks so much for the advice, I've tried this afternoon but I don't think I have the coding skills. I'll digest for a few days and see if I can work through the ideas again and then send through what I have.

In the meantime as an alternative, is there a way to compress the mjpeg stream with  ffmpegoutput whilst saving video files. I tried using the -crf (constant rate factor) and -b (bitrate) flags but they don't seem to change the saved file size. I'm sure there is a way.

Thanks again,
Sam

tennisparty, 2022-08-12:
Sorry I realise I've asked lots of questions so will stop after this. I think on reflection the simplest method for me will be to start a h264 encoder and then in one thread convert the h264 to mjpeg with ffmpeg (or another means) for the webpage. Is this possible? I think this should be simpler

davidplowman, 2022-08-12:
Hi, well that sounds plausible to me, though the devil is always in the details of course. That project you linked looks quite interesting. Clearly someone's put a bit of time and effort into it, though there's a bit too much magic Javascript for me to understand it easily!!

So I was going to make one final contribution too. I've taken the original mjpeg server and made the jpeg encoding run "by hand", that is, it's no longer the _Picamera2_ object that's driving it. It seems to run reasonably, though as I said previously, I would expect lower framerates. But the point is that you can then get _Picamera2_ to run the h.264 encoder, and save the result to h.264 or mp4 files in the "usual" way. The code is just for illustration but anyway, here it is:
```
import logging
import socketserver
from http import server
from threading import Condition, Thread
import simplejpeg
import cv2

from picamera2 import Picamera2
from picamera2.outputs import FileOutput

PAGE = """\
<html>
<head>
<title>picamera2 MJPEG streaming demo</title>
</head>
<body>
<h1>Picamera2 MJPEG Streaming Demo</h1>
<img src="stream.mjpg" width="640" height="480" />
</body>
</html>
"""

def mjpeg_encode():
    global mjpeg_frame
    while not mjpeg_abort:
        yuv = picam2.capture_array("lores")
        rgb = cv2.cvtColor(yuv, cv2.COLOR_YUV420p2RGB)
        buf = simplejpeg.encode_jpeg(rgb, quality=80, colorspace='BGR', colorsubsampling='420') 
        with mjpeg_condition:
            mjpeg_frame = buf
            mjpeg_condition.notify_all()

class StreamingHandler(server.BaseHTTPRequestHandler):
    def do_GET(self):
        if self.path == '/':
            self.send_response(301)
            self.send_header('Location', '/index.html')
            self.end_headers()
        elif self.path == '/index.html':
            content = PAGE.encode('utf-8')
            self.send_response(200)
            self.send_header('Content-Type', 'text/html')
            self.send_header('Content-Length', len(content))
            self.end_headers()
            self.wfile.write(content)
        elif self.path == '/stream.mjpg':
            self.send_response(200)
            self.send_header('Age', 0)
            self.send_header('Cache-Control', 'no-cache, private')
            self.send_header('Pragma', 'no-cache')
            self.send_header('Content-Type', 'multipart/x-mixed-replace; boundary=FRAME')
            self.end_headers()
            try:
                while True:
                    with mjpeg_condition:
                        mjpeg_condition.wait()
                        frame = mjpeg_frame
                    self.wfile.write(b'--FRAME\r\n')
                    self.send_header('Content-Type', 'image/jpeg')
                    self.send_header('Content-Length', len(frame))
                    self.end_headers()
                    self.wfile.write(frame)
                    self.wfile.write(b'\r\n')
            except Exception as e:
                logging.warning(
                    'Removed streaming client %s: %s',
                    self.client_address, str(e))
        else:
            self.send_error(404)
            self.end_headers()

class StreamingServer(socketserver.ThreadingMixIn, server.HTTPServer):
    allow_reuse_address = True
    daemon_threads = True

picam2 = Picamera2()
picam2.configure(picam2.create_video_configuration(main={"size": (640, 480)}, lores={}))
picam2.start()

mjpeg_abort = False
mjpeg_frame = None
mjpeg_condition = Condition()
mjpeg_thread = Thread(target=mjpeg_encode, daemon=True)
mjpeg_thread.start()

try:
    address = ('', 8000)
    server = StreamingServer(address, StreamingHandler)
    server.serve_forever()
finally:
    mjpeg_abort = True
    mjpeg_thread.join()
```

tennisparty, 2022-08-12:
Amazing, thank you, that is so useful :) I'll try and digest what you have done!

tennisparty, 2022-08-17:
Hello,

I have got the code recording to H264 files 

I was wondering if there is a way to add in a motion detection trigger before the H264 files record. Ideally with a circular buffer but I am working towards this. I have tried using the example [capture_motion.py](https://github.com/raspberrypi/picamera2/blob/main/examples/capture_motion.py) code but because this initiates another lores buffer I am getting threading issues `RuntimeError: Failure to wait for previous operation to finish!`. I think this is conflicting with the lores buffer array capture in the mjpeg_encode function. 

Here is what I have so far. The motion() function defined at line 79 and called at line 128/129

If there is an alternative way to detect motion using openCV or otherwise which makes more sense do let me know
```
import logging
import socketserver
from http import server
from threading import Condition, Thread
import simplejpeg
import cv2

from picamera2 import Picamera2
from picamera2.outputs import FileOutput
from picamera2.encoders import H264Encoder
from picamera2.outputs import CircularOutput
import time

PAGE = """\
<html>
<head>
<title>picamera2 MJPEG streaming demo</title>
</head>
<body>
<h1>Picamera2 MJPEG Streaming Demo</h1>
<img src="stream.mjpg" width="640" height="480" />
</body>
</html>
"""

def mjpeg_encode():
    global mjpeg_frame
    while not mjpeg_abort:
        yuv = picam2.capture_array("lores")
        rgb = cv2.cvtColor(yuv, cv2.COLOR_YUV420p2RGB)
        buf = simplejpeg.encode_jpeg(rgb, quality=80, colorspace='BGR', colorsubsampling='420') 
        with mjpeg_condition:
            mjpeg_frame = buf
            mjpeg_condition.notify_all()

class StreamingHandler(server.BaseHTTPRequestHandler):
    def do_GET(self):
        if self.path == '/':
            self.send_response(301)
            self.send_header('Location', '/index.html')
            self.end_headers()
        elif self.path == '/index.html':
            content = PAGE.encode('utf-8')
            self.send_response(200)
            self.send_header('Content-Type', 'text/html')
            self.send_header('Content-Length', len(content))
            self.end_headers()
            self.wfile.write(content)
        elif self.path == '/stream.mjpg':
            self.send_response(200)
            self.send_header('Age', 0)
            self.send_header('Cache-Control', 'no-cache, private')
            self.send_header('Pragma', 'no-cache')
            self.send_header('Content-Type', 'multipart/x-mixed-replace; boundary=FRAME')
            self.end_headers()
            try:
                while True:
                    with mjpeg_condition:
                        mjpeg_condition.wait()
                        frame = mjpeg_frame
                    self.wfile.write(b'--FRAME\r\n')
                    self.send_header('Content-Type', 'image/jpeg')
                    self.send_header('Content-Length', len(frame))
                    self.end_headers()
                    self.wfile.write(frame)
                    self.wfile.write(b'\r\n')
            except Exception as e:
                logging.warning(
                    'Removed streaming client %s: %s',
                    self.client_address, str(e))
        else:
            self.send_error(404)
            self.end_headers()

class StreamingServer(socketserver.ThreadingMixIn, server.HTTPServer):
    allow_reuse_address = True
    daemon_threads = True

def motion():
    lsize = (320, 240)
    w, h = lsize
    prev = None
    encoding = False
    ltime = 0
    while True:
    # for x in range(0, 260000):
    #     print(x)
        cur = picam2.capture_buffer("lores")
        # cur = yuv
        cur = cur[:w*h].reshape(h, w)
        if prev is not None:
            # Measure pixels differences between current and
            # previous frame
            mse = np.square(np.subtract(cur, prev)).mean()
            if mse > 7:
                if not encoding:
                    encoder.output = FileOutput("{}.h264".format(int(time.time())))
                    picam2.start_encoder()
                    encoding = True
                    print("New Motion", mse)
                    print("sleeping for 1..")
                    time.sleep(1)
                    print("out of loop?")
                ltime = time.time()
            else:
                if encoding and time.time() - ltime > 2.0:
                    print("in already encoding loop")
                    picam2.stop_encoder()
                    encoding = False
        prev = cur

picam2 = Picamera2()
picam2.configure(picam2.create_video_configuration(main={"size": (640, 480)}, lores={}))
encoder = H264Encoder(1000000, repeat=True)
# initiate circular output - also works without
encoder.output = CircularOutput()
picam2.encoder = encoder
picam2.start()
picam2.start_encoder()

mjpeg_abort = False
mjpeg_frame = None
mjpeg_condition = Condition()
mjpeg_thread = Thread(target=mjpeg_encode, daemon=True)
mjpeg_thread.start()

motion_thread = Thread(target=motion, daemon=True)
motion_thread.start()

def func1():
    try:
        address = ('', 8000)
        server = StreamingServer(address, StreamingHandler)
        server.serve_forever()
    finally:
        mjpeg_abort = True
        mjpeg_thread.join()

def func2():
    mp4string = time.strftime("%Y%m%d-%H%M%S") + '.h264'
    output2 = FileOutput(mp4string)
    for i in range(30):
        output2.start()
        # picam2.start_recording('bloop.mjpeg')
        print("sleeping")
        time.sleep(10)
        print("sleep ends")
        output2.stop()
        mp4string = time.strftime("%Y%m%d-%H%M%S") + '.h264'
        output2 = FileOutput(mp4string)
        encoder.output = [output2]
        print(i)

if __name__ == '__main__':
    a = Thread(target = func1)
    b = Thread(target = func2)
    a.start()
    b.start()
    a.join()
    b.join()
```

davidplowman, 2022-08-17:
Hi, yes, if you had two threads both trying to call something like `picam2.capture_buffer` then that won't work and you'll get errors when the calls collide. You could protect such accesses with a lock so that each call is guaranteed to finish before the next can start, but I don't think that's what you want - because each call will "consume" a camera frame, with the result that both threads would run at half the full framerate. To make that kind of thing work you'd probably need a single thread calling `capture_buffer`, and then notifying all the other client threads when a new frame has been copied out (a bit like `mjpeg_encode` does above). But that is getting more complicated!

As regards doing motion detection, the code above just computes the MSE between two successive frames (at least, the greyscale channel of each). This isn't a bad way to do it, but it's fairly simplistic. It will detect "lots of pixels changing a bit", but would miss "fewer pixels changing a lot". It depends a bit what you want the detector to be sensitive to.

If I were doing this sort of thing, I'd be tempted to compute the absolute difference between the greyscale channels, either directly with numpy or using `cv2.absdiff`. Then I'd calculate the histogram of the resulting image (`np.histogram` or `cv2.calcHist`) and also probably make a cumulative histogram (`np.cumsum`?) out of the result because certain operations are simply easier with cumulative values.

Finally, these histograms easily let you answer questions like "has this percentage of pixels changed by at least this amount?" so it's easy to cover both the many/few changed pixels scenarios that we had earler. It's also (relatively) efficient to calculate things like the mean (or even MSE) because you only have to traverse the histogram, not the entire image.

On the downside, histograms lose spatial information about where changes are. You could compute multiple histograms for different parts of the image, but that's starting to feel complicated again. Like I said, it all depends what you want to detect. Does that help at all?

tennisparty, 2022-08-17:
That's really helpful. I've seen a few motion detection algorithms in cv2 so will have a think about the best formula when I get to that stage. I'm recording bats jumping out of holes in trees, which are usually small, fast and only a portion of the image.

 I've had a go at starting capturing a buffer and feeding this into the motion code but can't quite get a result. This is what I have so far

```
#!/usr/bin/python3

# Mostly copied from https://picamera.readthedocs.io/en/release-1.13/recipes2.html
# Run this script, then point a web browser at http:<this-ip-address>:8000
# Note: needs simplejpeg to be installed (pip3 install simplejpeg).
print("hello world")
import io
import logging
import socketserver
from http import server
from threading import Condition, Thread

from picamera2 import Picamera2
from picamera2.encoders import JpegEncoder
from picamera2.outputs import FfmpegOutput
from picamera2.outputs import FileOutput
# from datetime import datetime
import time

import numpy as np

PAGE = """\
<html>
<head>
<title>picamera2 MJPEG streaming demo</title>
</head>
<body>
<h1>Picamera2 MJPEG Streaming Demo</h1>
<img src="stream.mjpg" width="640" height="480" />
</body>
</html>
"""

def capturebuffer():
    global cb_frame
    global buf2
    while not cb_abort:
        buf2 = picam2.capture_buffer("lores")
        with cb_condition:
            cb_frame = buf2
            cb_condition.notify_all()

class StreamingOutput(io.BufferedIOBase):
    def __init__(self):
        self.frame = None
        self.condition = Condition()

    def write(self, buf):
        with self.condition:
            self.frame = buf
            self.condition.notify_all()

class StreamingHandler(server.BaseHTTPRequestHandler):
    def do_GET(self):
        if self.path == '/':
            self.send_response(301)
            self.send_header('Location', '/index.html')
            self.end_headers()
        elif self.path == '/index.html':
            content = PAGE.encode('utf-8')
            self.send_response(200)
            self.send_header('Content-Type', 'text/html')
            self.send_header('Content-Length', len(content))
            self.end_headers()
            self.wfile.write(content)
[... 101 lines cut ...]
        output2.start()
        # picam2.start_recording('bloop.mjpeg')
        print("sleeping")
        time.sleep(300)
        print("sleep ends")
        output2.stop()
        mp4string = time.strftime("%Y%m%d-%H%M%S") + '.mjpg'
        output2 = FileOutput(mp4string)
        encoder.output = [output1, output2]
        print(i)

def func3():
    motion()

if __name__ == '__main__':
    a = Thread(target = func1)
    b = Thread(target = func2)
    c = Thread(target = func3)
    a.start()
    b.start()
    c.start()
    a.join()
    b.join()
    c.join()
```

tennisparty, 2022-08-18:
I've stripped everything back whilst working on it a bit this morning and seem to have a capture buffer feeding into a motion loop. The motion loop exits after a few runs which I don't quite understand but probably need some time to digest. Is this on the right track? If I use the same 'capture buffer' to stream to a server is there anything I need to consider?

Sorry for asking so many questions, I'm new to this

```
#!/usr/bin/python3

# Mostly copied from https://picamera.readthedocs.io/en/release-1.13/recipes2.html
# Run this script, then point a web browser at http:<this-ip-address>:8000
# Note: needs simplejpeg to be installed (pip3 install simplejpeg).
print("hello world")
import io
import logging
import socketserver
from http import server
from threading import Condition, Thread

from picamera2 import Picamera2
from picamera2.encoders import JpegEncoder
from picamera2.outputs import FfmpegOutput
from picamera2.outputs import FileOutput
# from datetime import datetime
import time

import numpy as np

def capturebuffer():
    global cb_frame
    global buf2
    while not cb_abort:
        # print("while cb abort")
        buf2 = picam2.capture_buffer("lores")
        # print(buf2)
        with cb_condition:
            cb_frame = buf2
            # print(cb_frame)
            # print("with cb cond")
            cb_condition.notify_all()

def motion():
    print("in motion loop")
    # global buf2
    # global cb_condition
    lsize = (320, 240)
    w, h = lsize
    prev = None
    encoding = False
    ltime = 0
    try:
        while True:
            print("motion true")
            with cb_condition:
                print("printing cb frame")
                print(cb_frame)
                cb_condition.wait()
                print("printing cb frame after wait")
                print(buf2)
                cur = cb_frame
                # print(cur)
                cur = cur[:w*h].reshape(h, w)
                print("reshaped cur:")
                print(cur)
                if prev is not None:
                    # print("prev is not none")
                    # Measure pixels differences between current and
                    # previous frame
                    mse = np.square(np.subtract(cur, prev)).mean()
                    # print(cur)
                    # print(prev)
                    # print(mse)
                    if mse > 7:
                        if not encoding:
                            encoder.output = FileOutput("{}.h264".format(int(time.time())))
                            picam2.start_encoder()
                            encoding = True
                            print("New Motion", mse)
                            print("sleeping for 1..")
                            time.sleep(1)
                            print("out of loop?")
                        ltime = time.time()
                    else:
                        # if encoding and time.time() - ltime > 2.0:
                        print("in already encoding loop")
                            # picam2.stop_encoder()
                            # encoding = False
                print("changing prev to cur")
                prev = cur
                print(prev)
    except:
        print("exception at end of motion loop")

picam2 = Picamera2()
picam2.configure(picam2.create_video_configuration(main={"size": (1280, 720)}, lores={}))
encoder = JpegEncoder()
picam2.start_encoder(encoder)
picam2.start()

cb_abort = False
cb_frame = None
buf2 = None
cb_condition = Condition()
cb_thread = Thread(target=capturebuffer, daemon=True)
cb_thread.start()

# print(buf2)

motion_thread = Thread(target=motion, daemon=True)
motion_thread.start()
motion_thread.join()
cb_thread.join()
```

davidplowman, 2022-08-18:
I think that looks OK to me. I possibly would avoid trying to hold the lock so long (`with cb_condition:`) in the `motion` loop, I think you can give it up once you've done `cur = cb_frame`. That should make things run a bit better, especially if you add a jpeg encode loop (you really don't want to be holding the lock while doing the encode).

It looks to me like all you really need to do is add back the jpeg encode which then feeds the web server. If you use the "do it by hand" technique from before, then the Picamera2's "built-in" encoder remains available for recording h.264 or mp4 files. The `mjpeg_encode` loop should look just like it did before, but waiting for the `cb_frame` instead of calling `picam2.capture_array` as we did previously.

tennisparty, 2022-08-18:
Thankyou so much. I have changed the cb_condition and all seems good.

I have added the streaming code and modified the mjpeg_encode loop as follows:
```
#!/usr/bin/python3

# Mostly copied from https://picamera.readthedocs.io/en/release-1.13/recipes2.html
# Run this script, then point a web browser at http:<this-ip-address>:8000
# Note: needs simplejpeg to be installed (pip3 install simplejpeg).
print("hello world")
import io
import logging
import socketserver
from http import server
from threading import Condition, Thread

from picamera2 import Picamera2
from picamera2.encoders import JpegEncoder
from picamera2.encoders import H264Encoder
from picamera2.outputs import FfmpegOutput
from picamera2.outputs import FileOutput
# from datetime import datetime
import time

import numpy as np
import simplejpeg
import cv2

PAGE = """\
<html>
<head>
<title>picamera2 MJPEG streaming demo</title>
</head>
<body>
<h1>Picamera2 MJPEG Streaming Demo</h1>
<img src="stream.mjpg" width="640" height="480" />
</body>
</html>
"""

def capturebuffer():
    global cb_frame
    global buf2
    while not cb_abort:
        # print("while cb abort")
        buf2 = picam2.capture_buffer("lores")
        # print(buf2)
        with cb_condition:
            cb_frame = buf2
            # print(cb_frame)
            # print("with cb cond")
            cb_condition.notify_all()

def mjpeg_encode():
    global mjpeg_frame
    while not mjpeg_abort:
            with cb_condition:
                # print("printing cb frame")
                # print(cb_frame)
                # cb_condition.wait()
                # print("printing cb frame after wait")
                # print(cb_frame)
                yuv = cb_frame
                print("yuv")
                print(yuv)
            rgb = cv2.cvtColor(yuv, cv2.COLOR_YUV420p2RGB)
            buf = simplejpeg.encode_jpeg(rgb, quality=80, colorspace='BGR', colorsubsampling='420') 
            with mjpeg_condition:
                mjpeg_frame = buf
                mjpeg_condition.notify_all()
[... 121 lines cut ...]
cb_condition = Condition()
cb_thread = Thread(target=capturebuffer, daemon=True)
cb_thread.start()

mjpeg_abort = False
mjpeg_frame = None
mjpeg_condition = Condition()
mjpeg_thread = Thread(target=mjpeg_encode, daemon=True)
mjpeg_thread.start()

motion_thread = Thread(target=motion, daemon=True)
motion_thread.start()
motion_thread.join()
mjpeg_thread.join()
cb_thread.join()
```

I am getting the following error - 
`  File "/home/pi/18Augverysimple2.py", line 63, in mjpeg_encode
    rgb = cv2.cvtColor(yuv, cv2.COLOR_YUV420p2RGB)
cv2.error: OpenCV(4.6.0) /tmp/pip-wheel-u79916uk/opencv-python_ea2489746b3a43bfb3f2b5331b7ab47a/opencv/modules/imgproc/src/color.simd_helpers.hpp:108: error: (-215:Assertion failed) sz.width % 2 == 0 && sz.height % 3 == 0 in function 'CvtHelper'`

The yuv variable is printing as None and cv2 isn't receiving the capture buffer cb_frame data. I'm a bit confused why that is different from the cur variable in the motion code which does receive the capture buffer cb_frame data.

Thanks again

davidplowman, 2022-08-18:
Ah, I think you need to use `capture_array` instead of `capture_buffer` in your `capturebuffer` function. You'll have to update this line `cur = cur[:w*h].reshape(h, w)` in your `motion` function too, something like `cur = cur[:h,:]` should do.

The reason is that `capture_buffer` gives you a flat and very long 1d array, which OpenCV doesn't like. `capture_array` gives you a 2d array which is what OpenCV expects, but now your motion function doesn't need to convert the 1d array into a 2d one. You can just pick the first h rows, which are the greyscale part of the image.

tennisparty, 2022-08-18:
Amazing, thank you, phwoar, that's all working! Thankyou so much for your help :)

Is there a way to increase the frame rate to 60fps? I am using a ov9281 and have tried the following video config but files are still written at 30fps:
picam2.configure(picam2.create_video_configuration(controls={"FrameDurationLimits": (16666, 16666)},main={"size": (1280, 720)}, lores={"size": (640, 480)}))

davidplowman, 2022-08-18:
How are you saving the files? Certain file formats (h.264 files, mjpeg files for example) don't actually contain any timing information so it's up to the playing software to pick a random framerate, which may well be 30fps. It depends on the software, but I'd expect there to be an option to change that.

If you're saving something like an mp4 file, that should contain timing information, so you'd have to check what framerate the camera is really running at (just start the camera and try `picam2.capture_metadata()["FrameDuration"]`).

tennisparty, 2022-08-18:
Perfect thank you, I'll check that :)

tennisparty, 2022-08-18:
I seem to get some strange banding at the top of the videos at higher frame rates 

I'm using a circular loop now. I've tried changing the number of buffers but it doesn't seem to change the image
[image]
This is less pronounced but still visible on recordings
[image]

davidplowman, 2022-08-19:
My initial reaction might be some kind of hardware or signal integrity problem, but of course it's hard to be sure. Some things to check:

* How long is the ribbon cable between your Pi and the camera?
* Do you see this corruption in both the main and lores streams?
* Maybe try running the camera with a very minimal script to see if there's any difference.
* Could you save one of these correct images as a png so that there's no compression and post that?
* At what framerate do you start to see a problem?

Thanks.

tennisparty, 2022-08-20:
Hi, I've tried to recreate the problem but had no success. I did re-install the pi OS, now 64 bit and have moved the camera around a few times so can't be sure what caused the issue. I'll troubleshoot with those steps if it does start again.

closed 2022-09-08 by tennisparty

azatalex96, 2022-12-15:
is there a way to check if camera recording? 

in picamera camera.recording == True was doing the job, but i couldn't find the function in picamera2

azsde, 2023-01-11:
@tennisparty I would like to achieve a similar result, would you mind sharing your code ? :)

jevchance, 2023-05-26:
@tennisparty would you please share your code? Thank you.