raspberrypi/picamera2 #258: [HOW-TO] Multiple cameras using CM4 & Picamera2

neil-benn, 2022-08-11:
Hello,

  I'm running a CM4 with libcamera; the libcamera apps can capture from both cameras using the command line apps.  I'm now trying to capture from both cameras using picamera2 but I can't see how to specify which camera to connect to.  Is this available in picamera2?

  I could look at using v4l2 or picking through the code from the RPi libcamera-apps to adapt into a python library but that would take much longer to handle and be duplicating efforts made here.

Thanks.

Regards,

Neil

davidplowman, 2022-08-12:
Hi, the _Picamera2_ constructor takes a `camera_num` argument, defaulting to zero, which should in theory specify which camera to use.

The catch is that I haven't ever got round to trying this on a CM4. With luck and a following wind, it might work like libcamera-apps do, i.e. you can start separate Python interpreters and attach each to a different camera (there is no real handling of dual cameras within libcamera itself). Though the usual rule applies that all software is broken until you've tested it. I'd be interested to hear what happens...

neil-benn, 2022-08-12:
Hello,

  Great news - I'm on the case and I'll see what happens!

Cheers,

Neil

neil-benn, 2022-08-12:
Hello,

  I tested this today and accessing one camera in one process worked just fine using picamera2 on cam 0 and cam 1.  I used a still config, started the camera and took 20 or 30 images without a problem.

  Thanks

Cheers,

Neil

http://www.ziath.com<http://www.ziath.com/>

CorvetteCole, 2022-08-31:
this should probably be closed. camera interface works just fine on CM4 with multiple processes even

closed 2022-08-31 by davidplowman