raspberrypi/picamera2 #259: [HOW-TO] Example for MQTT

gbernal, 2022-08-13:
Hi,

I would like to use picamera2 to send images periodically or video stream from my headless raspberry pi 3b+ to a reactjs client that uses MQTT to receive data. 
Would the process be similar to https://github.com/raspberrypi/picamera2/blob/main/examples/capture_stream_udp.py ?

Thank you in advance.

davidplowman, 2022-08-15:
Hi, I think it sounds like you might want to create your own Output object, with its own write method? Maybe take a look [here](https://github.com/raspberrypi/picamera2/blob/main/examples/mjpeg_server.py#L30), where it does something a bit like this. So perhaps in your `__init__` method you want to store the "client" object, and in the `write` method you then do `self.client.publish`?

gbernal, 2022-08-16:
Thank you for the suggestion @davidplowman, I will explore your suggestion and see how far I'm able to get with. I will post back in here any progress or issues.

davidplowman, 2022-09-13:
Hi, just wondering if there's any else to follow up on this one? Please post back if there is, otherwise I'll consider closing this issue in a few days. Thanks!

davidplowman, 2022-09-26:
Closing this now. Please feel free to open a new issue if further help is needed. Thanks!

closed 2022-09-26 by davidplowman