raspberrypi/picamera2 #260: how to capture in BGR ?

zoldaten, 2022-08-14:
hi!
in Picamera there was something like this:
```
for frame in camera.capture_continuous(capture, format="bgr", use_video_port=True,
                                       resize=(img_width,img_height)):
```
for Picamera2 there it goes:
```
while True:    
    origin_img = camera.capture_array()
    origin_img = cv2.cvtColor(origin_img, cv2.COLOR_BGRA2RGB)
```
i have to convert each frame to BGR.
Perhaps, there is a better way to do that ?

I thied camera.format = "RGB888" but failed.

Tim-Brown-NZ, 2022-08-15:
Have you tried specifying the format in the definition of main when creating the configuration. Something like:
_config = self.picam2.create_preview_configuration(main={"format": "RGB888"}) 
self.picam2.configure(config)
self.picam2.start()_

zoldaten, 2022-08-15:
got it!

closed 2022-08-15 by zoldaten