raspberrypi/picamera2 #261: how to capture in stereo mode ?

zoldaten, 2022-08-14:
there was an option in Picamera:
```camera = PiCamera(stereo_mode='side-by-side',stereo_decimate=False)```
that lets to capture from both cameras.

Now i see that 2 cameras registered:
```
[3:12:24.466472615] [28917]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3700-f30ad033
[3:12:24.502559212] [28929]  INFO RPI raspberrypi.cpp:1368 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media0 and ISP device /dev/media1
[3:12:24.506693130] [28917]  INFO Camera camera.cpp:1029 configuring streams: (0) 640x480-XBGR8888
```
But i can use just one camera.
Is there possible to start up second camera at once ?

davidplowman, 2022-08-15:
Hi, I'm afraid _libcamera_ doesn't have any stereo capture support, so _Picamera2_ doesn't either. It is on the list of things to do, but unfortunately it still seems like quite a long way off.

You can start a second camera in a second Python process. The `Picamera2` constructor takes a `camera_num` argument which can be used to select the other camera.

zoldaten, 2022-08-15:
ok, thanks.

closed 2022-08-15 by zoldaten