raspberrypi/picamera2 #269: [SUGGESTION] Make optional GUI dependencies actually optional

mpizenberg, 2022-08-17:
Hi, I am trying to use picamera2 with [poetry (the package manager)](https://python-poetry.org/), but GUI dependencies are brought in and failed at installation saying there is no valid version for the qt5 deps. This issue happens because there is no precompiled wheel for those for the aarch64 architecture if I understand correctly.

I've seen in the readme that removing the GUI dependencies can be done by setting the NOGUI environment variable, but this does not seem to work when using the poetry package manager. However, I know that poetry support optional dependencies since I could specify some for my tinyrpc dependency as such:

```toml
tinyrpc = {extras = ["gevent", "wsgi", "httpclient"], version = "^1.1.5"}
```

I'm not super familiar with all the complexities of how python packaging work, but could it be possible to make the GUI dependencies of picamera2 optional in the same way?

davidplowman, 2022-08-17:
@chrisruk Hi Chris, I know we've looked at this, do you know what the latest state of play is? Thanks!

chrisruk, 2022-08-17:
I've not used poetry before I'm afraid.  pip/pypi also support extra deps via a wheel, but the problem is there's no way to make the GUI deps with that a 'default'.  

You should be able to install picamera2 using pip though - 
```
NOGUI=1 pip3 install git+https://github.com/raspberrypi/picamera2.git
```

It might be necessary for us to create 2 pypi packages, one without the GUI deps, if people want to use other python package managers.

davidplowman, 2022-08-17:
Very soon we're going to have an apt package which you'll be able to install with `--no-install-recommends` to avoid all the GUI stuff. Is that likely to be a workable substitute for everyone, do you think?

mpizenberg, 2022-08-17:
Poetry is very similar to pipenv, it enables handling virtual environments declaratively at the project level (instead of System or User levels) by using the standard `pyproject.toml` file introduced in PEP-518.

I've just had a look at how tinyrpc does its optional dependencies and it seems they just have an `extras_require` section in their `setup.py`:

```py
setup(
    name='tinyrpc',
    ...
    install_requires=['six'],
    extras_require={
        'gevent': ['gevent'],
        'httpclient': ['requests', 'websocket-client', 'gevent-websocket'],
        ...
        'rabbitmq': ['pika']
    }
)
```

Could picamera2 do the same for the GUI dependencies? Meaning have a `gui` extra that includes all the GUI dependencies for people who need them? As such people could both use declarative systems (like poetry, pipenv, ...) or use pip if they prefer.

```sh
pip install 'picamera2[gui]'
```

In my usecase, installing some of the dependencies via the OS (apt) and others via virtual env package manager (such as poetry) would be less than ideal. And the problem with using `NOGUI=1 pip ...` is that the dependency is not declaratively specified in the `pyproject.toml` (PEP-518).

chrisruk, 2022-08-17:
Yeah the problem is we want the GUI deps to be installed by default, rather than the user having to tell pip to install the GUI deps.  Ideally it'd be great if pip/pypi supported a 'default' extra require, but alas it doesn't

The deb packages will be great @davidplowman.  I did wonder if we want 2 pypi wheels though, for non-debian based distros, or do we only need to target Raspberry Pi OS?

mpizenberg, 2022-08-17:
> Yeah the problem is we want the GUI deps to be installed by default, rather than the user having to tell pip to install the GUI deps. Ideally it'd be great if pip/pypi supported a 'default' extra require, but alas it doesn't

Ah I see. Well I thought adding `[gui]` to the pip command was a very small price to pay to be PEP-518 compatible. (In addition to the fact that making optional dependencies substractive instead of additive is a very bad idea in general for the dependency solvers, that I know since I've been working on a dependency solver for the rust ecosystem).

But I totally understand if it's you preference.

mpizenberg, 2022-08-17:
> (see - https://discuss.python.org/t/adding-a-default-extra-require-environment/4898 )

Very interesting thread, from all parties. I can see how the core python devs tend to agree more on my side and how from a users perspective, it could be more useful to have defaults with GUI, especially for people with little understanding of the packaging tooling and installing things via for example pycharm.

It's a tough one. And from that thread, it seems to me that the most viable solution is having two packages distributed. Something like `picamera2` be the default on with gui, and something like `picamera2-lib` be the stripped down to minimal dependencies one.

chrisruk, 2022-08-18:
We've altered it now so that picamera2 doesn't install GUI dependencies by default, and you specify picamera2[gui] 
to install them.  (This is in branch 'next' but should be merged into 'main' soon and then into PyPi)

mpizenberg, 2022-08-22:
Awesome! I can confirm it’s working now for me with version 0.3.2

closed 2022-08-22 by mpizenberg