raspberrypi/picamera2 #270: Video recieve problem

smababou, 2022-08-17:
## Describe the bug
I am using picamera2 in an application for AGV localization, where one channel is used for markers localization and the other channel for line following
Even though i have set the resolution in video configuration, the output frames are disturbed

## Code
```python
import cv2
from picamera2 import Picamera2
from picamera2.encoders import H264Encoder

video_capture = Picamera2()
# Video Size Config
video_config = video_capture.create_video_configuration(main={'size': (1920, 1080), 'format': 'XRGB8888'}, 
                                                        lores={'size': (1920, 1080), 'format': 'YUV420'},
                                                        raw={'size': (640, 480), 'format': 'SGBRG10_CSI2P'}
                                                        )
video_capture.configure(video_config)
# Encoder Config 1
video_capture.encoder = H264Encoder(bitrate=1000000000, repeat=True, iperiod=100)
# Encoder Config 2
# video_capture.encoder = H264Encoder()
video_capture.start()
# opencv image write
frame = video_capture.capture_array("main")
cv2.imwrite('image.png', frame)
# picamera capture request
request = self.video_capture.capture_request()
request.save(name='main', file_output="main.png", format='png')
request.release()
```

## Console Output
```bash
{'use_case': 'video', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'JPEG'>, 'buffer_count': 6, 'main': {'format': 'XRGB8888', 'size': (1920, 1080), 'stride': 7680, 'framesize': 8294400}, 'lores': {'format': 'YUV420', 'size': (1920, 1080), 'stride': 1920, 'framesize': 3110400}, 'raw': {'format': 'SGBRG10_CSI2P', 'size': (640, 480), 'stride': 800, 'framesize': 384000}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Fast: 1>, 'FrameDurationLimits': (33333, 33333)}, 'display': 'main', 'encode': 'main'}
```

## Hardware
Raspberry Pi 4 Model B Rev 1.4

## Output frames
Add any other context about the problem
### main frame output
[image]
### lores frame output
[image]

## Expected behavior
Kindly support me to create the proper configuration for the camera in order to process the frames

davidplowman, 2022-08-17:
Hi, thank you for the question, but perhaps you can explain a little more what is happening.

Firstly, it looks like you're using the v1 camera, is that correct?

Secondly, you're asking for a heavily binned (640x480) mode from the sensor and then upscaling the images to 1920x1080 which causes the very poor image quality, though I think that is not the issue you are raising.

I ran your script and it saves two files ("image.png" and "main.png") and both look fine (apart from the poor image quality). I note that both are being taken from the "main" stream. (I also had to delete the `self.` characters when it captures the request near the end, otherwise the script crashes.)

Finally, I'm wondering whether you meant to capture the "lores" stream for the first image? This is a YUV420 format image, so the 2nd image that you attached shows exactly what I would expect. OpenCV functions mostly don't handle YUV420 so it is saving the Y, U and V planes all as if the whole thing is a single greyscale image. To convert the YUV image to RGB before saving you would have to use something like `rgb = cv2.cvtColor(yuv, cv2.COLOR_YUV420p2RGB)` first.

Hope that helps!

smababou, 2022-08-17:
## Hardware Description
I am currently using Joy-it camera (Chip: OV5647), which is the same chip as in rpi-cam v2, and with which i have tested the libcamera app, and it functions good
[image]
[image]

## Edit done
i have updated the code with these configuration
- [x] problem fixed the resolution 

```python
video_config = video_capture.create_video_configuration(main={'size': (1920, 1080), 'format': 'XRGB8888'}, 
                                                        lores={'size': (1920, 1080), 'format': 'YUV420'},
                                                        raw={'size': (1920, 1080), 'format': 'SGBRG10_CSI2P'}
```

## Problem
### I store lores frame using the following
```python
lores_frame = video_capture.capture_array("lores")
cv2.imwrite('lores_frame.png', lores_frame)
```
- I can be seen in the output image, it's half is corrupted 
[image]

### I store main frame using the following
```python
request = video_capture.capture_request()
request.save(name='main', file_output="main_frame.png", format='png')
request.release()
```
- I can be seen in the output image, it's half is corrupted 
[image]

## libcamera-still Output
[image]

## Console Output
```python
[0:01:25.688279157] [760]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3700-f30ad033
[0:01:25.759740569] [764]  INFO RPI raspberrypi.cpp:1368 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media4 and ISP device /dev/media0
[0:01:25.764248891] [760]  INFO Camera camera.cpp:1029 configuring streams: (0) 1920x1080-XRGB8888 (1) 1920x1080-YUV420 (2) 1920x1080-SGBRG10_CSI2P
[0:01:25.764668310] [764]  INFO RPI raspberrypi.cpp:759 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1920x1080-SGBRG10_1X10 - Selected unicam format: 1920x1080-pGAA
pi-nhs-0007 2022-08-17 16:35:17 cameraAccess[760] __init__:49 INFO: Camera is now running:
{'use_case': 'video', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'JPEG'>, 'buffer_count': 6, 'main': {'format': 'XRGB8888', 'size': (1920, 1080), 'stride': 7680, 'framesize': 8294400}, 'lores': {'format': 'YUV420', 'size': (1920, 1080), 'stride': 1920, 'framesize': 3110400}, 'raw': {'format': 'SGBRG10_CSI2P', 'size': (1920, 1080), 'stride': 2400, 'framesize': 2592000}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Fast: 1>, 'FrameDurationLimits': (33333, 33333)}, 'display': 'main', 'encode': 'main'}
```

davidplowman, 2022-08-17:
Thanks for the clarifications.

1. The lores image

The picture that you have attached is what I would expect. You've got a planar YUV420 image and OpenCV treats it as a greyscale image so the U and V planes appear below the Y. You need to use `cv2.cvtColor` as I explained previously.

2. The main image

I don't understand why you see this. When I capture an image like this I get a complete picture. What happens if you capture it as a JPEG? What happens if you display it in the preview (try `video_capture.start(show_preview=True)`)?

Thanks!

smababou, 2022-08-17:
1. for some applications i need a planar YUV420 image, so lores image fits me good.
2. I used the following code, and i got this image
```python
import time

from picamera2 import Picamera2
from picamera2.encoders import JpegEncoder

picam2 = Picamera2()
video_config = picam2.create_video_configuration(main={"size": (1920, 1080)})
picam2.configure(video_config)

picam2.start_preview()
encoder = JpegEncoder(q=70)

picam2.start_recording(encoder, 'test.mjpeg')
time.sleep(10)
picam2.stop_recording()
```

## Output

[image]

smababou, 2022-08-17:
i have solved the problem of receiving only part of the image, by adding a delay before storing the frame

while the lores image is fixed, by the following, as i needed gray image
```python
gray = cv2.cvtColor(yuv, cv2.COLOR_YUV420p2GRAY)
```

Thank you so much for your support

davidplowman, 2022-08-18:
Glad that things are working. I'll close this issue in that case, but please feel free to open a new one if anything else comes up. Thanks!

closed 2022-08-18 by davidplowman

0xaaaa, 2024-01-21:
> OpenCV functions mostly don't handle YUV420 so it is saving the Y, U and V planes all as if the whole thing is a single greyscale image. To convert the YUV image to RGB before saving you would have to use something like `rgb = cv2.cvtColor(yuv, cv2.COLOR_YUV420p2RGB)` first.

thank you so much, it works for me!