raspberrypi/picamera2 #282: [BUG] Start / Stop Picamera2 breaks the event loop in QPicamera2 / QGlPicamera2

Mauroq, 2022-08-23:
I'm trying to create an application for the Arducam 64mp. I have followed the instructions from Arducam to get a video output from the camera. So far everything is working. I can control the camera using the examples and the apps from this repository. The problem arises when trying the use the function "switch_mode_and_capture_file", after which the entire application stops responding.

I have traced this down to stopping and starting the Picamera2 class. 

Attaching a debugger, I noticed that after stopping and starting the class, the QPicamera2.handle_requests functions is not called any more. It would seem that the event loop here is broken somehow.

I have tested this with the QGlPicamera2 and QPicamera2 classes.

uname -a:
`Linux raspberrypi 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l GNU/Linux`

/boot/config.txt:
```
dtparam=audio=on
camera_auto_detect=1
display_auto_detect=1
max_framebuffers=2
disable_overscan=1
[pi4]
arm_boost=1
dtoverlay=vc4-kms-v3d,cma-512
gpu_mem=128

[all]
dtoverlay=arducam_64mp
```

/boot/cmdline.txt:
`console=serial0,115200 console=tty1 root=PARTUUID=a6f07418-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cma=512M`

Minimal example that shows the issue (inspired from app_capture2.py):
```
#!/usr/bin/python3

# This example is essentially the same as app_capture.py, however here
# we use the Qt signal/slot mechanism to get a callback (capture_done)
# when the capture, that is running asynchronously, is finished.

from PyQt5 import QtCore
from PyQt5.QtWidgets import (QApplication, QHBoxLayout, QLabel, QPushButton,
                             QVBoxLayout, QWidget)

from picamera2 import Picamera2
from picamera2.previews.qt import QGlPicamera2, QPicamera2

def post_callback(request):
    label.setText(''.join("{}: {}\n".format(k, v) for k, v in request.get_metadata().items()))

picam2 = Picamera2()
picam2.post_callback = post_callback
picam2.configure(picam2.create_preview_configuration(main={"size": (800, 600)}))

app = QApplication([])

def on_button_clicked():
    button.setEnabled(False)
    cfg = picam2.create_still_configuration()
    picam2.switch_mode_and_capture_file(cfg, "test.jpg", signal_function=qpicamera2.signal_done)

def on_button2_clicked():
    picam2.stop()
    picam2.start()

def capture_done():
    picam2.wait()
    button.setEnabled(True)

#qpicamera2 = QGlPicamera2(picam2, width=800, height=600, keep_ar=False)
qpicamera2 = QPicamera2(picam2, width=800, height=600, keep_ar=False)
button = QPushButton("Click to capture JPEG")
button2 = QPushButton("Start Stop")
label = QLabel()
window = QWidget()
qpicamera2.done_signal.connect(capture_done)
button.clicked.connect(on_button_clicked)
button2.clicked.connect(on_button2_clicked)

label.setFixedWidth(400)
label.setAlignment(QtCore.Qt.AlignTop)
layout_h = QHBoxLayout()
layout_v = QVBoxLayout()
layout_v.addWidget(label)
layout_v.addWidget(button)
layout_v.addWidget(button2)
layout_h.addWidget(qpicamera2, 80)
layout_h.addLayout(layout_v, 20)
window.setWindowTitle("Qt Picamera2 App")
window.resize(1200, 600)
window.setLayout(layout_h)

picam2.start()
window.show()
app.exec()

```

waisaacs, 2022-08-23:
For reference are you using an Arducam build of libcamera?  I had troubles some time ago trying to get one of the Arducam autofocus cameras working with picamera2, and it boiled down to some incompatibilities at the libcamera level.  Maybe things have improved since I tried, but still it may be relevant to know whether you are using the standard libcamera or not.

Mauroq, 2022-08-24:
I'm using the libcamera build of Arducam. 
For reference, the following command works and produces an Image output: 
`libcamera-still -e png -o img.png --autofocus --width 4624 --height 3472`

davidplowman, 2022-08-24:
@Mauroq Hi, can I start by getting a little more information?

1. If you run the app_capture2.py example completely unchanged, are you finding that the "capture JPEG" button causes the app to fail?
2. When you say the event loop isn't running, are you finding that the preview window stops updating?
3. When the app fails, are there any messages printed to the console? Is there anything suspicious in dmesg?

One thing I might suspect is that it's running out of memory and failing to allocate buffers, though I see you have your CMA bumped up to 512MB already. But be aware that the buffers required for 64MP stills capture in Python are truly vast (libcamera-still is more efficient because it can work with YUV420).

Another thought is that Arducam might not be keeping up with mainline libcamera, in which case working your way through older releases of _Picamera2_ might (or might not...) find one that works. But if the problem is related to the Arducam device/software then we're generally unable to help and you'd have to ask them for support.

Mauroq, 2022-08-24:
Hey Davidplowman,

1. If you run the app_capture2.py example completely unchanged, are you finding that the "capture JPEG" button causes the app to fail?
Yes the app still fails. The preview window stops updating, and an image is not taken. The log below:
```
[17:34:34.088596699] [9383]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3730-67300b62
[17:34:34.152791751] [9384]  WARN CameraSensorProperties camera_sensor_properties.cpp:174 No static properties available for 'arducam_64mp'
[17:34:34.152879861] [9384]  WARN CameraSensorProperties camera_sensor_properties.cpp:176 Please consider updating the camera sensor properties database
[17:34:34.178342143] [9384]  WARN RPI raspberrypi.cpp:1274 Mismatch between Unicam and CamHelper for embedded data usage!
[17:34:34.179093929] [9384] ERROR DelayedControls delayed_controls.cpp:87 Delay request for control id 0x009a090a but control is not exposed by device /dev/v4l-subdev0
[17:34:34.179373685] [9384]  INFO RPI raspberrypi.cpp:1398 Registered camera /base/soc/i2c0mux/i2c@1/arducam_64mp@1a to Unicam device /dev/media0 and ISP device /dev/media3
[17:34:34.183125652] [9383]  INFO Camera camera.cpp:1029 configuring streams: (0) 800x600-XBGR8888
[17:34:34.183571017] [9384]  INFO RPI raspberrypi.cpp:763 Sensor: /base/soc/i2c0mux/i2c@1/arducam_64mp@1a - Selected sensor format: 1280x720-SBGGR10_1X10 - Selected unicam format: 1280x720-pBAA
qt.qpa.xcb: QXcbConnection: XCB error: 148 (Unknown), sequence: 192, resource id: 0, major code: 140 (Unknown), minor code: 20
[17:34:45.087645084] [9383]  INFO Camera camera.cpp:1029 configuring streams: (0) 9248x6944-BGR888
[17:34:45.090125699] [9384]  INFO RPI raspberrypi.cpp:763 Sensor: /base/soc/i2c0mux/i2c@1/arducam_64mp@1a - Selected sensor format: 9152x6944-SBGGR10_1X10 - Selected unicam format: 9152x6944-pBAA
```
Also the log with the app_capture2.py with verbose_console=2 in Picamera2()
```
[17:37:57.636075975] [9447]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3730-67300b62
[17:37:57.699740219] [9448]  WARN CameraSensorProperties camera_sensor_properties.cpp:174 No static properties available for 'arducam_64mp'
[17:37:57.699789311] [9448]  WARN CameraSensorProperties camera_sensor_properties.cpp:176 Please consider updating the camera sensor properties database
[17:37:57.721693846] [9448]  WARN RPI raspberrypi.cpp:1274 Mismatch between Unicam and CamHelper for embedded data usage!
[17:37:57.722158081] [9448] ERROR DelayedControls delayed_controls.cpp:87 Delay request for control id 0x009a090a but control is not exposed by device /dev/v4l-subdev0
[17:37:57.722323153] [9448]  INFO RPI raspberrypi.cpp:1398 Registered camera /base/soc/i2c0mux/i2c@1/arducam_64mp@1a to Unicam device /dev/media0 and ISP device /dev/media3
2022-08-24T09:03:32.100Z | INFO     | Initialization successful.
2022-08-24T09:03:32.100Z | INFO     | Camera now open.
2022-08-24T09:03:32.100Z | DEBUG    | <libcamera._libcamera.CameraManager object at 0xaf8dc420>
2022-08-24T09:03:32.102Z | DEBUG    | Requesting configuration: {'use_case': 'preview', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'JPEG'>, 'buffer_count': 4, 'main': {'format': 'XBGR8888', 'size': (800, 600), 'stride': 3200, 'framesize': 1920000}, 'lores': None, 'raw': None, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Minimal: 3>, 'FrameDurationLimits': (100, 83333)}, 'display': 'main', 'encode': 'main'}
[17:37:57.725881623] [9447]  INFO Camera camera.cpp:1029 configuring streams: (0) 800x600-XBGR8888
[17:37:57.726276377] [9448]  INFO RPI raspberrypi.cpp:763 Sensor: /base/soc/i2c0mux/i2c@1/arducam_64mp@1a - Selected sensor format: 1280x720-SBGGR10_1X10 - Selected unicam format: 1280x720-pBAA
2022-08-24T09:03:32.104Z | INFO     | Configuration successful!
2022-08-24T09:03:32.105Z | DEBUG    | Final configuration: {'use_case': 'preview', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'JPEG'>, 'buffer_count': 4, 'main': {'format': 'XBGR8888', 'size': (800, 600), 'stride': 3200, 'framesize': 1920000}, 'lores': None, 'raw': None, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Minimal: 3>, 'FrameDurationLimits': (100, 83333)}, 'display': 'main', 'encode': 'main'}
2022-08-24T09:03:32.105Z | DEBUG    | Streams: {'main': <libcamera._libcamera.Stream object at 0xa7dd2560>, 'lores': None, 'raw': None}
2022-08-24T09:03:32.110Z | DEBUG    | Allocated 4 buffers for stream 0.
EGL Mesa Project 1.4
2022-08-24T09:03:33.605Z | INFO     | Camera started
Make buffer for request Request(3:C:0/1:0)
qt.qpa.xcb: QXcbConnection: XCB error: 148 (Unknown), sequence: 192, resource id: 0, major code: 140 (Unknown), minor code: 20
Make buffer for request Request(6:C:0/1:0)
Make buffer for request Request(8:C:0/1:0)
2022-08-24T09:03:40.437Z | DEBUG    | Execute function: functools.partial(<bound method Picamera2.switch_mode_ of <picamera2.picamera2.Picamera2 object at 0xb66d1358>>, {'use_case': 'still', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'JPEG'>, 'buffer_count': 1, 'main': {'format': 'BGR888', 'size': (9248, 6944)}, 'lores': None, 'raw': None, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.HighQuality: 2>, 'FrameDurationLimits': (100, 1000000000)}, 'display': None, 'encode': None})
2022-08-24T09:03:40.456Z | INFO     | Camera stopped
[... cut ...]
2. When you say the event loop isn't running, are you finding that the preview window stops updating?
Yes the main window stop updating. Setting a breakpoint in Picamera2.process_requests() reveals its not called

3. When the app fails, are there any messages printed to the console? Is there anything suspicious in dmesg?
dmesg yields no new messages

I have tested a still configuration with "main={"size": (4624, 3472)}" and main={"size": (800, 600)} to no success.

I will test some older version of Picamera2 in the meantime

davidplowman, 2022-09-13:
Hi. is there anything still to follow up on this issue? Please let us know if there is. Thanks!

davidplowman, 2022-10-17:
I think I'll close this one for now, but please open a new issue if anything else comes up. Thanks!

closed 2022-10-17 by davidplowman