raspberrypi/picamera2 #284: [BUG]After Updating Picamera2 will not instantiate

ktorimaru, 2022-08-24:
This seems very similar to:
https://github.com/raspberrypi/picamera2/issues/166

This is what is happening:
pi@raspberrypi:~ $ python3
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from picamera2 import Picamera2
>>> picam = Picamera2()
[21:08:49.458778567] [34135]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3700-f30ad033
[21:08:49.527748197] [34151]  INFO RPI raspberrypi.cpp:1368 Registered camera /base/soc/i2c0mux/i2c@1/imx290@1a to Unicam device /dev/media0 and ISP device /dev/media1
2022-08-24T21:29:24.838Z | ERROR    | Camera __init__ sequence did not complete.
Traceback (most recent call last):
  File "/home/pi/picamera2/picamera2/picamera2.py", line 77, in __init__
    self.open_camera()
  File "/home/pi/picamera2/picamera2/picamera2.py", line 178, in open_camera
    if self.initialize_camera():
  File "/home/pi/picamera2/picamera2/picamera2.py", line 162, in initialize_camera
    self.sensor_resolution = self.camera.properties["PixelArraySize"]
KeyError: 'PixelArraySize'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/picamera2/picamera2/picamera2.py", line 81, in __init__
    raise RuntimeError("Camera __init__ sequence did not complete.")
RuntimeError: Camera __init__ sequence did not complete.

Note: libcamera-hello is working. The version is v0.0.0+3700-f30ad033

Prior to updating everything was working great.

davidplowman, 2022-08-25:
Hi, the "PixelArraySize" error is normally a result of not updating libcamera, for example [here](https://github.com/raspberrypi/picamera2/issues/231). Could you try updating as described there and let us know what happens? Thanks!

ktorimaru, 2022-08-26:
Ran these:
```
sudo apt install -y python3-libcamera
pip3 install picamera2 --upgrade
```

They reported that everything is current.
Running `cam = Picamera2()` failed the same way.

Tim-Brown-NZ, 2022-08-26:
I've had this error in the past when I had inadvertently installed two versions/copies of libcamera. I think one was under /lib/python3/dist-packages/libcamera and one under /**usr**/lib/python3/dist-packages/libcamera. sudo apt install was finding the newer one, python the older one.  Disclaimer: Linux is not my first language.

davidplowman, 2022-08-26:
Yes, I think something like this is likely. I don't know if starting from a fresh image is an option for you, otherwise maybe try `find / -name libcamera.so |& grep -v denied` to see how many versions might be lurking on your system?

waisaacs, 2022-08-26:
This was one of the issues I hit when I was using the Arducam build of libcamera.  Just an indication of the version in use being out of sync with the latest build, one way or another, as mentioned.

Tim-Brown-NZ, 2022-08-27:
On my system the file is called _libcamera.cpython-39-arm-linux-gnueabihf.so and is in the /usr/lib/python3/dist-packages/libcamera/ folder.  At least I think that's the file, again Linux not my first language.
When I run:
sudo find / -name *libcamera*.so |& grep -v denied
I get that file and two other files that match the pattern but don't look interesting.

davidplowman, 2022-08-30:
Hi, thanks for the various responses.

Can you say what OS you are using and whether you're using Arducam supplied versions of libcamera?

waisaacs, 2022-08-31:
I have just finally upgraded picamera2 on my RPi4 system and am getting complaints when trying to instantiate about the Sycc color class not being defined.  This is a recent change in libcamera for which you updated picamera2 to keep in sync.  But the package installed version of libcamera on Raspberry Pi OS is the 2022 July 7 build 3700, which does not have these changes (and there are probably other incompatibilities, not sure if the PixelArraySize is one of them).

Trying to build libcamera now and figure out how to properly point to it.  But just know that the posted instructions for keeping libcamera and picamera2 up to date are not currently functional/compatible.

davidplowman, 2022-08-31:
Hi, yes I'm very sorry about this. We're just right in the throes of updating all the packages and getting ready to build a new set of Raspberry Pi OS images. On this occasion libcamera has changed (the JPEG colour space has been renamed to sYCC, which is the cause of all the trouble), and Picamera2 is updated but the python3-libcamera package isn't.

For now I would downgrade yourself back to Picamera 0.3.2 which should be fine. Very soon things though be back to normal, and in fact even better because there'll be a single apt package which will update everything.

davidplowman, 2022-09-01:
Hi again, I think things are all sorted out now and there are updated libcamera apt packages that will work with _Picamera2_ version 0.3.3 (indeed earlier versions won't work any more). Also, as I indicated, you can simply do the following
```
sudo apt install -y python3-picamera2
```
and that should fetch everything you need.

waisaacs, 2022-09-01:
I found that the above pulled in several dependencies but was not until I did a separate apt upgrade that the libcamera stuff refreshed (to an Aug 31 build).  In any case, it looks like everything is back to functional out of the box.  Curious if this solves the OP's issue.

davidplowman, 2022-09-02:
Thanks for the confirmation on that. There might be a way of forcing apt to update dependencies too when updating the main package, we can investigate that one!

davidplowman, 2022-09-15:
Just wondering if there's anything else to do on this issue, or if I can consider closing it in a few days time? Please post back if there are any updates. Thanks!

ktorimaru, 2022-09-15:
I ended up starting from scratch.

closed 2022-09-15 by ktorimaru