raspberrypi/picamera2 #285: [BUG] app_full.py: "Reset" button in the "Image Tuning" pane updates the sliders but not the camera

davidplowman, 2022-08-26:
**Describe the bug**
Noticed while running the app_full.py demo. It appears that pressing the "Reset" button in the "Image Tuning" pane updates the various sliders back to their default locations, but I don't think the default values get sent back to the camera system.

**To Reproduce**
Grab, for example, the "Sharpness" slider and pull it all the way to the right. The image now looks over-sharpened. Next press the "Reset" button. The slider goes back to 1.0 but the image doesn't change. If you now click "up" to change the sharpness to 1.10, then it does change.

**Expected behaviour**
The camera image should change as soon as you click "Reset".

**Hardware :**
 Pi 4B with imx477, though I don't think that's particularly relevant!

davidplowman, 2022-08-26:
@will-v-pi Could you maybe take a look? I expect it's pretty straightforward, no great urgency, though. Thanks!!

will-v-pi, 2022-08-26:
Yeah, the sliders weren't emitting the value changed signal when they were updated by the reset button, so the controls weren't applying. Should be fixed with #286

davidplowman, 2022-08-30:
Thanks for the fix!

closed 2022-08-30 by davidplowman