raspberrypi/picamera2 #287: CameraConfiguration' object is not callable

SaailN, 2022-08-27:
While running the python script with configuration object i m getting the following error. Seems like object is not callable.
review_config = picam2.preview_configuration()
TypeError: 'CameraConfiguration' object is not callable
[image]

Tim-Brown-NZ, 2022-08-27:
In the latest release all the xxx_configuration methods were renamed to to create_xxx_configuration.
So you need to call create_preview_configuration() as in the updated example here:
https://github.com/raspberrypi/picamera2/blob/main/examples/preview.py

SaailN, 2022-08-27:
Thanks @Tim-Brown-NZ

closed 2022-08-27 by SaailN