raspberrypi/picamera2 #291: Expected date to be out of Beta

pageauc, 2022-08-31:
When is the expected date for the official release with production install in place?

davidplowman, 2022-08-31:
Hi, I'm afraid I don't have a date yet, it will be influenced by feedback from the beta release among other things.

pageauc, 2022-08-31:
Can you narrow it down to possibly a monthly quarter?  Eg second quarter of
2023
It would make more sense to have a rough target for developers and not just
and open ended Beta.

davidplowman, 2022-09-01:
Hi again, Raspberry Pi doesn't generally give too many details about future plans but I agree that something would be helpful here, so I'll see what I can do. Perhaps once the beta release has been out there with folks for a little while we'll start to get some idea how things are going and I might be able to give some "guidance".

pageauc, 2022-09-01:
Thanks.  Appreciate it

aguaviva, 2022-09-15:
Also, do you have updates on this? Thanks for this great platform!

davidplowman, 2022-09-16:
Sorry, I don't really have anything as yet. Mostly I think I've been trouble-shooting installation issues with people so far, I'm keen to get beyond that and find what problems there are in real world usage.

closed 2023-11-05 by pageauc