raspberrypi/picamera2 #292: [OTHER] Why do versions >1.1.1 require python 3.9+?

CorvetteCole, 2022-08-31:
Hello, we were originally intending to piecemeal upgrade our camera stack first on Buster, then slowly move to Bullseye over time. Unfortunately this plan has been thwarted by the fact that recent version of picamera2 seem to requires Python 3.9+, which Buster does not have access to. As any good scientist knows, don't change two variables at once! We are trying to avoid combining problems from shifting camera stack and OS at the same time.

Please let me know if this can be changed, but if not we understand.

Thanks!

davidplowman, 2022-08-31:
We generally have such headaches with libcamera changes, different kernel versions, different kernel drivers and so on, that I'm probably a bit paranoid and want to limit everything to Bullseye. Which doesn't mean you can't make it work - I would try checking Picamera2 out from the repository, point your PYTHONPATH environment variable at it, and see what happens.

CorvetteCole, 2022-08-31:
I'll give it a try and update this issue with results

CorvetteCole, 2022-09-12:
@mikkelam I had issues with libcamera being out of date and the colorspaces not matching. I suspect picamera2 0.3.2 would've worked. We've since decided to fully switch to bullseye instead of trying to do this the "wrong" way.

CorvetteCole, 2022-09-13:
I'm closing this issue, since I do not intend to move further with it.

closed 2022-09-13 by CorvetteCole