raspberrypi/picamera2 #293: [BUG][HOW-TO] Preview.QTGL on RasPi OS 64bit

JamesRayn, 2022-08-31:
**Describe the bug**
I tried to run Picamera2 on an RasPi 3B+ with RasPi OS 64bit lite. I installed everything like described. After that I executed some examples. But I was not able to run a Qt_Gl preview, I was able to run a DRM preview in console, but the Gl powered preview crashed whatever I tried.

**To Reproduce**
Run preview.py out of the examples folder on RasPI OS 64bit lite (RasPI 3B+).

**Console Output, Screenshots**
```
pi@fotobox:~/picamera2-main/apps $ python3 app_capture2.py
Traceback (most recent call last):
  File "/home/pi/picamera2-main/apps/app_capture2.py", line 12, in <module>
    from picamera2.previews.qt import QGlPicamera2
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/previews/qt.py", line 1, in <module>
    from .q_gl_picamera2 import EglState, QGlPicamera2
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/previews/q_gl_picamera2.py", line 21, in <module>
    from OpenGL.GLES2.OES.EGL_image import *
  File "/home/pi/.local/lib/python3.9/site-packages/OpenGL/GLES2/__init__.py", line 2, in <module>
    from OpenGL.raw.GLES2._types import *
  File "/home/pi/.local/lib/python3.9/site-packages/OpenGL/raw/GLES2/_types.py", line 5, in <module>
    _error_checker = _ErrorChecker( _p, _p.GLES2.glGetError )
AttributeError: 'NoneType' object has no attribute 'glGetError'
```

**Hardware :**
RasPi 3B+ with RasPI OS 64bit lite and a PiCam 2

**Additional context**
Is it a bug or a known incompatibility?

davidplowman, 2022-08-31:
Not sure, to be honest. Just to understand exactly what you're doing, have you installed the whole of X-Windows and the desktop environment on your Pi 3 and now you're trying to run it from a terminal window?

(If you're not running X-Windows, then the QT and QTGL previews aren't meant to work - only the DRM one will.)

JamesRayn, 2022-09-01:
Oh excuse me. I installed openbox as X Window system. If I install exactly the same named packages at an 32bit environment it does work. And I tried to start the QTGL preview out of the terminal emulator window. Just in case of the DRM preview I started it from the raw console.

davidplowman, 2022-09-02:
In general, sorting out all the different ways and combinations in which folks might install stuff is probably a bit beyond what I can do! Having said that, if you can tell me the exact image you started from (that latest Bullseye Lite 64-bit, I assume) and the exact sequence of install commands then I would take a look.

Also, we've released an apt package for Picamera2 now, so the most official way to install it will be (after `sudo apt update` and `sudo apt upgrade`)
```
sudo apt install -y python3-picamera2
```
so it would be good to reproduce any problem using this method. This will fetch all the Picamera2 dependencies including the Qt/OpenGL ones. If you don't want those, add `--no-install-recommends` to that install command.

JamesRayn, 2022-09-06:
I tried it again.
I installed RasPi OS 64bit lite another time fresh on my RasPi 3B+.
I installed openbox again and installed the camera system with `sudo apt install -y python3-picamera2` this time.
But I get the same error again:

```
[0:15:20.896968462] [1151] [1;32m INFO [1;37mCamera [1;34mcamera_manager.cpp:293 [0mlibcamera v0.0.0+3866-0c55e522
[0:15:20.947169795] [1152] [1;33m WARN [1;37mRPI [1;34mraspberrypi.cpp:1258 [0mMismatch between Unicam and CamHelper for embedded data usage!
[0:15:20.949291873] [1152] [1;32m INFO [1;37mRPI [1;34mraspberrypi.cpp:1374 [0mRegistered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3/dist-packages/picamera2/previews/qt_previews.py", line 22, in thread_func
    self.qpicamera2 = self.make_picamera2_widget(picam2, width=self.width, height=self.height, transform=self.transform)
  File "/usr/lib/python3/dist-packages/picamera2/previews/qt_previews.py", line 74, in make_picamera2_widget
    from picamera2.previews.qt import QGlPicamera2
  File "/usr/lib/python3/dist-packages/picamera2/previews/qt.py", line 1, in <module>
    from .q_gl_picamera2 import EglState, QGlPicamera2
  File "/usr/lib/python3/dist-packages/picamera2/previews/q_gl_picamera2.py", line 21, in <module>
    from OpenGL.GLES2.OES.EGL_image import *
  File "/usr/lib/python3/dist-packages/OpenGL/GLES2/__init__.py", line 2, in <module>
    from OpenGL.raw.GLES2._types import *
  File "/usr/lib/python3/dist-packages/OpenGL/raw/GLES2/_types.py", line 5, in <module>
    _error_checker = _ErrorChecker( _p, _p.GLES2.glGetError )
AttributeError: 'NoneType' object has no attribute 'glGetError'
```

davidplowman, 2022-09-07:
Hi again, thanks for the extra info. As I said previously, if you could confirm exactly what commands you typed to install openbox (and anything else) then I could try it and be sure I have done the same thing as you. Could you also please confirm precisely any configuration changes you have made, for example, being a Pi 3, I assume you have had to do something about Glamor? Display stacks and OpenGL are not really anything I know about, so being as explicit as you can really helps me. Thanks!

davidplowman, 2022-09-07:
Thanks for those instructions. Using them I was able to see the problem easily enough. I used one of the just-released updated OS images so Picamera2 (without GUI dependencies) was already installed.

Anyway, I may have a fix for you, here's what I did:
```
sudo apt update
sudo apt upgrade # does almost nothing as it's a new image

sudo apt install openbox xorg lightdm unclutter

# and here's the magic fix:
sudo apt install libgles2

# installing python3-picamera2 would normally fetch these, but not on a Lite image:
sudo apt install python3-pyqt5 python3-opengl
```
And that seemed to work, maybe you could verify it. Don't ask me why, I don't even begin to understand.

Another thing I noticed is that I didn't need to enable Glamor. It seems that by not using a standard image we haven't gone through the setup/initialisation bits that would have disabled it. Generally the reaction here at Pi Towers to "do-it-yourself" installing of X-Windows and OpenGL involved lots of teeth-sucking and telling me that I was "on my own"...

JamesRayn, 2022-09-07:
Perfect, thank you for digging into this.
`sudo apt install libgles2` is indeed the magic fix and works for me to.

davidplowman, 2022-09-08:
Great, glad that's working. I think I'll probably close this one now, though I appreciate there is something weird/broken/puzzling somewhere down in OpenGL/GLES on 64-bit platforms. I'll mention it to the folks here and see if anyone understands anything, but I think it's rather beyond my pay grade to figure it out...!

closed 2022-09-08 by davidplowman