raspberrypi/picamera2 #294: [OTHER] libcap development headers required python-prctl dependency

jamesdolezal, 2022-08-31:
Running the installation commands in the README yields an error when attempting to install the `python-prctl` dependency:

```
Collecting python-prctl
  Using cached python-prctl-1.8.1.tar.gz (28 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      You need to install libcap development headers to build this module
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.
```

This can be solved by installing the libcap development headers. It might be worth including this in the README installation instructions, changing this:

`sudo apt install -y python3-libcamera python3-kms++`

to:

`sudo apt install -y python3-libcamera python3-kms++ libcap-dev`

davidplowman, 2022-09-01:
Hi, thanks for letting us know. Perhaps you're using OS Lite which would be why I hadn't noticed this? I'll get the README and the documentation updated!

davidplowman, 2022-09-01:
Or is it related to the fact that you were installing prctl with pip? I generally have more trouble with pip because stuff tends not to be available pre-compiled which causes all sorts of headaches. Does installing from apt work correctly for you? This is normally what I would recommend.

davidplowman, 2022-09-05:
Hi again, I think I'm going to close this because I believe the problem is actually down to installing through pip rather than apt. But if I've got that wrong, please feel free to reopen the issue.

It may also be helpful to know that from now on, the recommended way to install Picamera2 will be
```
sudo apt update
sudo apt upgrade
sudo apt install -y python3-picamera2
```

closed 2022-09-05 by davidplowman

davidplowman, 2022-09-06:
Well, I didn't realise there are some users who can't install through apt. Generally I find pip a real pain for certain packages so I really don't want to encourage most users down that route. But if this is really a thing that some users need to do then I'm in principal open to adding an "Application Note" to the documentation explaining how to do this, and we could add a link from the "regular" installation instructions which "ordinary folk" wouldn't be tempted to follow.

But having said that, thing are about to change. We're going to be issuing images that contain Picamera2 by default, so there'd be no need to install anything separately. Where does that leave this question? Does it mean you don't need to install anything extra so everything is solved? I'm not sure I know enough about this... what do you think?

flatsiedatsie, 2022-09-06:
> We're going to be issuing images that contain Picamera2 by default

Also on the Lite image? Doesn't it rely on some big dependencies? (another separate issue)

I've just added it by default to the [Candle disk image](https://www.candlesmarthome.com/download) too, so we're on the same track there. 

However, I wasn't too happy about doing that because it means that the various plugins can no longer bring their own (more up to date) version of Picamera2. This matters because the disk image isn't updated that frequently (e.g. every 6-12 months), so bringing their own version is a way for these plugins to have a never version of Picamera2, despite the underlying OS being updated less frequently.

This matters even more since Picamera2 is such a new project, likely to have significant changes at random intervals. Locking all plugins into an 'old' version of Picamera2 isn't ideal.

That being said, being pragmatic, at the moment there is only one plugin that uses Picamera2, so that's why in practise it's ok. But I can't speak for the future, or for other, bigger projects? It kind of depends on how stable PiCamera2 is too.

In general I guess I wish Picamera2 had (the option to have) less big dependencies, more akin to the 'old' picamera package. It would make all this less of an issue in the first place. Here's a l[ist of failed Github actions](https://github.com/createcandle/candlecam/actions) from my failed attempts to build the plugin in the 'normal' way for my project. In the end I had to give up and go the Apt route...

davidplowman, 2022-09-06:
The Lite version of the OS doesn't include Qt or OpenGL, so it's still quite small (and those features of Picamera2 won't work unless you fetch those dependencies explicitly).

I would also caution a bit about updating Picamera2 on the fly. libcamera doesn't have a stable API yet so it's very easy for libcamera and Picamera2 to get out of sync. In fact we've just been through another bout of this (bugs to do with "Jpeg" and "Sycc") and it's just pain all round. If you can stick to `sudo apt upgrade` then you'll always get packages that work together.

gamer9-create, 2024-10-25:
you are such a sigma chad!!!! thanks so much i was having this issue!!!!

wegunterjr, 2025-05-03:
This is old issue, but I have same questions.  If we are using pyenv and installing with apt get, does that cause issues with versions. I installed using apt, but when I try to import into my pyenv it says picamera2 not available...
`Python 3.9.22 (main, May  2 2025, 07:53:18) 
[GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import picamera2
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'picamera2'
>>> from picamera2 import Picamera2
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'picamera2'
`

davidplowman, 2025-05-06:
Normally Picamera2 should be installed on your system already. To use it in a _venv_, make sure you create your venv with the `--system-site-packages` flag. Have you done that?