raspberrypi/picamera2 #295: [BUG] 'libcamera._libcamera.ColorSpace' has no attribute 'Sycc'

jamesdolezal, 2022-08-31:
**Describe the bug**
After following the installation instructions, importing picamera2 raises the following error:

```
>>> import picamera2
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/venv/lib/python3.9/site-packages/picamera2/__init__.py", line 5, in <module>
    from .picamera2 import Picamera2, Preview
  File "/home/pi/venv/lib/python3.9/site-packages/picamera2/picamera2.py", line 45, in <module>
    class Picamera2:
  File "/home/pi/venv/lib/python3.9/site-packages/picamera2/picamera2.py", line 458, in Picamera2
    def create_preview_configuration(self, main={}, lores=None, raw=None, transform=libcamera.Transform(), colour_space=libcamera.ColorSpace.Sycc(), buffer_count=4, controls={}, display="main", encode="main"):
AttributeError: type object 'libcamera._libcamera.ColorSpace' has no attribute 'Sycc'
```
**To Reproduce**
``` python
import picamera2  # Raises the error
```
**Hardware :**
Raspberry Pi 4B, camera module v2

**Additional context**
I suspect this is related to a recent change in libcamera switching from "Jpeg" to "Sycc": https://patchwork.libcamera.org/patch/17185/

libcamera was installed with apt install `python3-libcamera`, which installed version `0~git20220705+f30ad033-1`.

This issue is resolved by importing libcamera first and setting `libcamera.ColorSpace.Sycc` equal to `.Jpeg`: 

``` python
import libcamera
libcamera.ColorSpace.Sycc = libcamera.ColorSpace.Jpeg
import picamera2  # No error raised
```

I expect this can be fixed by either checking the installed version of libcamera, or verifying that `Sycc` exists (using `Jpeg` if not)

davidplowman, 2022-09-01:
Hi, yes I'm very sorry about this. We're just in the throes of a whole load of updates which include the latest libcamera (which has change the Jpeg() colour space to Sycc() causing the trouble). We're building apt packages for _Picamera2_ for the first time which is why some _Picamera2_ stuff has got updated (incorrectly, probably) while the libcamera packages haven't caught up yet.

For now I would recommend downgrading back to 0.3.2 while I sort everything out. In a couple of days, hopefully, you'll be able to install the whole of _Picamera2_, and all the dependencies with `sudo apt install -y python3-picamera2`.

davidplowman, 2022-09-01:
Actually I think this is all fixed now. If you do
```
sudo apt update
sudo apt upgrade
```
you should get new versions of _libcamera_ which will make the 0.3.3 version of _Picamera2_ work (indeed previous versions won't). Also, you can now just do
```
sudo apt install python3-picamera2
```
and I think this will be the recommended installation procedure going forwards.

jamesdolezal, 2022-09-06:
Thanks for the clarification on this - I can confirm this solves the issue.

closed 2022-09-06 by jamesdolezal