raspberrypi/picamera2 #298: AttributeError: type object 'libcamera._libcamera.ColorSpace' has no attribute 'Jpeg'

mohamed-29, 2022-09-04:
AttributeError: type object 'libcamera._libcamera.ColorSpace' has no attribute 'Jpeg'

Bra1nsen, 2022-09-04:
[image]
 
example: raw.py

same error

smababou, 2022-09-05:
i am getting the same error also

```bash
pi@pi:~ $ python3
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import picamera2
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.9/dist-packages/picamera2/__init__.py", line 5, in <module>
    from .picamera2 import Picamera2, Preview
  File "/usr/local/lib/python3.9/dist-packages/picamera2/picamera2.py", line 42, in <module>
    class Picamera2:
  File "/usr/local/lib/python3.9/dist-packages/picamera2/picamera2.py", line 400, in Picamera2
    def create_preview_configuration(self, main={}, lores=None, raw=None, transform=libcamera.Transform(), colour_space=libcamera.ColorSpace.Jpeg(), buffer_count=4, controls={}, display="main", encode="main"):
AttributeError: type object 'libcamera._libcamera.ColorSpace' has no attribute 'Jpeg'
```

davidplowman, 2022-09-05:
Hi, yes I'm afraid the libcamera API has been updated and the latest version of Picamera2 matches it. Run
```
sudo apt update
sudo apt upgrade
```
to fix the problem.

smababou, 2022-09-05:
I have did that, and libcamera is upgraded. 
according to this issue: https://github.com/raspberrypi/picamera2/issues/295#issuecomment-1234291659
But still having the same error

davidplowman, 2022-09-05:
Ah, sorry, my mistake. If Picamera2 is looking for a symbol ColorSpace.Jpeg() then it's Picamera2 that is out-of-date, not libcamera. The latest version that matches the updated libcamera is 0.3.3. - what does `pip3 show picamera2` report?

Note that you can now also use `sudo apt install -y python3-picamera2`. This will probably be a better way to install Picamera2 in future because `sudo apt upgrade` will update everything so that the versions should stay compatible.

Bra1nsen, 2022-09-05:
```
pi@allskyLA:~ $ pip3 show picamera2
Name: picamera2
Version: 0.2.3
Summary: The libcamera-based Python interface to Raspberry Pi cameras, based on the original Picamera library
Home-page: https://github.com/RaspberryPi/picamera2
Author: Raspberry Pi & Raspberry Pi Foundation
Author-email: picamera2@raspberrypi.com
License: BSD 2-Clause License
Location: /home/pi/.local/lib/python3.9/site-packages
Requires: numpy, PiDNG, piexif, pillow, pyopengl, PyQt5, python-prctl, simplejpeg, v4l2-python3
Required-by: 
```

davidplowman, 2022-09-05:
Yes, for the latest version of libcamera you need to be on Picamera2 0.3.3. As I said, the following is probably recommended now:
```
sudo apt update
sudo apt upgrade
sudo apt install -y python3-picamera2
```

Bra1nsen, 2022-09-05:
error remains :/
```
pi@allskyLA:~ $ sudo apt install -y python3-picamera2
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-picamera2 is already the newest version (0.3.3-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@allskyLA:~ $ pip3 show picamera2
Name: picamera2
Version: 0.2.3
Summary: The libcamera-based Python interface to Raspberry Pi cameras, based on the original Picamera library
Home-page: https://github.com/RaspberryPi/picamera2
Author: Raspberry Pi & Raspberry Pi Foundation
Author-email: picamera2@raspberrypi.com
License: BSD 2-Clause License
Location: /home/pi/.local/lib/python3.9/site-packages
Requires: numpy, PiDNG, piexif, pillow, pyopengl, PyQt5, python-prctl, simplejpeg, v4l2-python3
Required-by: 
pi@allskyLA:~ $ 
```

smababou, 2022-09-05:
@Bra1nsen you must uninstall the pip3 package
```bash
pip3 uninstall picamera2
```
or
```bash
sudo pip3 uninstall picamera2
```
then reinstall using:
```bash
sudo apt install -y python3-picamera2
```
pip show output
```bash
pi@pi:~ $ pip3 show picamera2
Name: picamera2
Version: 0.3.3
Summary: The libcamera-based Python interface to Raspberry Pi cameras, based on the original Picamera library
Home-page: https://github.com/RaspberryPi/picamera2
Author: Raspberry Pi & Raspberry Pi Foundation
Author-email: picamera2@raspberrypi.com
License: BSD 2-Clause License
Location: /usr/lib/python3/dist-packages
Requires: 
Required-by: 
```

Bra1nsen, 2022-09-05:
nice thanks =)!

closed 2022-09-06 by mohamed-29