raspberrypi/picamera2 #299: [BUG] H264Encoder no longer accepting iperiod parameter

sandyol55, 2022-09-04:
After updating/upgrading a fresh 32-bit install on a 2G Pi4 to the latest picamera2 0.3.3 with 
 libcamera build: v0.0.0+3866-0c55e522
the H264Encoder is failing with the following message if an iperiod parameter is passed

`Traceback (most recent call last):
  File "/home/sandy/picamera2/examples/capture_video.py", line 11, in <module>
    encoder = H264Encoder(10000000, repeat=True, iperiod=45)
  File "/usr/lib/python3/dist-packages/picamera2/encoders/h264_encoder.py", line 26, in __init__
    self._controls += [(V4L2_CID_MPEG_VIDEO_H264_I_PERIOD, iperiod)]
NameError: name 'V4L2_CID_MPEG_VIDEO_H264_I_PERIOD' is not defined`

Able to generate the error with the following snippet.....

`#!/usr/bin/python3
import time

from picamera2 import Picamera2
from picamera2.encoders import H264Encoder

picam2 = Picamera2()
video_config = picam2.create_video_configuration()
picam2.configure(video_config)

encoder = H264Encoder(10000000, repeat=True, iperiod=45)

picam2.start_recording(encoder, 'test.h264')
time.sleep(10)
picam2.stop_recording()`

davidplowman, 2022-09-05:
Not sure what's going on there. What happens if you enter just
```
import v4l2
v4l2.V4L2_CID_MPEG_VIDEO_H264_I_PERIOD
```
Also, what does `pip3 show v4l2` report? Thanks!

@chrisruk Any ideas on this??

sandyol55, 2022-09-05:
Hi, and thanks for the quick response.

import v4l2
v4l2.V4L2_CID_MPEG_VIDEO_H264_I_PERIOD

gives this response

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'v4l2' has no attribute 'V4L2_CID_MPEG_VIDEO_H264_I_PERIOD'_CID_MPEG_VIDEO_H264_I_PERIOD

and .....tellingly ......
pip3 show v4l2
returns
WARNING: Package(s) not found: v4l2

Will recheck I've followed all the proper installation steps and report back.
Thanks again
Sandy

chrisruk, 2022-09-05:
You need to do the following to get the version number -

```
pip3 show v4l2-python3
```

I'm just testing everything works correctly with the .deb package at the moment too

sandyol55, 2022-09-05:
Have just flashed a fresh image and update/upgraded and then installed picamera2 with the 'old' 

sudo apt install -y python3-libcamera python3-kms++
sudo apt install -y python3-pyqt5 python3-prctl libatlas-base-dev ffmpeg python3-pip
pip3 install numpy --upgrade
pip3 install picamera2[gui]

instructions.

pip3 show v4l2-python3

returns

Name: v4l2-python3
Version: 0.3.1
Summary: Python bindings for the v4l2 userspace api.
Home-page: https://pypi.org/project/v4l2-python3/
Author: None
Author-email: None
License: GPLv2
Location: /home/sandy/.local/lib/python3.9/site-packages
Requires: 
Required-by: picamera2

and the error messages are gone when H264Encoder is called with iperiod parameter.

chrisruk, 2022-09-05:
Thanks, yeah the issue seems to be with the .deb package being an older version than the pip package

davidplowman, 2022-09-05:
Thanks everyone, I'll take a look at that apt package and see what I need to do.

davidplowman, 2022-09-05:
Hi again, we've posted an updated version of the python3-v4l2 package. To fetch it use:
```
sudo apt update
sudo apt install -y python3-v4l2
```
Sorry for the hassle and thanks for spotting it!

davidplowman, 2022-09-05:
Actually I can probably marked this as fixed now - please feel free to reopen if it's still broken. Thanks!

closed 2022-09-05 by davidplowman

sandyol55, 2022-09-05:
Thanks again. Back to tinkering with a simple Wildlife/Security Cam application, heavily based on examples plus the discourse in #255.