raspberrypi/picamera2 #300: Camera __init__ sequence did not complete.

robokrishan, 2022-09-05:
I have been working with a camera module which worked perfectly with picamera2 as of a month ago. My big mistake was performing an apt update. Now I am unable to use the picamera2 library for interfacing my camera module. I am working with the OV5647 camera. I was initially on picamera2 0.2.3 (now 0.3.3). I tried reverting back to the older version but still not working. I think there are a bunch of other libraries that got updated as well. My working script is now completely broken. The script fails to initalize the camera. I get the following output on my terminal when I try to run the script:

`[0:28:02.822312015] [1895]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3866-0c55e522
2022-09-05T13:23:01.484Z | ERROR    | Camera(s) not found (Do not forget to disable legacy camera with raspi-config).
2022-09-05T13:23:01.485Z | ERROR    | Camera __init__ sequence did not complete.
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/picamera2.py", line 123, in __init__
    self._open_camera()
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/picamera2.py", line 305, in _open_camera
    if self._initialize_camera():
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/picamera2.py", line 271, in _initialize_camera
    raise RuntimeError("Camera(s) not found (Do not forget to disable legacy camera with raspi-config).")
RuntimeError: Camera(s) not found (Do not forget to disable legacy camera with raspi-config).

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/RaspberryPi/Motorized_Focus_Camera/python/FocuserExample.py", line 187, in <module>
    main()
  File "/home/pi/RaspberryPi/Motorized_Focus_Camera/python/FocuserExample.py", line 175, in main
    camera = Camera()
  File "/home/pi/RaspberryPi/Motorized_Focus_Camera/python/RpiCamera.py", line 20, in __init__
    self.open_camera(width, height)
  File "/home/pi/RaspberryPi/Motorized_Focus_Camera/python/RpiCamera.py", line 23, in open_camera
    self.cam = Picamera2()
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/picamera2.py", line 130, in __init__
    raise RuntimeError("Camera __init__ sequence did not complete.")
RuntimeError: Camera __init__ sequence did not complete.`

I have ensured that the Legacy Camera stack is disabled in raspi-config. I tried initializing the camera on a fresh script but the moment the interpreter runs Picamera2(), this error is thrown. I am at a loss here. Can someone please tell me which libraries were updated that broke the install? I want to revert all the dependencies to the previous version and see if I can salvage my script.

Thanks in advance.

tvoverbeek, 2022-09-05:
Your traceback shows the picamera2 in your .local directory.
The latest update (using the apt package) installs in the system python directory.
Get rid of your .local installation (Probably with pip3 uninstall picamera2) or brute force by deleting all picamera2 related files in /home/pi/.local/lib/python3.9/site-packages/
This willa avoid conflicts caused by mixing the 2 versions.

robokrishan, 2022-09-05:
Thanks for the response! I uninstalled the picamera2 installation using pip, then installed the apt package (as explained on the github page for picamera2). I still get the same error as shown in the post. Here is the output:

```
>>> from picamera2 import Picamera2, Preview
>>> cam = Picamera2()
[0:17:12.952950772] [2093]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3866-0c55e522
2022-09-05T13:51:21.581Z | ERROR    | Camera(s) not found (Do not forget to disable legacy camera with raspi-config).
2022-09-05T13:51:21.582Z | ERROR    | Camera __init__ sequence did not complete.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 123, in __init__
    self._open_camera()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 305, in _open_camera
    if self._initialize_camera():
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 271, in _initialize_camera
    raise RuntimeError("Camera(s) not found (Do not forget to disable legacy camera with raspi-config).")
RuntimeError: Camera(s) not found (Do not forget to disable legacy camera with raspi-config).

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 130, in __init__
    raise RuntimeError("Camera __init__ sequence did not complete.")
RuntimeError: Camera __init__ sequence did not complete.
```

tvoverbeek, 2022-09-05:
Which version of Raspberry Pi OS? You need the latest Bullseye version for the automatic camera recognition to work.

robokrishan, 2022-09-05:
@tvoverbeek Here is my output from uname -a:

```
pi@arducam-32:/usr/lib/python3/dist-packages/picamera2 $ uname -a
Linux arducam-32 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l GNU/Linux
```

Tim-Brown-NZ, 2022-09-06:
I've had problems with autodetect with two of the cameras models I use.  The solution was to edit the /boot/config.txt file and set _camera_auto_detect=0_ and add _dtoverlay=ov5647_ at the end, basically force the pi to use this camera.  I haven't used the ov5647 camera but I have had problems with Arducam's latest version of imx477 cameras not being recognized.  Incidentally the newest imx477 boards don't work with the original picamera software either.  So our older boards work with our old code and the new ones don't.  The available dtoverlay settings files are in /usr/share/libcamera/ipa/raspberrypi.

robokrishan, 2022-09-06:
@Tim-Brown-NZ Thanks for the insight. I will try to force the overlay. I have used this approach in the past but after the previous update to picamera2, it wasn't necessary. I will use dtoverlay=ov5647 and see if that makes things work. Will update the thread with the results.

robokrishan, 2022-09-06:
Hi @davidplowman . Thanks for looking into it. `libcamera-hello` works. Preview window opens for 5 seconds, then closes. All connections are good. Yes I have spare modules and I have tried 3 different ones. 

Here is my output from `dmesg | grep ov5`:
```

pi@arducam-revert:~/RaspberryPi/Motorized_Focus_Camera/python $ dmesg | grep ov5
[   10.229959] ov5647 10-0036: Consider updating driver ov5647 to match on endpoints
```

Here is my output from `cat /etc/os-release` :

```
pi@arducam-revert:~/RaspberryPi/Motorized_Focus_Camera/python $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```

davidplowman, 2022-09-06:
Ah, are you using an Arducam camera for which you have downloaded an Arducam version of libcamera and libcamera-apps? I think the problem is that their software isn't up to date with the latest version of libcamera and therefore just won't work with the official python bindings (python3-libcamera). It's also proprietary, so far as I know, which means there's nothing I can do to help you. You would probably have to go to Arducam for support. Sorry not to be more help.

robokrishan, 2022-09-06:
Hi @davidplowman . Thanks for your quick response! Yes the camera I am using is from Arducam. I am NOT using Arducam's custom `libcamera` or `libcamera-apps`. Like I mentioned, I am using `python3-libcamera`. Here is the error that I am getting now:

`AttributeError: type object 'libcamera._libcamera.ColorSpace' has no attribute 'Jpeg'`

For context, here is my requirements.txt:

```
python3-kms++=0~git20211110~105525+2236a8c-1
python3-libcamera=0~git20220705+f30ad033-1                # Unfortunately no longer available, new update, old version unlinked
python3-pyqt5=5.15.2+dfsg-3
python3-prctl=1.7-2+b2
libatlas-base-dev=3.10.3-10+rpi1
ffmpeg=7:4.3.4-0+deb11u1+rpt2
numpy==1.23.1
picamera2==0.2.3
opencv-python==4.6.0.66
```

davidplowman, 2022-09-06:
Hi again, the "Jpeg" error is because your Picamera2 is out of date - the latest version is 0.3.3. The best way to install it now is `sudo apt install python3-picamera2` but I think if you've installed via pip previously then you might need to uninstall that first.

robokrishan, 2022-09-06:
OMG You are a lifesaver! It worked! I uninstalled from pip, then installed as you suggested and voila! Thanks so much @davidplowman , appreciate your time and effort.

davidplowman, 2022-09-06:
Glad to hear it! I'll close this now - please feel free to open another issue if anything else comes up.

closed 2022-09-06 by davidplowman

Cande8a, 2022-10-31:
I have the same problem, with the difference I have installed the picamera2 on a docker container, but I tried opening python from the terminal INSIDE docker and the picamera2 is imported and everything is ok, but when I run my whole project, where my python file for the camera is, I get this error. I made everything this issue explain, but is still the same. I didn't installed picamera2 with pip, actually on my container I have it like it is recommended on github page.
The only difference I can see is that I don't have any output with dmesg | grep ov5
I've been working through this issue many days and it's for my job, that'S why I'm asking for help. Has anyone any idea about what to try? Thanks!!

sunmouren, 2023-05-26:
This is probably due to permissions, try adding the user name to the vidoe and render user groups, e.g. 
`
sudo gpasswd --add pi video
sudo gpasswd -add pi render

`
then reboot the system.