raspberrypi/picamera2 #301: [HOW-TO] framerate max limit in raspberrypi 4b with hq camera? or how to change the camera mode?

Jiangshan00001, 2022-09-06:
framerate max limit in raspberrypi 4b with hq camera

now i use hq camera with rapberrypi 4b. it can reach 40fps. 
how can i reach 120fps with low resolution?

https://forums.raspberrypi.com/viewtopic.php?t=314581
https://www.raspberrypi.com/documentation/accessories/camera.html#libcamera-vid

is seems that what ever resolution is set, the mode is always:"Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC"

so, the problem maybe: how to change the camera mode?

1 : imx477 [4056x3040] (/base/soc/i2c0mux/i2c@1/imx477@1a)
    Modes: 'SRGGB10_CSI2P' : 1332x990 [120.05 fps - (696, 528)/2664x1980 crop]
           'SRGGB12_CSI2P' : 2028x1080 [50.03 fps - (0, 440)/4056x2160 crop]
                             2028x1520 [40.01 fps - (0, 0)/4056x3040 crop]
                             4056x3040 [10.00 fps - (0, 0)/4056x3040 crop]

davidplowman, 2022-09-06:
Hi, you can force the choice of a particular camera mode by requesting a raw stream, like this:
```
config = picam2.create_preview_configuration(raw={"size": (1332, 990), "format": "SRGGB10_CSI2P"})
```
There is some further discussion around this topic in section 4.2.2 of the [manual](https://datasheets.raspberrypi.com/camera/picamera2-manual-draft.pdf) which may help further.

Jiangshan00001, 2022-09-09:
@davidplowman thanks, it works!

closed 2022-09-09 by Jiangshan00001