raspberrypi/picamera2 #302: [HOW-TO] Spot metering mode

tennisparty, 2022-09-07:
Is there a way of metering the exposure on a single point in the image rather than an average of the entire image?

I can't find any relevant information in the docs apart from how to update the exposure and analog gain,

Thankyou

davidplowman, 2022-09-07:
There is a spot metering mode which you can use like this
```
...
from libcamera import controls
picam2.set_controls({"AeMeteringMode": controls.AeMeteringModeEnum.Spot})
```
which will metre on a spot in the centre of the image. You can't change the location or size of the spot, though it looks like you can mess around with the weights in the tuning file which will give you some scope to tweak the behaviour.

There is something about this in Appendix C of the latest version of the Picamera2 manual, though unfortunately we haven't quite managed to publish the update yet!

tennisparty, 2022-09-07:
Thanks, that's really useful, spot on as always

davidplowman, 2022-09-08:
Great! I'll probably close this one now but as ever, please post back if anything else comes up.