raspberrypi/picamera2 #303: Error Attribute Libcamera

Aaronngp, 2022-09-07:
Can't run example code (tuning_file.py), because libcamera give an error

I try to use the camera to take screenshots but I get an attribute error. 

This didn't happen before, after several days without using the camera now this error has come out. 

Console output: AttributeError: type object 'libcamera._libcamera.ColorSpace' has no attribute 'Jpeg'

I'm using RPI 4 and camera pi noir v2.1. I try this code in Thonny.

davidplowman, 2022-09-08:
Hi, the libcamera API has changed so you need to update Picamera2 to match. Updating to the latest version (currently 0.3.3) should fix this.

Moving forward, we're telling people that they can install Picamera2 using `sudo apt install -y python3-picamera2`. You will need to uninstall the pip version first, however. Then you can update your software with `sudo apt update` and `sudo apt upgrade` and you'll always get compatible versions of libcamera and Picamera2.

faeq123, 2022-09-08:
whats the command to uninstall the pip version?

davidplowman, 2022-09-08:
I think it's `pip3 uninstall picamera2`

faeq123, 2022-09-08:
thank you soo much it worked

davidplowman, 2022-09-08:
Great! I think I'll close this one now - please feel free to post a new issue if anything else comes up. Thanks!

closed 2022-09-08 by davidplowman

Alpha-Badger, 2022-10-06:
@davidplowman 

> Hi, the libcamera API has changed so you need to update Picamera2 to match. Updating to the latest version (currently 0.3.3) should fix this.
>
> Moving forward, we're telling people that they can [...]

Hello, 

`sudo apt install -y python3-picamera2` gives me the following message:

**rhinosd@raspberrypi:~ $ sudo apt install -y python3-picamera2
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-picamera2 is already the newest version (0.3.3-1).
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
python3-picamera2 : Depends: python3-libcamera but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).**

Obviously the next thing I should do is `install python3-libcamera`. However, the solution to another issue I have with my Arducam conflicts with doing this. Here is a link that outlines what I am experiencing: https://forum.arducam.com/t/how-is-it-possible-to-get-autofocus-working-along-with-picamera2-for-imx519/3460/6

"If you have python3-libcamera installed, this will cause you to use the official libcamera, which has no focus control"

They explicitly mention that to not use install python3-libcamera. How can I use PiCamera2, Arducam, and implement their functionalities via Python? Perhaps it's out of your purview as it may be a question for Arducam. However, I feel as though more synergy between PiCamera2 and Arducam is in order. Could that be something that is addressed with them?

davidplowman, 2022-10-06:
Hi, yes I think it's a question for Arducam, really. Probably the best solution would be for them to implement autofocus properly as part of the libcamera framework, but obviously there would be more work involved in this.