raspberrypi/picamera2 #307: [BUG] Library throwing DEBUG log msgs, even when set for INFO Level.

ajonnet, 2022-09-09:
Debug level logs are being thrown by the library, irrespective of what value is supplied for "verbose_console" param for constructor Picamera2(). It happens when log handlers are defined prior to calling of constructor.

Due to above behavior, Logs are filled up with debug statements being generated by the library. 

**To Reproduce**
import logging
from picamera2 import Picamera2

# Initialize Logger
logger = logging.getLogger("")
formatStr = '[%(asctime)s] [%(threadName)s] [%(levelname)s]: %(message)s'
console = logging.StreamHandler()
console.setFormatter(logging.Formatter(formatStr))
logger.addHandler(console)
logger.setLevel(logging.INFO)

# Initialize camera
picam=Picamera2(verbose_console=1)

**Expected behaviour**
As verbose_console is set to 1, below debug statement should not have generated
[2022-09-09 11:34:32,216] [MainThread] [DEBUG]: <libcamera._libcamera.CameraManager object at 0x7fae3a4bb0>

**Hardware :**
Pi 3B
OS: BullsEye

For now, to avoid above issue i have to patch below snippet (to be called after initialization of Picamera2)
logger = logging.getLogger('picamera2')
logger.setLevel(logging.INFO)

davidplowman, 2022-09-13:
Hi, thanks for raising this. Unfortunately I don't know much about the Python logging module, but here's what I found out.

Firstly, it's clear the the debug message is being printed by the `logger` that gets created in the test script, not by the Picamera2's logger (this is clear from the format of the message).

So it seems to me like the Picamera2 logger is correctly not outputting anything but "propagating" the message to the parent, which is `logger` in the script. (I think `logger` is actually the "root" logger.)

Anyway, could it be that this "propagation" by-passes filters on the logger, but only heeds filters on the handlers? If you change `logger.setLevel(logging.INFO)` to `consode.setLevel(logging.INFO)` then the message no longer appears. In fact, I found [this flow diagram](https://docs.python.org/2/howto/logging.html#logging-flow) which, if I'm interpreting it correctly, demonstrates exactly this behaviour.

Does this sound correct? Like I said, I hesitate a little just because I'm not terribly familiar with the behaviour of this module. But in any case I can't really see that Picamera2 could do anything to provoke this behaviour, it does seem intrinsic to the logging module.

davidplowman, 2022-09-16:
I've been experimenting a bit with this simplified test case:
```
import logging

class Dummy:
    def __init__(self, level=logging.INFO):
        self.logger = logging.getLogger("Dummy")
        logger.setLevel(logging.DEBUG)
        console = logging.StreamHandler()
        console.setLevel(level)
        console.setFormatter(logging.Formatter('[Dummy]: %(message)s'))
        logger.addHandler(console)

    def do_something(self):
        self.logger.debug("I am doing something")

logger = logging.getLogger()
logger.setLevel(logging.INFO)
console = logging.StreamHandler()
console.setFormatter(logging.Formatter('[Script]: %(message)s'))
logger.addHandler(console)
logger.setLevel(logging.INFO)
logger.debug("This should not appear")

dummy = Dummy(level=logging.INFO)
dummy.do_something()
```
which exhibits the same problem - it outputs the `I am doing something` message which is perhaps not expected.

If I replace `logger.setLevel(logging.DEBUG)` in the `__init__` method by `logger.setLevel(level)` then the problem goes away. But then, if I also change `dummy = Dummy(level=logging.INFO)` to `dummy = Dummy(level=logging.DEBUG)` then the `I am doing something` message gets output by _both_ loggers!

So it really does seem to be down to the rules of propagation and filters on loggers. If anyone can say what the most correct implementation of the `Dummy` class should be, then I'll copy that in Picamera2. Thanks!

tvoverbeek, 2022-09-16:
@davidplowman I think you forgot something in the  DUMMY __init__ function:
```
    def __init__(self, level=logging.INFO):
        self.logger = logging.getLogger("Dummy")  <---
        self.logger.setLevel(logging.DEBUG)       <---
        console = logging.StreamHandler()
        console.setLevel(level)
        console.setFormatter(loggin
        self.logger.addHandler(console)           <---
```

Without self you define properties of the global logger.
I have been bitten myself a few times by forgetting self for a class object ;-)
```

davidplowman, 2022-09-16:
@tvoverbeek Ah, thanks for that, have edited the original in case anyone wants to try it! Though weirdly none of the behaviour seems to change. The rules here seem a bit twisty...  :(

tvoverbeek, 2022-09-17:
OK, I believe I now understand how it works.
In the original post from @ajonnet there are 2 loggers defined.
The root logger in the main program and the picamera2 logger.
They are arranged hierarchically: any message from the picamera2 logger is also passed to the root logger.
The picamera2 logger has a setting logging.DEBUG.
The picamera2 messages are filtered by the streamhandler(s).
The picamera2 streamhandler blocks DEBUG mesages (console_verbosity=1)
But the root streamhandler has no level defined so it passes everything.
You see it in @ajonnet's output all INFO messages occur twice: first from the picamera2 logger followed by the root logger.
In @ajonnet's code (and in @davidplowman's example) replace the `logger.setLevel(logging.INFO)` just before
the `picam=Picamera2(verbose_console=1)` line with `console.setLevel(logging.INFO`. Similar in @davidplowman's example.
Then only the INFO messages appear twice and no DEBUG message.

However I think the logging concept should be changed.
From the 'Advanced Logging Tutorial', the section on "Configuring logging for a library"(https://docs.python.org/3.9/howto/logging.html#configuring-logging-for-a-library):
```
It is strongly advised that you do not add any handlers other than [NullHandler]
(https://docs.python.org/3.9/library/logging.handlers.html#logging.NullHandler) to your library’s loggers.
This is because the configuration of handlers is the prerogative of the application developer who uses your
library. The application developer knows their target audience and what handlers are most appropriate for
their application: if you add handlers ‘under the hood’, you might well interfere with their ability to
carry out unit tests and deliver logs which suit their requirements.
```
The picamera2 logger should only set the logging level and not add any handlers.
console-verbosity becomes logging-verbosity.
Without a NullHandler even when the user does not use a logger, warning and error messages will still be written
to stderr. I suppose this is the behavior we want.
This avoids the double logging messages as in @ajonnet's example.
What do you think?

ajonnet, 2022-09-20:
> OK, I believe I now understand how it works. In the original post from @ajonnet there are 2 loggers defined. The root logger in the main program and the picamera2 logger. They are arranged [...]

"configuration of handlers is the prerogative of the application developer who uses your library."
-This is the principle that should hold.
As part of my application code, i am already defining the required handlers and logging level.  But when initialize_logger() from picamera2_logger.py is called, it overrides the behavior i expect by defining log level to DEBUG for picamera2. Logs from picamera2, should have merged as per logging configuration defined in application code.

"The picamera2 logger should only set the logging level and not add any handlers."
-Here not even logging level should be defined by the library. It should be possible for application code to control what level of logs are to be generated.

I am naive with logs propagation principle. But expect it to be such that, logs from library principle should be controllable via application code.

davidplowman, 2022-09-27:
I've taken a bit more advice, and the feeling seemed to be that it would be better not to set the logger's level at all. Also, the `Picamera2.logger()` function is probably unnecessary as folks who understand the logging module would use `getLogger("picamera2")`, and then configure everything for themselves. So I'll prepare a PR along those lines in due course and then link it to this bug report.

closed 2022-10-18 by davidplowman