raspberrypi/picamera2 #309: [HOW-TO] Disable all log output (including from libcamera)

CorvetteCole, 2022-09-09:
**Describe what it is that you want to accomplish**
We are using picamera2 in a curses based application. Obviously, printing to stdout while this is running is not desirable, since it messes up the TUI. We need to disable log output entirely or redirect it elsewhere. 

This includes libcamera log messages, which I'm not sure how to disable given that the python bindings are pretty poorly documented (or I just can't find the documentation).

I've been digging around in the libcamera source code, and I've found the following promising code `libcamera::logSetTarget(libcamera::LoggingTargetNone);`, but that is C++ which obviously isn't applicable to the python bindings. 

**Describe alternatives you've considered**
I've thought about running the picamera stuff in a separate process instead of thread, but that has numerous negative implications and massively complicates the camera lifecycle management for our application.

I've also considered briefly pausing the UI while capture is on-going, but this is obviously not a good solution and was not needed for picamera2.

**Additional context**
If libcamera's python bindings don't provide this and this would be a more applicable issue on their git, please let me know. Desperately trying to figure this out or implement it ourselves!

davidplowman, 2022-09-13:
Hi, the first thing I was going to suggest was to set `LIBCAMERA_LOG_LEVELS`! Can you say in what way that solution doesn't work for you?

My other thought was that I believed (correct me if I'm wrong) that this debug gets send to `stderr` rather than `stdout`. Does redirecting `stderr` help at all?

If you still need something else, it might be worth dropping a message on their mailing list, folks there are normally very response. You can join [here](https://lists.libcamera.org/listinfo/libcamera-devel).

CorvetteCole, 2022-09-13:
it's non-optimal because ideally I'd like to get that logging output sometimes, and other than setting it system wide we have to remember to call the program with that environment variable. Ideally I'd love to log it with python's logger library to a file but I think there are technical reasons that would make that impossible (libcamera being a different language, namely).

I'll see if I can redirect stderr or something. Maybe it would even be possible to capture stderr and redirect to python's logger without breaking curses. For now, the environment variable works well.

CorvetteCole, 2022-09-13:
Our use case is somewhat unique since we have to access remote devices periodically that are extremely remote and isolated (in space). Just for context, since it makes it more clear why we are trying to prioritize compatibility with a 'common' base to reduce room for human error.

CorvetteCole, 2022-10-18:
I think it probably isn't worth the time to implement that kind of thing right now, but it would be very nice to document some info on this. Is that manual PDF in git somewhere? I could contribute some if so. 

Ideally it would be nice to have python bindings for that kind of thing but it's somewhat of a niche use case and I don't think it would be important to most users.

davidplowman, 2022-10-18:
I've just pushed a minor update (version 0.3.4) to apt, but haven't published the updated manual yet. That now has a short section on using logging, but doesn't mention libcamera logging. If you happened to have something on that I could add it, otherwise I don't mind putting something together myself - I don't think it's more than a few sentences and a list of the logging levels.

closed 2022-10-18 by CorvetteCole