raspberrypi/picamera2 #310: [BUG]

realizator, 2022-09-09:
**Describe the bug**
When using [preview.py](https://github.com/raspberrypi/picamera2/blob/main/examples/preview.py) example on the two-camera system, it works fine with the first cam (#0), but shows glitched image with the cam #1.

**To Reproduce**
OS: Bullseye (5.15.61-v7l+) - "RPiOS with desktop" 32bit version (2022-09-06-raspios-bullseye-armhf.img.xz + autoupdate)
Python: 3.9.2 (default, Mar 12 2021, 04:06:34) 
Picamera2:  0.3.3
[preview.py](https://github.com/raspberrypi/picamera2/blob/main/examples/preview.py)
Step 1: Change "[Picamera2()](https://github.com/raspberrypi/picamera2/blob/7094d7a94354cf860a58454e2d35e31c5f2e055f/examples/preview.py#L10)" to "Picamera2(0)" and run the script
Step 2: stop the script
Step 3: Change "[Picamera2(0)](https://github.com/raspberrypi/picamera2/blob/7094d7a94354cf860a58454e2d35e31c5f2e055f/examples/preview.py#L10)" to "Picamera2(1)" and run the script
Step 4: stop the script 

**Expected behaviour**
Running the script with Picamera2(0) and Picamera2(1) options should show preview image from the first and from the second cameras respectively.

**Console Output, Screenshots**
57 seconds [video with the experiment](https://youtu.be/rhW-0ocEjYA) 

**Hardware :**
I've used CM4-based board (StereoPi V2) and two OV5647 sensors.
boot/config.txt has been updated by adding two rows to enable cameras:

dtoverlay=ov5647 
dtoverlay=ov5647,cam0

**Additional context**

From the first view the glitch looks like a bad camera ribbon connection. That's why after the test I ran the libcamera app to show both cameras are working correctly.

davidplowman, 2022-09-13:
Hi, thanks for reporting this, I'll try and get hold of a CM4 and see if I can reproduce it too.

In the meantime, a couple of things to try:

* You ran the 2 instances of libcamera-hello simultaneously. Do they both still work if you run them one after the other (as you did Picamera2)?
* If you change the Picamera2 script to use `create_video_configuration` instead of `create_preview_configuration` is the behaviour the same? (This would cause the camera to be fixed at 30fps and would rule out any effect of higher framerates.)
* Do you by chance have a pair of v2 cameras to try? That's what I have and so that's what I'll test. (I'm a bit short of ov5647s but will see if I can find another.)

Thanks!

realizator, 2022-09-14:
Hi @davidplowman,

I've tested it with V2 (IMX219) and HQ (IMX477). 
For both IMX219 and IMX477 the script preview.py works fine!
So we have the OV5647 exclusion only.

davidplowman, 2022-09-15:
Thanks for the update. I'm still struggling to get all the exact same hardware, but your results are suggesting fairly strongly that the issue is related to ov5647 sensors running at higher framerates. If you could bear to do some more experiments, here are a couple more things to try:

1. If you could add `--framerate 60` to your libcamera-hello commands, that would provide extra confirmation that it's not specific to Picamera2, it's just dependent on the camera framerate.
2. Have you tried swapping the cameras over? That would confirm whether it's related to the camera connector on the IO board or whether it's something to do with the actual sensor (or indeed cable, it would clearly be helpful to check that too!).

Thanks!

davidplowman, 2022-09-20:
Hi again, I've finally got hold of a CM4 and a pair of ov5647s. Thus far I haven't managed to reproduce the problem, they seem to be working as expected.

Is there any update on what happens if you swap the sensors/cables over, or on the `--framerate 60` experiment? Can you also say how long the ribbon cables are? Thanks!

davidplowman, 2022-10-18:
Hi again, just wondering if there were any updates here. If not I'll consider closing this issue in a few days. Thanks!

davidplowman, 2022-11-01:
I'll close this one for the time being. Please feel free to open a new issue if there are any updates or anything else we can look at. Thanks.