raspberrypi/picamera2 #313: [BUG] UDP messages can be too long, aren't broken up

CorvetteCole, 2022-09-12:
**Describe the bug**
When utilizing UDP streaming as described in the docs, a resolution, bitrate, etc higher than shown easily cause an OSError: [Errno 90] Message too long. This appears to be because the UDP packets are larger than allowed by the OS, and need to be split up.

**To Reproduce**
Use a resolution and bitrate large enough to create big packets. Example script (must change target ip):
```
#!/usr/bin/python3

import socket
import time

from picamera2 import Picamera2
from picamera2.encoders import H264Encoder
from picamera2.outputs import FileOutput

picam2 = Picamera2()
video_config = picam2.create_video_configuration({"size": (1640, 1232)})
picam2.configure(video_config)
encoder = H264Encoder(1000000, repeat=True)

with socket.socket(socket.AF_INET, socket.SOCK_DGRAM) as sock:
    sock.connect(("10.11.9.112", 1234))
    stream = sock.makefile("wb")
    picam2.start_recording(encoder, FileOutput(stream))
    try:
        time.sleep(10000)
    finally:
        picam2.stop_recording()
```

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Console Output, Screenshots**
```
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/encoders/v4l2_encoder.py", line 190, in thread_poll
    self.outputframe(b, keyframe, (buf.timestamp.secs * 1000000) + buf.timestamp.usecs)
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/encoders/encoder.py", line 183, in outputframe
    out.outputframe(frame, keyframe, timestamp)
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/outputs/fileoutput.py", line 77, in outputframe
    self._write(frame, timestamp)
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/outputs/fileoutput.py", line 95, in _write
    self._fileoutput.write(frame)
  File "/usr/lib/python3.9/socket.py", line 722, in write
    return self._sock.send(b)
OSError: [Errno 90] Message too long
```

**Hardware :**
Raspberry Pi 4, Raspbian Bullseye, IMX219 (Camera v2.1)

**Additional context**
TCP streaming works fine, but is not applicable to our use case where we can't directly access the device.

chrisruk, 2022-09-12:
Thanks for reporting, will have a look at this

chrisruk, 2022-09-13:
Yes I think you're write it might be tricky to fix in FileOutput.  I'm just having a little look into UDP 
based protocols such as RTP.  But will do some more investigation into how they work.

It sounds like RTP might be able to split the payload for us, as well as detecting out of order delivery and packet loss.

davidplowman, 2022-09-13:
I'm feeling that we really ought to be able to send raw UDP, for example the `libcamera-vid` application manages this ([here](https://github.com/raspberrypi/libcamera-apps/blob/main/output/net_output.cpp#L99)).

To be honest I'm a bit disappointed that `socket.makefile` doesn't take care of this. It knows it's got a socket, it can check the type of the socket, why can't it make a file object that we can use like a normal file object?

I'm nervous about RTP for various reasons, I think some players may not like it unless we get an SDP message that says what's in the RTP stream, at which point we may be looking at RTSP servers (which means more complexity, or more libraries, or both). It may be that RTP is better for some applications, but I think the basic UDP should work too.

chrisruk, 2022-09-13:
Thanks for that c++ example David, I'll see if I can replicate that approach.  But yeah it's a shame
makefile can't abstract this.

CorvetteCole, 2022-09-13:
I've created a draft SocketOutput which simply takes in a socket and then sends data through that. If it is a datagram based socket, it will split up the frames. 

Whatever socket passed in to it must be opened before the SocketOutput is started.

This has not been tested and is more meant as a proof of concept for now, want to hear thoughts on it. It is quite similar to FileOutput

CorvetteCole, 2022-09-15:
will be solved by #317

closed 2022-09-16 by CorvetteCole