raspberrypi/picamera2 #315: [HOW-TO] Calling switch_mode_and_capture_image from function and then exit

eat-sleep-code, 2022-09-14:
**Please only ask one question per issue!**

**Describe what it is that you want to accomplish**
I have this code:

```python
def captureImage(filepath, raw = True):
	captured = camera.switch_mode_and_capture_image(configStill)
	captured.save('main', filepath)
	if raw == True:
		filepathDNG = filepath.replace('.jpg', '.dng')
		captured.save_dng(filepathDNG)
	return
```
I am calling it like this...
```python
print(' Capturing image: ' + filepath + '\n') 
captureImage(filepath, raw)  
print('did I come back here?')   #This line and any subsequent lines never fire.   I would expect it should return to this section after the capture is complete. 
```

**Describe alternatives you've considered**
I am trying to migrate an existing picamera project to picamera2

**Additional context**
Here is the current repo: https://github.com/eat-sleep-code/camera/tree/picamera-2-migration

davidplowman, 2022-09-15:
Hi, thanks for the question.

So I think the problem is that you're using `switch_mode_and_capture_image` which returns a PIL image in RGB format, and there's no way to get back from this to the raw buffer from which you could save a DNG.

There are many ways to do all this, but we also provide a couple of examples that might help: https://github.com/raspberrypi/picamera2/blob/main/examples/capture_dng.py captures just the DNG, whereas https://github.com/raspberrypi/picamera2/blob/main/examples/capture_dng_and_jpeg.py captures both a JPEG and a DNG.

Please let us know if there's anything else we can help with. Thanks!

davidplowman, 2022-10-03:
Hi, just wondering if there was any update here or anything related that we can help with? Thanks!

davidplowman, 2022-10-17:
Hi, I think I'll close this one for now if that's ok, but please open a new issue if there's anything else for us to look at. Thanks!

closed 2022-10-17 by davidplowman