raspberrypi/picamera2 #316: [BUG] Install package needs to be updated

ktorimaru, 2022-09-14:
The current install package do not contain this change:

https://github.com/raspberrypi/picamera2/commit/e2598de37f64b1f93ac109020476361a836f7d49

And therefore will not run.

davidplowman, 2022-09-15:
Hi, can you say what install package you are referring to? I believe both the python3-picamera2 apt package and version 0.3.3 in PyPI have this commit, but perhaps something else is going wrong?

ktorimaru, 2022-09-15:
Yesterday I used:
sudo apt update
sudo apt install -y python3-picamera2 --no-install-recommends

And got the Colorspace.jpeg error.

I made the changes in:
https://github.com/raspberrypi/picamera2/commit/e2598de37f64b1f93ac109020476361a836f7d49

and everything works.

davidplowman, 2022-09-16:
So this commit is definitely in all the current releases. I am wondering if you're not running quite the code that you expect. When you say "I made the changes in", what exactly do you mean - have you checked out the source code and are running from that?

One thing to check is that if you want to use the apt version after using the pip version, people have found that they need to uninstall the pip version first (`pip3 uninstall picamera2`). Could this be what's happening?

davidplowman, 2022-10-07:
Hi, have you been able to resolve the problems here or is there anything else for us to be looking at? Thanks!

davidplowman, 2022-10-17:
Hi, I think I'll close this one for now, but please open a new issue if anything else comes up. Thanks!