raspberrypi/picamera2 #321: [HOW-TO] Use picamera2 on Raspberry Pi Zero 2 W

slabua, 2022-09-16:
Is there any known issue in making picamera2 work on the Zero 2 W or am I missing something?

I had tried various settings but the farther I had been to was to open the capture window, which just showed in black and systematically froze the zero.

If any specific information related to my system is needed in order to make it work please let me know and I will comment accordingly.

davidplowman, 2022-09-16:
Hi, could you perhaps supply some further information please:

* Can you say what image you have installed. Is this one of the Raspberry Pi OS images released a few days ago? Can you say which one it is, i.e. Full/Lite, 32/64 bit?
* Have you edited anything in the /boot/config.txt, and if so, what?
* Can you say what camera you are using?
* Have you installed any other camera related software from anywhere else?
* Do libcamera-apps work, that is, if you enter `libcamera-hello` do you see a preview window? If not, are there any error messages in the console output?
* When you tried Picamera2, what script did you run? Were there any messages in the console window?

Thanks!

slabua, 2022-09-16:
> Hi, could you perhaps supply some further information please:

Hi David, thank you for your reply

> * Can you say what image you have installed. Is this one of the Raspberry Pi OS images released a few days ago? Can you say which one it is, i.e. Full/Lite, 32/64 bit?

It is a Bullseye 32 bit updated from Buster full

> * Have you edited anything in the /boot/config.txt, and if so, what?

I have tried various combinations of the following settings:
```
dtoverlay=vc4-kms-v3d
camera_auto_detect=1
gpu_mem=128 (also 256)
max_framebuffers=2 
```

> * Can you say what camera you are using?

Tried with both Pi Camera Rev 1.3 and 2.1

> * Have you installed any other camera related software from anywhere else?

Hm, do the legacy stack and qv4l2 from apt count?

> * Do libcamera-apps work, that is, if you enter `libcamera-hello` do you see a preview window? If not, are there any error messages in the console output?

I'm running headless, this is the output of libcamera-hello:
```
Preview window unavailable
[0:03:09.808692354] [1288]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3866-0c55e522
[0:03:09.940236212] [1292]  WARN RPI raspberrypi.cpp:1258 Mismatch between Unicam and CamHelper for embedded data usage!
[0:03:09.941751671] [1292]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media2 and ISP device /dev/media0
[0:03:09.942971383] [1288]  INFO Camera camera.cpp:1035 configuring streams: (0) 1640x1232-YUV420
[0:03:09.944043638] [1292]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
```
At this point ^C doesn't do anything, but Pi is responsive on another terminal.

> * When you tried Picamera2, what script did you run? Were there any messages in the console window?

```
In [1]: from picamera2 import Picamera2
   ...: picam2 = Picamera2()
   ...: picam2.start_and_capture_file("test123.jpg")
```
If I had previously run libcamera-apps the script is not able to access the camera already being in use, otherwise on a fresh boot:
```
[0:03:05.440438425] [1533]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3866-0c55e522
[0:03:05.526880818] [1554]  WARN RPI raspberrypi.cpp:1258 Mismatch between Unicam and CamHelper for embedded data usage!
[0:03:05.528808396] [1554]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
[0:03:05.539083530] [1533]  INFO Camera camera.cpp:1035 configuring streams: (0) 640x480-XBGR8888
[0:03:05.540122358] [1554]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 640x480-SBGGR10_1X10 - Selected unicam format: 640x480-pBAA
libEGL warning: DRI3: failed to query the version
libEGL warning: DRI2: failed to authenticate
qt.qpa.xcb: QXcbConnection: XCB error: 1 (BadRequest), sequence: 325, resource id: 779, major code: 155 (Unknown), minor code: 1
```
[image]

> 
> Thanks!

Hope this clarifies the situation a bit. Thanks to you!

davidplowman, 2022-09-17:
Thanks for the info. That gives us a few more things to check.

1. Make sure you delete the `gpu_mem` line from your /boot/config.txt file, that would only make things worse. The rest looks OK, though.
2. Seeing as you're running headless, and assuming you're logged in with X-forwarding, use `libcamera-hello --qt-preview`. Does that show an image? We really need to get this to work first before worrying about anything else.

The other thing I'm nervous about is upgrading a Buster image. I know it's a thing that should work, but we don't particularly recommend it because the different settings that people may have when they upgrade is very hard to guess at, and the testing burden in trying to guarantee that it works correctly in all cases is very high.

If you have a spare SD card lying around it might be worth installing the new 32-bit Bullseye with Picamera2 pre-installed. If that behaves differently (and works) then we know there's something amiss with the setup of your other image.

slabua, 2022-09-17:
> Thanks for the info. That gives us a few more things to check.
> 
> 1. Make sure you delete the `gpu_mem` line from your /boot/config.txt file, that would only make things worse. The rest looks OK, though.

Commented out

> 2. Seeing as you're running headless, and assuming you're logged in with X-forwarding, use `libcamera-hello --qt-preview`. Does that show an image? We really need to get this to work first before worrying about anything else.

This is the output of the command, but no preview window showed up at all
```
libcamera-hello --qt-preview
Preview window unavailable
[0:01:50.113288234] [1100]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3866-0c55e522
[0:01:50.239841964] [1101]  WARN RPI raspberrypi.cpp:1258 Mismatch between Unicam and CamHelper for embedded data usage!
[0:01:50.241442874] [1101]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
[0:01:50.242690758] [1100]  INFO Camera camera.cpp:1035 configuring streams: (0) 1640x1232-YUV420
[0:01:50.243533257] [1101]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
```

> The other thing I'm nervous about is upgrading a Buster image. I know it's a thing that should work, but we don't particularly recommend it because the different settings that people may have when [...]

Thanks that makes sense, I'll try to make a fresh installation and see what happens.

davidplowman, 2022-09-17:
I don't really understand why libcamera-hello wouldn't give you a preview window. Do other applications work with X forwarding?

Is there any possibility this was at one time a Lite image which had the desktop added later? That might be one explanation for libcamera-hello not doing quite what I expect. But as I said, it's hard to know for sure...

slabua, 2022-09-19:
Hi David,
> I don't really understand why libcamera-hello wouldn't give you a preview window. Do other applications work with X forwarding?

Yes other applications forward as expected.

> Is there any possibility this was at one time a Lite image which had the desktop added later? That might be one explanation for libcamera-hello not doing quite what I expect. But as I said, it's hard to know for sure...

I think it wasn't a Lite image but I am not 100% sure.

[Update]
I installed a fresh 64 bit Bullseye image and libcamera-hello --qt-preview works.
Then I tried to capture an image with the same script as before and it froze

```
>>> from picamera2 import Picamera2
>>> picam2 = Picamera2()
[0:14:40.562413980] [1530]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3866-0c55e522
[0:14:40.660492654] [1534]  WARN RPI raspberrypi.cpp:1258 Mismatch between Unicam and CamHelper for embedded data usage!
[0:14:40.663274934] [1534]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
>>> picam2.start_and_capture_file("test123.jpg")
[0:14:59.107016793] [1530]  INFO Camera camera.cpp:1035 configuring streams: (0) 640x480-XBGR8888
[0:14:59.108810603] [1534]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 640x480-SBGGR10_1X10 - Selected unicam format: 640x480-pBAA
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
[0:15:06.795421758] [1537]  INFO Camera camera.cpp:1035 configuring streams: (0) 3280x2464-BGR888
[0:15:06.797843142] [1534]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 3280x2464-SBGGR10_1X10 - Selected unicam format: 3280x2464-pBAA
[0:15:12.498568837] [1537]  INFO Camera camera.cpp:1035 configuring streams: (0) 640x480-XBGR8888
[0:15:12.530857948] [1534]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 640x480-SBGGR10_1X10 - Selected unicam format: 640x480-pBAA
>>>
Message from syslogd@raspberrypi at Sep 19 08:29:08 ...
 kernel:[  914.465961] Internal error: SP/PC alignment exception: 8a000000 [#1] PREEMPT SMP

Message from syslogd@raspberrypi at Sep 19 08:29:08 ...
 kernel:[  914.466654] Code: bad PC value

Message from syslogd@raspberrypi at Sep 19 08:29:08 ...
 kernel:[  914.839600] Internal error: Oops: 96000004 [#2] PREEMPT SMP

Message from syslogd@raspberrypi at Sep 19 08:29:08 ...
 kernel:[  914.840353] Code: a90153f3 aa0003f3 f9402014 b40000b4 (f9400680)

```
capture window showed up, though.

davidplowman, 2022-09-19:
Thanks for the update. At least libcamera-hello seems to be working. Might be worth trying libcamera-still and seeing if that saves a JPEG successfully (e.g. `libcamera-sitll -o test.jpg`).

As regards Picamera2, maybe break the script down a bit to see what does and doesn't work. Let's start very simple:
```
import time
from picamera2 import Picamera2, Preview
picam2 = Picamera2()
picam2.start_preview(Preview.QT)
picam2.start()
time.sleep(5)
picam2.capture_file("test.jpg")
```
Does that capture a (very small) JPEG file, or does it fail somewhere?

To be fair the errors you have above look very nasty, kind of kernel/hardware related. It would probably be worth attaching or posting the contents of `dmesg`. We have been having some trouble with Pi Zeros, though I wasn't aware this extended to Zero 2s. I'll see what I can find out. There has been some suggestion the Pi Zero issues are related to power supplies, though I don't think that's ever been proven. Out of interest, what kind of power supply are you using?

slabua, 2022-09-19:
> Thanks for the update. At least libcamera-hello seems to be working. Might be worth trying libcamera-still and seeing if that saves a JPEG successfully (e.g. `libcamera-sitll -o test.jpg`).

It couldn't save the picture
```
$ libcamera-still -o test123.jpg
libEGL warning: DRI3: failed to query the version
libEGL warning: DRI2: failed to authenticate
X Error of failed request:  BadRequest (invalid request code or no such operation)
  Major opcode of failed request:  155 ()
  Minor opcode of failed request:  1
  Serial number of failed request:  16
  Current serial number in output stream:  16
```

> As regards Picamera2, maybe break the script down a bit to see what does and doesn't work. Let's start very simple:
>
> ```
> import time
> from picamera2 import Picamera2, Preview
> picam2 = Picamera2() [...]

it worked fine
```
>>> import time
>>> from picamera2 import Picamera2, Preview
>>> picam2 = Picamera2()
[0:03:44.715456968] [1437]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3866-0c55e522
[0:03:44.837526995] [1438]  WARN RPI raspberrypi.cpp:1258 Mismatch between Unicam and CamHelper for embedded data usage!
[0:03:44.839502866] [1438]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
>>> picam2.start_preview(Preview.QT)
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
>>> picam2.start()
[0:04:15.226485388] [1437]  INFO Camera camera.cpp:1035 configuring streams: (0) 640x480-XBGR8888
[0:04:15.228532911] [1438]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 640x480-SBGGR10_1X10 - Selected unicam format: 640x480-pBAA
>>> picam2.capture_file("test456.jpg")
{'ScalerCrop': (1000, 752, 1280, 960), 'SensorBlackLevels': (4096, 4096, 4096, 4096), 'ColourTemperature': 3691, 'DigitalGain': 1.0000416040420532, 'ColourGains': (1.3881982564926147, 1.9284855127334595), 'AeLocked': True, 'Lux': 17.741283416748047, 'FrameDuration': 66701, 'ColourCorrectionMatrix': (1.702760100364685, -0.5760537385940552, -0.12670700252056122, -0.4733457863330841, 1.873021125793457, -0.39967527985572815, -0.07629621028900146, -0.8147128224372864, 1.891008973121643), 'SensorTimestamp': 295642770000, 'AnalogueGain': 8.0, 'ExposureTime': 66663}
>>> picam2.stop()
>>> picam2.close()
>>> exit()

```

> To be fair the errors you have above look very nasty, kind of kernel/hardware related. It would probably be worth attaching or posting the contents of `dmesg`. We have been having some trouble with [...]

Long ago when the Pi Zero2 came out I had read something about the way the camera is handled, somehow differently than before, but I can't recall the source. Furthermore, the camera led does not turn on, despite it being active in config.txt. I also have a Pi Zero (non 2) at office, I will check it later too.

As for the power supply, I'm using a ups pack with a lipo battery, but it is also connected to the pc via a usb cable. I will also check with a regular power supply.

slabua, 2022-09-21:
I was able to try it with a different power supply, and it worked fine with the following script:
```
In [1]: from picamera2 import Picamera2
   ...: picam2 = Picamera2()
   ...: picam2.start_and_capture_file("test123.jpg")
```
Still fails with `libcamera-sitll -o test.jpg`, though.

davidplowman, 2022-09-21:
I have a few more things to try. Let's try them one at a time so as not to get ourselves muddled as to which, if any, had an effect!

Firstly, can you confirm that this is using the latest image that we released this month that is still failing in `libcamera-still`?

Now, there was a recent fix to the CSI-2 receiver that didn't quite make that release, so you could do a `sudo rpi-update` to fetch it. (Warning: not fully tested software, usual caveats about important data and backups apply.) Let's try that and see if it makes a difference.

If that doesn't help then that's useful to know in itself, and we can go on afterwards and try a couple of other things. Thanks!

slabua, 2022-09-21:
> I have a few more things to try. Let's try them one at a time so as not to get ourselves muddled as to which, if any, had an effect!
> 
> Firstly, can you confirm that this is using the latest image that we released this month that is still failing in `libcamera-still`?

Yes, I just flashed it yesterday and there is no modification at all.

> Now, there was a recent fix to the CSI-2 receiver that didn't quite make that release, so you could do a `sudo rpi-update` to fetch it. (Warning: not fully tested software, usual caveats about [...]

I ran the update and the command equally failed.
Then I thought, should I perhaps need to run it with sudo? And it worked.
On the other sdcard with Buster updated to Bullseye I didn't need to use sudo, though, it didn't work regardless.

Groups the user belongs to: 4(adm),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),105(input),107(render),109(netdev),117(lpadmin),997(gpio),998(i2c),999(spi)

Wrapping up with the current information:

- A proper power supply is needed to capture large images
- libcamera-still works if used with sudo
- The simple capture script in python doesn't require root privileges

davidplowman, 2022-09-21:
It absolutely shouldn't be necessary to use sudo to make libcamera-still work, in fact I don't see how it can make any difference. Are you sure the behaviour isn't simply intermittent, sometimes working and sometimes failing regardless of using sudo?

slabua, 2022-09-23:
I don't know why, but it only works if executed with sudo.

davidplowman, 2022-09-23:
This is all very hard to understand. Here is one more thing to try. Could you add
```
over_voltage=4
```
into your /boot/ocnfig.txt file (and reboot). Let's see if that makes any difference.

If it is still failing, could you post whatever it is that you get in the console when it crashes. Could you also please post the contents of `dmesg` (maybe run `sudo dmesg -C` to clear it _before_ running `libcamera-still`, just to cut out all the rubbish). Thanks!

slabua, 2022-09-23:
I added the over_voltage line (which by the way I use indeed in the other sd card), but I still need sudo to execute the command.
The error is still the original one:
```
$ libcamera-still -o test0d.jpg
libEGL warning: DRI3: failed to query the version
libEGL warning: DRI2: failed to authenticate
X Error of failed request:  BadRequest (invalid request code or no such operation)
  Major opcode of failed request:  155 ()
  Minor opcode of failed request:  1
  Serial number of failed request:  16
  Current serial number in output stream:  16
```
After cleaning dmesg and running again the libcamera command, there is completely no log on dmesg.

For completeness as previously also posted, I attach here the log of the terminal when using sudo instead:
```
$ sudo libcamera-still -o test0d.jpg
X11 connection rejected because of wrong authentication.
Preview window unavailable
[0:08:46.184332720] [1494]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3866-0c55e522
[0:08:46.283800655] [1495]  WARN RPI raspberrypi.cpp:1258 Mismatch between Unicam and CamHelper for embedded data usage!
[0:08:46.285804677] [1495]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
[0:08:46.287461206] [1494]  INFO Camera camera.cpp:1035 configuring streams: (0) 1640x1232-YUV420
[0:08:46.288519518] [1495]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
[0:08:51.459103601] [1494]  INFO Camera camera.cpp:1035 configuring streams: (0) 3280x2464-YUV420 (1) 3280x2464-SBGGR10_CSI2P
[0:08:51.469876508] [1495]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 3280x2464-SBGGR10_1X10 - Selected unicam format: 3280x2464-pBAA
Still capture image received
```
dmesg still empty after this too.

chrisruk, 2022-09-23:
Does adding '--nopreview' to libcamera-still make any different out of interest to the output, when not using sudo

slabua, 2022-09-23:
It worked

chrisruk, 2022-09-23:
Thanks.  I just checked a note I made a while ago, I used xfce with picamera and noticed I got 'libEGL warning: DRI2: failed to authenticate' when the screen was 'locked'.

Could you try unlocking the screen then running 
```
libcamera-still -o test0d.jpg
```

slabua, 2022-09-23:
Thank you for checking, but there is no screen. Perhaps running it headlessly is the source of all problems.

davidplowman, 2022-09-23:
Hi, the documentation possibly needs improving, but here are some links to the discussion of preview windows.

* [libcamera-apps](https://www.raspberrypi.com/documentation/accessories/camera.html#preview-window) - if you're running headless you either need to use `-n`, or `--qt-preview` if you want to use X-forwarding (note that this latter option is expensive as it involves software rendering, to say nothing of the uncompressed video travelling over the network).
* [Picamera2](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf) - see section 3.4 on page 12.

slabua, 2022-09-24:
> Ah sorry. Are you still using an environment though like XFCE/LXDE?

@chrisruk  It's the official image, I believe there is LXDE installed. (Yes, dpkg says so)

> if you're running headless you either need to use `-n`, or `--qt-preview`

@davidplowman I just verified that using --qt-preview also does not need sudo.
Sudo is only required if i run the command without neither -n and --qr-preview.

davidplowman, 2022-09-26:
I've been through re-checking the behaviour I get with my Pi Zero 2 and a clean image. I've logged in using `ssh`, and I've tried it with and without X forwarding.

- Without X forwarding. `libcamera-still -o test.jpg` works and does the capture, but complains that it can't make a preview window, which is expected.
- With X forwarding. `libcamera-still -o test.jpg` fails with a `failed to import fd 19` message and aborts. This is expected because it's trying to use the Pi's GPU to render onto the display, which is impossible.
- With X forwarding: both `-n` (no preview) and `--qt-preview` work, the latter displaying a small preview window on my laptop.

So I don't really understand what's different in your case, though I think I'm right in saying that the only case for which `sudo` is "required" is the one that shouldn't work anyway. I suspect that all `sudo` may be doing is to cause the preview to fail earlier, so it ends up behaving as though you weren't using X forwarding at all.

If that's all correct, then I'm inclined to say the answer is one of: use `--qt-preview` if you want a preview window, or if you don't either use `-n` or don't use X forwarding. This is exactly how everything is meant to work. There may be a bit of a question as to why the failure case fails in exactly that way, but given that we would at most be swapping one error message for another, I'm not sure it's worth worrying about too much. Have I understand that right? Thanks!

slabua, 2022-09-26:
Yes thank you for going thru all the testing scenarios!
I think the main turning point of this thread was to try using a new Bullseye image, which I didn't really want to do, but if it is necessary I will settle on patiently setting it up from scratch.

slabua, 2022-10-07:
Hi David, thanks a lot, I think that would be all for now~

davidplowman, 2022-10-07:
Great, thanks!

closed 2022-10-07 by davidplowman

davidplowman, 2022-10-17:
Hi, you can look in the image metadata, for example
```
...
metadata = picam2.capture_metadata()
print(metadata["FrameDuration"])
```

faeq123, 2022-10-17:
ya this worked thanks