raspberrypi/picamera2 #323: [BUG] CircularOutput video playback is slow before the output is specified

triple3567, 2022-09-17:
**Describe the bug**
When using the CircularOutput class to record a video, the frames recorded before the output is specified plays back at a much lower frame rate then expected. Once the output is specified, the frames are encoded properly and the video returns to its expected framerate. 

**To Reproduce**

Here is a simple python script to create a video using CircularOutput. The buffer size of CircularOutput should only have enough space for 5 seconds of 30 fps video. Once the output is specified, 10 more seconds of 30 fps video should be recorded. The video is saved to the file "recording.h264", and the timestamps associated with each frame is saved to the file "timestamps.txt"
````
from picamera2.encoders import H264Encoder
from picamera2.outputs import CircularOutput
from picamera2 import Picamera2
import time

#
# INITIALIZE CAMERA ENDCODER AND OUTPUT OBJECTS
#
picam2 = Picamera2()
video_config = picam2.create_video_configuration(main={"size": (1920,1080), "format": "RGB888"}, lores={"size": (640,480), "format": "YUV420"}, controls={"NoiseReductionMode": 2, "FrameDurationLimits": (33333, 33333)})
picam2.configure(video_config)
encoder = H264Encoder(bitrate=10000000)
output = CircularOutput(buffersize=150, pts="timestamps.txt")

picam2.start_recording(encoder, output)
time.sleep(10)

output.fileoutput = "recording.h264"
output.start()
time.sleep(10)
output.stop()
````

**Expected behaviour**
The video should playback with with a consistant framerate throughout, and the transition from frames recorded an output is specified to after the output is specified should be seemless. 

**Console Output, Screenshots**
[Here is a sample video and the timestamps associated with each frame of the video](https://drive.google.com/drive/folders/1AsbOLpUPKlXz245p_WM5paeG_7_nSfpa?usp=sharing). You can see that the first 5 seconds of the video have a much slower framerate than the following 10 seconds (Notice the speed of the fan in the video).

**Hardware :**
OS: Raspbian GNU/Linux 11 (bullseye)
Python: Python 3.9.2
Camera: Arducam IMX477 Raspberry Pi HQ Camera

davidplowman, 2022-09-20:
Hi, thanks for posting this.

I had a look through the timestamps and watched the recording. As far as I could tell, the timestamps all appear to show differences of 33.3ms to quite a high degree of accuracy, and when I played the file I couldn't see anything obviously wrong.

Can you say how you are playing the video? I know that we've had trouble playing these kinds of files with vlc because it doesn't seem handle plain h.264 bit streams very well, so I tend to use ffplay instead (I was also playing it on my laptop).

triple3567, 2022-09-21:
Yup, VLC seemed to be the issue. Works perfect with ffplay. Thanks for you help.

closed 2022-09-21 by triple3567