raspberrypi/picamera2 #324: [BUG] capture_image bombs out when specifying "lores" instead of "main" as argument

tvoverbeek, 2022-09-18:
**Describe the bug**
Capturing a PIL image only works for the "main" stream.
Taking the example in section 6.1.2 of the manual but replacing "main" with "lores"
causes an error and when used in a program  hangs it because the image request is never completed

**To Reproduce**
Use the following code:
```
from picamera2 import Picamera2
import time
picam2 = Picamera2()
picam2.start()
time.sleep(1)
image = picam2.capture_image("lores")
```
Here is a console log:
```
tvo@picam2-32:~ $ python bug.py
[4:07:33.004749969] [2906]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3866-0c55e522
[4:07:33.031900844] [2907]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media1 and ISP device /dev/media0
[4:07:33.035236964] [2906]  INFO Camera camera.cpp:1035 configuring streams: (0) 640x480-XBGR8888
[4:07:33.035597290] [2907]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGRBG10_1X10 - Selected unicam format: 640x480-pgAA
Traceback (most recent call last):
  File "/home/tvo/bug.py", line 6, in <module>
    image = picam2.capture_image("lores")
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1259, in capture_image
    return self._execute_or_dispatch(partial(self.capture_image_, name), wait, signal_function)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1038, in _execute_or_dispatch
    done, result = function()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1243, in capture_image_
    result = request.make_image(name)
  File "/usr/lib/python3/dist-packages/picamera2/request.py", line 150, in make_image
    return self.picam2.helpers.make_image(self.make_buffer(name), self.picam2.camera_config[name], width, height)
  File "/usr/lib/python3/dist-packages/picamera2/request.py", line 134, in make_buffer
    with _MappedBuffer(self, name) as b:
  File "/usr/lib/python3/dist-packages/picamera2/request.py", line 17, in __init__
    self.__fb = request.request.buffers[stream]
KeyError: None
tvo@picam2-32:~ $ 
```

**Expected behaviour**
Since PIL does not accept YUV420 format, anything other than "main" should be reported as an error.

**Hardware :**
RPI 4B 2G, latest OS (Linux picam2-32 5.15.61-v7l+ #1579 SMP Fri Aug 26 11:13:03 BST 2022 armv7l GNU/Linux)
OV5647 camera

**Additional context**
libcamera-apps build: 1bf0ccacf2ef 31-08-2022 (16:16:34)
libcamera build: v0.0.0+3866-0c55e522
picamera2 0.3.3-1

davidplowman, 2022-09-20:
Thanks for reporting this.

Looking at what's happening, the problem is actually that the "lores" stream is undefined, not specifically that it's YUV420 (which it would be if it were defined). So I definitely think the error message can be improved there!

On the subject of YUV420 (if the "lores" were defined), I'm slightly hesitant to tie an error directly to the attempt to use the "lores" stream because I still harbour fanciful thoughts that one day some magic on the GPU might lift the YUV-only restriction. But again the error message could certainly be more helpful.

I also agree with the point about the system not cleaning up very well after an error, I'll look at improving that too.

davidplowman, 2022-09-20:
Here's what I'm thinking... https://github.com/raspberrypi/picamera2/pull/327

tvoverbeek, 2022-09-20:
@davidplowman PR #327 looks good to me.
Closing this usse. Any further discussion (if needed) in the PR.

closed 2022-09-20 by tvoverbeek