raspberrypi/picamera2 #325: [HOW-TO] deploy frame_server.py to gunicorn

tennisparty, 2022-09-18:
**Please only ask one question per issue!**

**Describe what it is that you want to accomplish**
I have adapted the frame_server.py based on some excellent advice from David Plowman a few weeks ago - [issue 255](https://github.com/raspberrypi/picamera2/issues/255) and am running a lores stream (from an ov9281) onto a mjpeg server. This works fine if I run the app solely with python and flask. However if I attempt to run the app with gunicorn no frames are accessed by the mjpeg server. Let me know if this is the wrong place to post, I wasn't sure whether the gunicorn GitHub might be better.

ADAPTED frame_server.py
```#!/usr/bin/python3

# These two are only needed for the demo code below the FrameServer class.
from picamera2 import Picamera2
import time

from flask import Response
from flask import Flask, render_template, request

import cv2

from threading import Condition, Thread

import simplejpeg

class FrameServer:
    def __init__(self, picam2, stream='lores'):
        """A simple class that can serve up frames from one of the Picamera2's configured
        streams to multiple other threads.
        Pass in the Picamera2 object and the name of the stream for which you want
        to serve up frames."""
        self._picam2 = picam2
        self._stream = stream
        self._array = None
        self._condition = Condition()
        self._running = True
        self._count = 0
        self._thread = Thread(target=self._thread_func, daemon=True)
        

    @property
    def count(self):
        """A count of the number of frames received."""
        return self._count

    def start(self):
        """To start the FrameServer, you will also need to start the Picamera2 object."""
        self._thread.start()

    def stop(self):
        """To stop the FrameServer, first stop any client threads (that might be
        blocked in wait_for_frame), then call this stop method. Don't stop the
        Picamera2 object until the FrameServer has been stopped."""
        self._running = False
        self._thread.join()

    def _thread_func(self):
        while self._running:
            array = self._picam2.capture_array(self._stream)
            self._count += 1
            with self._condition:
                self._array = array
                # time.sleep(0)
                self._condition.notify_all()

    def wait_for_frame(self, previous=None):
        """You may optionally pass in the previous frame that you got last time you
        called this function. This will guarantee that you don't get duplicate frames
        returned in the event of spurious wake-ups, and it may even return more
        quickly in the case where a new frame has already arrived."""
        with self._condition:
            if previous is not None and self._array is not previous:
                return self._array
            while True:
                self._condition.wait()
                time.sleep(0)
                if self._array is not previous:
                    return self._array
                    

app = Flask(__name__)

@app.route("/mjpeg")
def mjpeg():
    return Response(thread1_func(), mimetype='multipart/x-mixed-replace; boundary=frame')

@app.route("/", methods=['GET', 'POST'])
def index():
    return render_template('index2.html')

if __name__ == "__main__":
    # Thread(target=lambda: app.run('0.0.0.0', port=8000, debug=True, use_reloader=False)).start()
    Thread(target=lambda: app.run('0.0.0.0', debug=True, use_reloader=False)).start()
#    Thread(target=lambda: app.run(host='0.0.0.0', threaded=True)).start()

mjpeg_abort = False
frame = None
mjpeg_condition = Condition()

# Below here is just demo code that uses the class:

def thread1_func():
    global thread1_count
    print("mjpeg_encode loop started")
    while not thread_abort:
        frame = server.wait_for_frame()
        thread1_count += 1
        print(frame)
        # with self._condition():
        #     self._condition.wait()
        yuv = frame
        rgb = cv2.cvtColor(yuv, cv2.COLOR_YUV420p2RGB)
        buf = simplejpeg.encode_jpeg(rgb, quality=80, colorspace='BGR', colorsubsampling='420')
        frame = buf
        yield (b'--frame\r\nContent-Type: image/jpeg\r\n\r\n' + frame + b'\r\n')

def thread2_func():
    global thread2_count
    frame = None
    while not thread_abort:
        frame = server.wait_for_frame(frame)
        thread2_count += 1

thread_abort = False
thread1_count = 0
thread2_count = 0
thread1 = Thread(target=thread1_func)
thread2 = Thread(target=thread2_func)

picam2 = Picamera2()
lores = (640, 400)
picam2.configure(picam2.create_video_configuration(controls={"FrameDurationLimits": (33333, 33333)},main={"size": (1280, 720)}, lores={"size": lores}))
server = FrameServer(picam2)
thread1.start()
thread2.start()
server.start()
picam2.start()

time.sleep(20)

thread_abort = True
thread1.join()
thread2.join()
server.stop()
picam2.stop()

print("Thread1 received", thread1_count, "frames")
print("Thread2 received", thread2_count, "frames")
print("Server received", server.count, "frames")`
```

INDEX.HTML
```
<!DOCTYPE html>
   <head>
      <title>TITLE</title>
      <link rel="stylesheet" href="../static/style.css/">
   </head>
   <body>
      <section>
         <h1>Header1</h1>
         <h2>Last Refresh: TIME</h2>
      </section>
      <img src="{{ url_for('mjpeg') }}" class="expanded-image">

   </body>
</html>
```

Running with just python:
``` python frame_server.py ```

 ```
Thread1 received 574 frames
Thread2 received 595 frames
Server received 596 frames
 ```
Note thread 1 starts receiving frame after webpage loaded

Running with gunicorn:
```gunicorn --worker-class gevent --workers 1 --bind 0.0.0.0:5000 wsgi:app```

```
Thread1 received 0 frames
Thread2 received 594 frames
Server received 595 frames
```
Note the webpage does not load until the sleep timer runs out and the mjpeg thread has stopped

**Describe alternatives you've considered**
I have tried adapting this code to run picamera2 as the simple example does work with gunicorn but I cannot get it to work. `https://github.com/miguelgrinberg/flask-video-streaming/blob/master/camera_pi.py`

**Additional context**
On this blog post [blog](https://blog.miguelgrinberg.com/post/flask-video-streaming-revisited) there is talk of needing to release the cpu to use gunicorn and gevent. I wonder if this could be the issue?

fipwmaqzufheoxq92ebc, 2022-09-19:
- gunicorn tries to import the app from your file (i.e. runs ```from wsgi import app```) which runs the code in your file, including the time.sleep(), so gunicorn gets your "app" after your file finished, when the timer ran out (and the threads exited).
If you run your app with "python wsgi.py", the ```if __name__ == '__main__'``` part runs the server in a background thread, before the timer starts.
To fix it, you could run a background thread which sleeps and then exits. (idea: use threading.Timer).

- thread1 does nothing because thread1_func creates an generator, which needs to be iterated over.

sample:
```
def x():
   while True:
       print("Hello World")
       yield 0
```
if you call it:
```x()```,
it does nothing, but if you use:
```
for a in x():
   pass
```
it will print "Hello World" forever.

- > On this blog post blog there is talk of needing to release the cpu to use gunicorn and gevent. I wonder if this could be the issue?

the ```self._condition.wait()``` in wait_for_frame already releases the cpu.

tennisparty, 2022-09-19:
Hello, thanks for the help. I've been trying to understand this but I don't think I quite follow. I've tried changing the code as below. It seems the timer in the background is now working. I still seem to be having problems with the line
```yield ((b'--frame\r\nContent-Type: image/jpeg\r\n\r\n' + frame + b'\r\n'))``` in thread1_func()  when running with gunicorn. The code loops normally (threaded) if run through python but when run through gunicorn thread1_func() only enters once. If I comment out the line then it loops as expected. Sorry I realise this is probably my misunderstanding

```
#!/usr/bin/python3

# These two are only needed for the demo code below the FrameServer class.
from picamera2 import Picamera2
import time

from flask import Response
from flask import Flask, render_template, request

import cv2

from threading import Condition, Thread, Timer

import simplejpeg

class FrameServer:
    def __init__(self, picam2, stream='lores'):
        """A simple class that can serve up frames from one of the Picamera2's configured
        streams to multiple other threads.
        Pass in the Picamera2 object and the name of the stream for which you want
        to serve up frames."""
        self._picam2 = picam2
        self._stream = stream
        self._array = None
        self._condition = Condition()
        self._running = True
        self._count = 0
        self._thread = Thread(target=self._thread_func, daemon=True)
        

    @property
    def count(self):
        """A count of the number of frames received."""
        return self._count

    def start(self):
        """To start the FrameServer, you will also need to start the Picamera2 object."""
        self._thread.start()

    def stop(self):
        """To stop the FrameServer, first stop any client threads (that might be
        blocked in wait_for_frame), then call this stop method. Don't stop the
        Picamera2 object until the FrameServer has been stopped."""
        self._running = False
        self._thread.join()

    def _thread_func(self):
        while self._running:
            array = self._picam2.capture_array(self._stream)
            self._count += 1
            with self._condition:
                self._array = array
                # time.sleep(0)
                self._condition.notify_all()

    def wait_for_frame(self, previous=None):
        """You may optionally pass in the previous frame that you got last time you
        called this function. This will guarantee that you don't get duplicate frames
        returned in the event of spurious wake-ups, and it may even return more
        quickly in the case where a new frame has already arrived."""
        with self._condition:
            if previous is not None and self._array is not previous:
                return self._array
            while True:
                self._condition.wait()
                time.sleep(0)
                if self._array is not previous:
                    return self._array
                    

app = Flask(__name__)

@app.route("/mjpeg")
def mjpeg():
    # print(framez)
    # hat = (b'--frame\r\nContent-Type: image/jpeg\r\n\r\n' + frame + b'\r\n')
    return Response(thread1_func(), mimetype='multipart/x-mixed-replace; boundary=frame')

@app.route("/", methods=['GET', 'POST'])
def index():
    return render_template('index2.html')

if __name__ == "__main__":
    # Thread(target=lambda: app.run('0.0.0.0', port=8000, debug=True, use_reloader=False)).start()
    Thread(target=lambda: app.run('0.0.0.0', debug=True, use_reloader=False)).start()
#    Thread(target=lambda: app.run(host='0.0.0.0', threaded=True)).start()

mjpeg_abort = False
frame = None
mjpeg_condition = Condition()

def thread1_func():
       # I have also tried - for a in thread1_func():
        i=0
        while i <= 10:
            global thread1_count
            print(thread1_count)
            frame = server.wait_for_frame()
            thread1_count += 1
            print(thread1_count)
            yuv = frame
            rgb = cv2.cvtColor(yuv, cv2.COLOR_YUV420p2RGB)
            buf = simplejpeg.encode_jpeg(rgb, quality=80, colorspace='BGR', colorsubsampling='420')
            frame = buf
            yield ((b'--frame\r\nContent-Type: image/jpeg\r\n\r\n' + frame + b'\r\n'))

def thread2_func():
    global thread2_count
    frame = None
    while not thread_abort:
        frame = server.wait_for_frame(frame)
        thread2_count += 1

thread_abort = False
thread1_count = 0
thread2_count = 0
thread1 = Thread(target=thread1_func)
thread2 = Thread(target=thread2_func)

picam2 = Picamera2()
lores = (640, 400)
picam2.configure(picam2.create_video_configuration(controls={"FrameDurationLimits": (33333, 33333)},main={"size": (1280, 720)}, lores={"size": lores}))
server = FrameServer(picam2)
thread1.start()
thread2.start()
server.start()
picam2.start()

def hello():
    print("timer ended")
    print("Thread1 received", thread1_count, "frames")
    print("Thread2 received", thread2_count, "frames")
    print("Server received", server.count, "frames")
t = Timer(3, hello)
t.start()
time.sleep(30)

thread_abort = True
t.join()
thread1.join()
thread2.join()
server.stop()
picam2.stop()
t.stop()
```

tennisparty, 2022-09-20:
Think I have it sorted, it was an issue with the `if __name__` statement

closed 2022-09-20 by tennisparty

mnbf9rca, 2022-10-06:
@tennisparty would you mind sharing your flask implementation? I'm also struggling to get implement picamera streaming with flask.