raspberrypi/picamera2 #328: [BUG]

PG1955, 2022-09-21:
When using a Arducam IMX462 Low Light Camera.
After running apt update and upgrade I found this error.

>>> %Run Capture.py
Traceback (most recent call last):
  File "/home/pi/examples/Capture.py", line 4, in <module>
    from picamera2 import Picamera2
  File "/usr/lib/python3/dist-packages/picamera2/__init__.py", line 1, in <module>
    from .configuration import CameraConfiguration, StreamConfiguration
  File "/usr/lib/python3/dist-packages/picamera2/configuration.py", line 1, in <module>
    from .controls import Controls
  File "/usr/lib/python3/dist-packages/picamera2/controls.py", line 4, in <module>
    from libcamera import ControlType, Size, Rectangle
  File "/usr/lib/python3/dist-packages/libcamera/__init__.py", line 4, in <module>
    from ._libcamera import *
ImportError: /usr/lib/python3/dist-packages/libcamera/_libcamera.cpython-39-arm-linux-gnueabihf.so: undefined symbol: _ZN9libcamera10ColorSpace4SyccE

**To Reproduce**
Ran examples/Capture.py

**Expected behaviour**
Expected to run without errors.

**Console Output, Screenshots**
tail /boot/config.txt
[all]
gpu_mem=128
dtoverlay=arducam
disable_camera_led=1

i@wild4:~ $ dpkg --list | grep libcamera
ii  libcamera-apps                       0.0.8                            armhf        libcamera-apps
ii  libcamera-dev                        0.0.8                            armhf        libcamera
ii  libcamera0:armhf                     0~git20220826+3fad116f-1         armhf        complex camera support library
ii  python3-libcamera                    0~git20220826+3fad116f-1         armhf        complex camera support library (Python 3 bindings)

pi@wild4:~ $ dpkg --list | grep picamera2
ii  python3-picamera2                    0.3.3-1                          all          Libcamera based python library

**Hardware :**
pi@wild4:~ $ uname -a
Linux wild4 5.15.61-v7l+ #1579 SMP Fri Aug 26 11:13:03 BST 2022 armv7l GNU/Linux

.https://www.arducam.com/docs/cameras-for-raspberry-pi/pivariety/2mp-imx462-pivariety-low-light-camera-module/

**Additional context**
I am not sure if this is a problem with picamera2 or libcamera.

davidplowman, 2022-09-21:
You appear to have an old version of libcamera. If you're using the standard Raspberry Pi version of libcamera you need to use `sudo apt upgrade` (or equivalent) to update it. If you're using a version downloaded from Arducam then you will need to go to Arducam for support.

PG1955, 2022-09-21:
Thanks David,

I will contact Arducam support.

Kind regards,

Peter

davidplowman, 2022-09-21:
The symbol that it can't find is "Sycc" which was a fairly recent change in the libcamera API (it was previously known as "Jpeg"). My guess is that Arducam haven't yet caught up with that latest version. I'll close this ticket for now, but if it does turn out to be something else that we can help you with, please come back to us and re-open it. Thanks!

closed 2022-09-21 by davidplowman