raspberrypi/picamera2 #331: [HOW-TO] Generate the negative of an image

jareff-g, 2022-09-22:
Hello,

Thanks for PiCamera2, it is great and easy to migrate to. Thahks to it I have been able to make an existing PiCamera legacy app run on Raspberry Pi OS 64 bit and benefit from the additional performance (which is quite bit in this case).

So far, I have been able to migrate all existing features without any problem, except for this one: **Generating the negative of an image** (which was achieved in PiCamera legacy with _image_effects='negative'_).

As a possible alternative, I have been looking at the Color Correction Matrix in case I could use it to alter the image in such a way, but did not arrive anywhere.

I would appreciat any help you can provide. If it cannot be handled by PiCamera2 I guess I could find an algorythm to do it, but if PiCamera2 could do it same as in legacy, it would be a nicer approach.

Regards,
Juan

davidplowman, 2022-09-23:
Hi, thanks for the question. Indeed I don't think there's any way the hardware can do that for you. I don't think the colour matrices have enough range, but I guess that's what you've been trying. Likewise the gamma function has to be strictly increasing, so that won't do it either.

You certainly have some options using either OpenCV or numpy. I think functions like `numpy.invert` or `cv2.bitwise_not` might do it for you in just a single line, you could try and see if one is more efficient that the other!

jareff-g, 2022-09-23:
Hello,

Thanks for the quick answer. PiCamera legacy is able to apply a negative transformation in real time, so that the effect is visible even in preview mode, this is why I was hoping it could be possible with PiCamera2 as well.

I tried your suggested solution with numpy and it works (thanks!), so I'll use it. It takes a bit of processing time, but since the effect is not mandatory in real time, it will do.

Thanks again, closing the issue now.

closed 2022-09-23 by jareff-g