raspberrypi/picamera2 #335: [BUG] Using two instances on separate interfaces simultaneously can sometimes fail with "[0:38:03.430892329] [2855] ERROR Camera camera.cpp:533 Camera in Configured state trying queueRequest() requiring state Running"

CorvetteCole, 2022-09-30:
**Please only report one bug per issue!**

**Describe the bug**
If you initialize two picamera2 objects on different interfaces, then attempt to take pictures with them back-to-back without closing them in-between, they can sometimes fail:
```
[0:38:03.430892329] [2855] ERROR Camera camera.cpp:533 Camera in Configured state trying queueRequest() requiring state Running
```

**To Reproduce**
1. Spawn two picamera2 objects on separate interfaces
2. start object one, take a picture, then stop it (but don't explicitly call close())
3. attempt to start object 2 

-- OR --

```python
from picamera2 import Picamera2

def take_pictures(should_close_between=False):
    cam1 = Picamera2(0)
    cam2 = Picamera2(1)

    print('camera 1 seq')
    config1 = cam1.create_still_configuration()
    cam1.configure(config1)
    cam1.start()
    cam1.capture_file("sim_test_c1.jpg")
    cam1.stop()
    
    if should_close_between:
        print('closing camera 1')
        cam1.close()

    print('camera 2 seq')
    config2 = cam2.create_still_configuration()
    cam2.configure(config2)
    cam2.start()
    cam2.capture_file("sim_test_c2.jpg")
    cam2.stop()
    
    if not should_close_between:
        print('closing camera 1')
        cam1.close()
    print('closing camera 2')
    cam2.close()

print("########## taking two photos, this will work")
take_pictures(should_close_between=True)
print("########## taking two photos, this will fail")
take_pictures()
```

**Expected behaviour**
It just works™ without having to close the object in between

**Console Output, Screenshots**
```
❯ python picamera2_sequential_test.py
########## taking two photos, this will work
[0:37:59.038259411] [2845]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3866-0c55e522
[0:37:59.069626442] [2846]  WARN RPI raspberrypi.cpp:1258 Mismatch between Unicam and CamHelper for embedded data usage!
[0:37:59.070797109] [2846]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@0/imx219@10 to Unicam device /dev/media2 and ISP device /dev/media0
[0:37:59.078736837] [2846]  WARN RPI raspberrypi.cpp:1258 Mismatch between Unicam and CamHelper for embedded data usage!
[0:37:59.079869542] [2846]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media4 and ISP device /dev/media1
camera 1 seq
[0:37:59.085781209] [2845]  INFO Camera camera.cpp:1035 configuring streams: (0) 3280x2464-BGR888
[0:37:59.086328219] [2846]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@0/imx219@10 - Selected sensor format: 3280x2464-SBGGR10_1X10 - Selected unicam format: 3280x2464-pBAA
closing camera 1
camera 2 seq
[0:38:00.432455763] [2851]  INFO Camera camera.cpp:1035 configuring streams: (0) 3280x2464-BGR888
[0:38:00.432982440] [2846]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 3280x2464-SBGGR10_1X10 - Selected unicam format: 3280x2464-pBAA
closing camera 2
########## taking two photos, this will fail
camera 1 seq
[0:38:01.833862282] [2853]  INFO Camera camera.cpp:1035 configuring streams: (0) 3280x2464-BGR888
[0:38:01.841378369] [2846]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@0/imx219@10 - Selected sensor format: 3280x2464-SBGGR10_1X10 - Selected unicam format: 3280x2464-pBAA
camera 2 seq
[0:38:02.607401706] [2855]  INFO Camera camera.cpp:1035 configuring streams: (0) 3280x2464-BGR888
[0:38:02.608323303] [2846]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 3280x2464-SBGGR10_1X10 - Selected unicam format: 3280x2464-pBAA
[0:38:03.430892329] [2855] ERROR Camera camera.cpp:533 Camera in Configured state trying queueRequest() requiring state Running
```

**Hardware :**
Compute Module 4, picamera2 0.3.3, bullseye, 2x imx219

**Additional context**
This doesn't always reproduce, but once it does it will consistently. I cannot figure out what makes that happen, seems very random.

CorvetteCole, 2022-09-30:
Pretty big issue for us at the moment, given that we can't consistently take automated images. Temporarily closing interface objects and re-initializing them for the time being but that should not be needed

chrisruk, 2022-09-30:
Is this with a compute module I assume?

CorvetteCole, 2022-09-30:
same issue. The below is what you were suggesting, right?

```python
from picamera2 import Picamera2

cam1 = Picamera2(0)
cam2 = Picamera2(1)

config1 = cam1.create_still_configuration()
config2 = cam2.create_still_configuration()

cam1.configure(config1)
cam2.configure(config2)

def take_pictures(should_close_between=False):

    print('camera 1 seq')
    cam1.start()
    cam1.capture_file("sim_test_c1.jpg")
    cam1.stop()
    
    if should_close_between:
        print('closing camera 1')
        cam1.close()

    print('camera 2 seq')
    cam2.start()
    cam2.capture_file("sim_test_c2.jpg")
    cam2.stop()
    
    if not should_close_between:
        print('closing camera 1')
        cam1.close()
    print('closing camera 2')
    cam2.close()

print("########## taking two photos, this will work")
take_pictures(should_close_between=True)
print("########## taking two photos, this will fail")
take_pictures()
```

CorvetteCole, 2022-09-30:
Actually I think it is a slightly different issue since I'm pretty sure the Picamera2 class isn't meant to be started after being closed, it needs to be re-initialized. But that isn't really the point this is just a simple reproduction script, problem presents itself in much more complex software

chrisruk, 2022-09-30:
Sorry just edited my comment with what I was thinking at the same time as you.

What you have looks the same as what I was thinking.

I don't have compute module to test currently, will see if I can get one.

CorvetteCole, 2022-09-30:
Appreciate it, if you have any further questions let me know. I will dig in to the library a bit myself if I have time

chrisruk, 2022-09-30:
Cheers, will do, one other idea

```
from picamera2 import Picamera2

cam1 = Picamera2(0)
config1 = cam1.create_still_configuration()
cam1.configure(config1)
cam1.start()

cam2 = Picamera2(1)
config2 = cam2.create_still_configuration()
cam2.configure(config2)
cam2.start()

def take_pictures():
    cam1.capture_file("sim_test_c1.jpg")   
    cam2.capture_file("sim_test_c2.jpg")

take_pictures()
take_pictures()
```

CorvetteCole, 2022-09-30:
That files with a completely different error, probably stemming from the same bug though:
```
[3:10:29.493452382] [4627]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3866-0c55e522
[3:10:29.524751075] [4628]  WARN RPI raspberrypi.cpp:1258 Mismatch between Unicam and CamHelper for embedded data usage!
[3:10:29.525897872] [4628]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@0/imx219@10 to Unicam device /dev/media2 and ISP device /dev/media0
[3:10:29.533908598] [4628]  WARN RPI raspberrypi.cpp:1258 Mismatch between Unicam and CamHelper for embedded data usage!
[3:10:29.535054025] [4628]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media4 and ISP device /dev/media1
[3:10:29.538704726] [4627]  INFO Camera camera.cpp:1035 configuring streams: (0) 3280x2464-BGR888
[3:10:29.539237903] [4628]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@0/imx219@10 - Selected sensor format: 3280x2464-SBGGR10_1X10 - Selected unicam format: 3280x2464-pBAA
[3:10:29.678599583] [4627]  INFO Camera camera.cpp:1035 configuring streams: (0) 3280x2464-BGR888
[3:10:29.679109631] [4628]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 3280x2464-SBGGR10_1X10 - Selected unicam format: 3280x2464-pBAA
[3:10:31.360322700] [4635] ERROR Camera camera.cpp:1132 Request was not created by this camera
Traceback (most recent call last):
  File "/home/pi/cubelab_cdh/picamera2_sequential_test.py", line 19, in <module>
    take_pictures()
  File "/home/pi/cubelab_cdh/picamera2_sequential_test.py", line 15, in take_pictures
    cam1.capture_file("sim_test_c1.jpg")   
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1047, in capture_file
    return self._execute_or_dispatch(partial(self.capture_file_, file_output, name, format=format),
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1043, in _execute_or_dispatch
    return self.wait()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 995, in wait
    result = self._future.result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 440, in result
    return self.__get_result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 389, in __get_result
    raise self._exception
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 958, in process_requests
    done, result = function()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1024, in capture_file_
    request.save(name, file_output, format=format)
  File "/usr/lib/python3/dist-packages/picamera2/request.py", line 154, in save
    return self.picam2.helpers.save(self.make_image(name), self.get_metadata(), file_output, format)
  File "/usr/lib/python3/dist-packages/picamera2/request.py", line 150, in make_image
    return self.picam2.helpers.make_image(self.make_buffer(name), self.picam2.camera_config[name], width, height)
  File "/usr/lib/python3/dist-packages/picamera2/request.py", line 134, in make_buffer
    with _MappedBuffer(self, name) as b:
  File "/usr/lib/python3/dist-packages/picamera2/request.py", line 17, in __init__
    self.__fb = request.request.buffers[stream]
KeyError: <libcamera._libcamera.Stream object at 0xb3b939e0>
```

davidplowman, 2022-10-03:
Hi, this isn't going to work at the moment. The only supported mode of operation for multiple cameras is to run them in separate processes. I think the main stumbling block is that libcamera gives all the completed frames (for all cameras) to whichever Picamera2 asks first. So clearly there's another "dispatch" layer that needs to be implemented to route the right requests to the right Picamera2 instance.

davidplowman, 2022-10-04:
Yes, this is in my plans. I'm busy with other stuff at the moment, and also on holiday at some point, so my guess would be that it's "a few weeks away".

davidplowman, 2022-10-27:
Just FYI this is being worked on (https://github.com/raspberrypi/picamera2/pull/376), though it still has a little way to go.

CorvetteCole, 2022-10-27:
Exciting, thank you! If you need any help with anything let me know

davidplowman, 2022-11-02:
Hi, we've actually pushed some code for this onto our "next" branch. If you're in a position to try out that code we'd be interested to know if it works for you. Thanks.

CorvetteCole, 2022-11-02:
That is great news... I took a look at the relevant commits. I'll see if I have time to give it a go this Friday, but if not I'll make sure to get it tested sometime next week.

Just a note, our application for this is entirely headless, so anything related to previews will not tested at all.

CorvetteCole, 2022-11-04:
Were tests introduced to ensure there are no regressions for this in the future? I see this commit https://github.com/raspberrypi/picamera2/commit/86b2981aa196ba3858cf138d7f2318c4269cc247 which I think would test this functionality (multi-threading specifically), but wasn't sure if it was intended to test this specifically

davidplowman, 2022-11-07:
There are some tests appearing for this, but I don't have any test boards with multiple cameras at the moment so they won't run automatically. Actually I probably want to make it easier to discover how many and what cameras are attached, then I can make this decision happen dynamically. So we're working on it!!

davidplowman, 2022-11-10:
This should now be available in the 0.3.6 release, so I'll close this issue. Please let us know of any problems!

closed 2022-11-10 by davidplowman