raspberrypi/picamera2 #336: [HOW-TO] camera stuck waiting for image

FattNerd, 2022-10-02:
**Please only ask one question per issue!**

**Describe what it is that you want to accomplish**
I use the camera in a python function to capture QRCODES.
first time the function is called, it works fine and when I close the Loop I use: 
picam.stop()
picam.close()

But when I call again the function in the same script, the camera starts but stay stuck when trying to get array from it:
im = camera.capture_array()

stuck forever until I CTRL+C where it tells me that the error is:

  File "/home/pi/camera.py", line 271, in <module>
    im = camera.capture_array()
  File "/usr/local/lib/python3.9/dist-packages/picamera2/picamera2.py", line 1188, in capture_array
    return self._execute_or_dispatch(partial(self.capture_array_, name), wait, signal_function)
  File "/usr/local/lib/python3.9/dist-packages/picamera2/picamera2.py", line 1043, in _execute_or_dispatch
    return self.wait()
  File "/usr/local/lib/python3.9/dist-packages/picamera2/picamera2.py", line 995, in wait
    result = self._future.result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 435, in result
    self._condition.wait(timeout)
  File "/usr/lib/python3.9/threading.py", line 312, in wait
    waiter.acquire()

Do you know or someone has got this issue already before? 

**Describe alternatives you've considered**
Tried to setup special settings back the the camera but seems that nothing is changing here
**Additional context**
Add any other context or screenshots about the request here.

davidplowman, 2022-10-03:
Hi, could you perhaps include a small self-contained test case that demonstrates the problem? That would help to understand exactly what you're doing. Thanks!

FattNerd, 2022-10-04:
this is the message from the camera after 30 min waiting:
[1:01:24.095017947] [11149]  INFO Camera camera.cpp:1029 configuring streams: (0) 640x480-XBGR8888
[1:01:24.096330720] [11146]  INFO RPI raspberrypi.cpp:763 Sensor: /base/soc/i2c0mux/i2c@1/arducam_64mp@1a - Selected sensor format: 1280x720-SBGGR10_1X10 - Selected unicam format: 1280x720-pBAA
[1:03:46.600185911] [11146]  WARN V4L2 v4l2_videodevice.cpp:1967 /dev/video0[35:cap]: Dequeue timer of 142044000.00us has expired!
[1:03:46.600385482] [11146] ERROR RPI raspberrypi.cpp:1895 Unicam has timed out!
[1:03:46.600417704] [11146] ERROR RPI raspberrypi.cpp:1896 Please check that your camera sensor connector is attached securely.
[1:03:46.600440444] [11146] ERROR RPI raspberrypi.cpp:1897 Alternatively, try another cable and/or sensor.

maybe camera.close() is not freeing the bus of the camera. Any idea how to do that ?

Thank you

davidplowman, 2022-10-04:
Hi, thanks for the update. Is there any way you can simplify it that so that it just contains the camera code? I don't have pygame installed, and it would be easier if the script has no dependencies on the barcodes or anything else. Thanks!

In the absence of that, the obvious things to check for would be the script stopping the camera (and asking for another frame), or possibly camera lock-ups caused by long cables (I would consider 0.5m to be "long").

FattNerd, 2022-10-04:
Hi

Without pygame:

from pyzbar import pyzbar
from picamera2 import Picamera2, CameraConfiguration, Preview
import re

camera = Picamera2()
video_config = camera.create_video_configuration({"size": (640, 480)})
camera.start()
refocus = 0

while loop:
				im = camera.capture_array()
				gray = cv2.cvtColor(im, cv2.COLOR_BGR2GRAY)
				frame = gray
				barcodes = pyzbar.decode(frame)
				for barcode in barcodes:
								x, y , w, h = barcode.rect
								barcode_info = barcode.data.decode('utf-8')
								cv2.rectangle(frame, (x, y),(x+w, y+h), (0, 255, 0), 2)
								font = cv2.FONT_HERSHEY_DUPLEX
								cv2.putText(frame, barcode_info, (x + 6, y - 6), font, 2.0, (255, 255, 255), 1)
								s = barcode_info
								substring = re.search(r'.*?\.com/(.*)\?utm.*', s)
								if substring.group(1) in taglist:
												print("already in list")
								else:

												taglist.append(substring.group(1))

												loop = False
												camera.stop()
												camera.close()

				refocus = refocus + 1
				if refocus == 80:
								refocus = 0
								camera.set_controls({"AfTrigger": 0})

davidplowman, 2022-10-04:
Thanks for the new version. Would you be able to fix the formatting so that I can copy and paste it, please? Thank you.

FattNerd, 2022-10-05:
Hi: 

This is a straight run program to test it and reproduce exactly the problem I am having:
[testcamera.txt](https://github.com/raspberrypi/picamera2/files/9714211/testcamera.txt)

```
"""
sudo apt install -y python3-kms++
sudo apt install -y python3-pyqt5 python3-prctl libatlas-base-dev ffmpeg
sudo pip3 install numpy --upgrade
sudo pip3 install picamera2==0.3.2
pip install pyzbar
pip install regex
pip install opencv-python
"""

from pyzbar import pyzbar
from picamera2 import Picamera2, CameraConfiguration, Preview
import re
import cv2

taglist = []
choice = "y"
camera = Picamera2()
video_config = camera.create_video_configuration({"size": (640, 480)})
camera.start()

while choice == "y":
  loop = True
  refocus = 0

  choice = input("one more scan? (y/n): ")

  while loop:
    im = camera.capture_array()
    frame = cv2.cvtColor(im, cv2.COLOR_BGR2GRAY)
    barcodes = pyzbar.decode(frame)
    print(frame)
    for barcode in barcodes:
      x, y , w, h = barcode.rect
      barcode_info = barcode.data.decode('utf-8')
      cv2.rectangle(frame, (x, y),(x+w, y+h), (0, 255, 0), 2)
      font = cv2.FONT_HERSHEY_DUPLEX
      cv2.putText(frame, barcode_info, (x + 6, y - 6), font, 2.0, (255, 255, 255), 1)
      s = barcode_info
      substring = re.search(r'.*?\.com/(.*)\?utm.*', s)
      if substring.group(1) in taglist:
        print("already in list")
      else:
        taglist.append(substring.group(1))
        loop = False
        camera.stop()
        camera.close()
        print(substring.group(1))
    refocus = refocus + 1
    if refocus == 80:
      refocus = 0
      camera.set_controls({"AfTrigger": 0})

```
thank you

davidplowman, 2022-10-05:
Hi, thanks for posting the new version. I went through the code and deleted everything that was not related to the camera. I replaced the `while choice == "y":` by `for i in range(2):` just to make it go round the loop twice. Anyway, I ended up with this:
```
from picamera2 import Picamera2

camera = Picamera2()
camera.start()

for i in range(2):
    im = camera.capture_array()
    camera.stop()
    camera.close()
```
Could you confirm that this looks equivalent in terms of the precise calls being made to the camera system? Thanks!

FattNerd, 2022-10-05:
Hi

Yes it does look similar.

On another level, I found a topic on stack overflow that mentioned that once the camera is closed, you need to open a new instance but it was not stable. (in my case once the camera is closed it doesn't want to open new instances as camera bus is busy ...)

I have corrected my code by not closing the camera.Just changing the configuration depending on settings required and using camera.start() and camera.stop() I manage to make it work (for my use)
camera.close() has to be used only if you hope to create a new camera instance later on in your code. But even for this I found it buggy sometimes.

davidplowman, 2022-10-05:
Generally speaking I wouldn't be recreating and closing the Picamera2 object if you can help it, nor would I be re-configuring the camera every time as that only de- and re-allocates a ton of memory that carries slight risks of failure depending on whatever else the system is doing. Similarly, maybe you don't even need to stop and re-start the camera every time (it's actually quite slow) - but that's up to you and your application of course!

If you do want to shutdown and restart the camera system the `with` idiom is often quite nice, so I'd consider something like (taking my simplified example)
```
from picamera2 import Picamera2

with Picamera2() as camera:
    camera.start()
    for i in range(2):
        im = camera.capture_array()
        # lots more code...
```
Also I wanted to mention that you created a configuration but never actually applied it. You probably wanted a `camera.configure(video_config)` after `video_config = ...`.

Lastly, if you have cases that are not working properly and can create nice simple examples like I have done, then please post them. Thanks!

FattNerd, 2022-10-05:
Thank you so much for your Help David, I will apply your method and leave the camera open all along the system. It is a first time for me using Picamera, I worked with webcam before most of the time.

Thank you so much

closed 2022-10-05 by FattNerd

davidplowman, 2022-10-17:
Hi, we have done some work in this area though it's not available in Picamera2 yet. However I will look into it when I have some time though I have a number of other things ahead of this in the queue.

Having said that, I should add that I'm very nervous about this sort of approach. If there is a problem with signal integrity, then all I can say is that "all bets are off". Almost anything could happen. A camera lock-up is just one (probably a typical) symptom. But the lock-ups might be "recoverable", or they might not be, or you might get a mixture. You might not get lock-ups but corrupt images (by which I mean complete garbage), I would expect this to be quite common as well. I'll see what I can do, but you're in a much better situation if you don't have random camera failures.

Simplecoding16, 2023-08-17:
Hello, 

I am having the same issue now when trying to capture and image with capture_array() getting stuck and freezing up my app

Thanks

davidplowman, 2023-08-18:
Is the system reporting camera timeouts? What kind of camera are you using, and what kind of Pi? How long is your ribbon cable? Please can you include any console error messages that you are seeing. It's also helpful if you can include code for the most minimal example that demonstrates the problem if you can - I would usually expect this to be half a dozen to a dozen lines of code, or thereabouts. Thank you.

Simplecoding16, 2023-08-19:
Hi @davidplowman, no, the system is not reporting any timeouts. The program freezes and,  similar to the original post, I have to use a keyboard interrupt to get the same waiter.acquire() error. 

For context, I have established a simple pyqt5 app showing a live Arducam preview. One the app, I have a toggle that when checked is supposed to capture_array() from the camera and begin edge detection of the capture array. When the code gets to capturing the array, it freezes up. I have tried to close the camera and create a new camera instance before imaging, I have also tried to switch over to cv.VideoCapture but comes back with a blank image 

List of equipment and packages 

- Waveshare CM4-ID-BaseB
- raspberry pi bullseye 64bit 
- Arducam IMX477 
 - Python 3.9 
 - OpenCV-Headless v3.6.8
 - pyqt5 v5.15.12
 - Picamera2

`class class1(QWidget):
    # Imports the image from Picamera Preview 

    def __init__(self):
        super().__init__()

    def theThing(self):

        # Capture an image array for circle detection
        img = picamera.capture_array("main")
       
       #rest of code for OpenCV to use. 

  
   def do_something_toggled(self, checked):
       if checked:

            # Run the theThing from class1
            class1.theThing(self)

        else:
            # rest of code `

davidplowman, 2023-08-21:
Hi, the first things I would do are:

1. Does `libcamera-hello` work correctly? This is the very first thing to check.
2. Does [this](https://github.com/raspberrypi/picamera2/blob/main/examples/capture_full_res.py) simple Picamera2 app work correctly?
3. Does [this](https://github.com/raspberrypi/picamera2/blob/main/apps/app_capture2.py) simple Picamera2 Qt app work correectly?
4. Have you read section 8.5 of the [manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf)? It wasn't clear to me whether you were using non-blocking calls correctly.

Simplecoding16, 2023-08-21:
Hi @davidplowman , thanks for your quick response. I have tried all 3 test codes, and they all work on my end. Additionally, I have gone through the documentation of PiCamera and tried to implement the signal_function.connect solve that was posted [here](https://github.com/raspberrypi/picamera2/issues/766). by using: 

`img= capture.array('main`, signal_function = qpicamera2.signal_done)`

This did get me past this line of code, as I added a print statement afterward to doublecheck, but I am unsure if this took an actual image since I am getting an AttributeError: 'Job' object has no attribute 'shape' when the image is supposed to be passed into the openCV edge detection function as shown below. 

```
        # Capture an image array for circle detection
        img = picam2a.capture_array("main", signal_function = qpicamera2a.signal_done)
       print('array taken')

        # Extract the dimensions of the image (height, width, and channels)
        height, width, channels = img.shape **<---- THIS IS WHERE THE ERROR IS RAISE**

        # Resize image for analysis
        imgReSize = cv.resize(img, (2028, 1520))

```

davidplowman, 2023-08-21:
I'm afraid I'm struggling a bit to understand what code you're actually running. If you were able to create a _minimal_ example - preferably no more than a dozen or so lines of code - that demonstrates the problem, that would be helpful.

The only thing I would add is that when you can capture_array asynchronously, you don't get back an image immediately. You have to wait for control to be returned to you through the Qt event loop, or wait for the result. As I say, it depends rather what your code is trying to do... Thanks!

Simplecoding16, 2023-08-23:
Hi @davidplowman, 

I have adapted your Picamera2 Qt app example to show my issue. The code is below

Let me know if you have further questions

`from PyQt5 import QtCore
from PyQt5.QtWidgets import (QApplication, QHBoxLayout, QLabel, QPushButton,
                             QVBoxLayout, QWidget)

from picamera2 import Picamera2
from picamera2.previews.qt import QGlPicamera2

def post_callback(request):
    label.setText(''.join(f"{k}: {v}\n" for k, v in request.get_metadata().items()))

picam2 = Picamera2()
picam2.post_callback = post_callback
picam2.configure(picam2.create_preview_configuration(main={"size": (800, 600)}))

app = QApplication([])

def on_button_clicked():
    button.setEnabled(False)

    # Capture an image array for openCV
    picam2.capture_array("main", signal_function=qpicamera2.signal_done)
    print('picam2a captured array')

    # Extract the dims for openCV
    height, width, channels = img.shape  # Attribution error raised: Job object has no attribute 'shape'

def capture_done(job):
    picam2.wait(job)
    button.setEnabled(True)

qpicamera2 = QGlPicamera2(picam2, width=800, height=600, keep_ar=False)
button = QPushButton("Click to capture JPEG")
label = QLabel()
window = QWidget()
qpicamera2.done_signal.connect(capture_done)
button.clicked.connect(on_button_clicked)

label.setFixedWidth(400)
label.setAlignment(QtCore.Qt.AlignTop)
layout_h = QHBoxLayout()
layout_v = QVBoxLayout()
layout_v.addWidget(label)
layout_v.addWidget(button)
layout_h.addWidget(qpicamera2, 80)
layout_h.addLayout(layout_v, 20)
window.setWindowTitle("Qt Picamera2 App")
window.resize(1200, 600)
window.setLayout(layout_h)

picam2.start()
window.show()
app.exec()`

davidplowman, 2023-08-23:
Hi, thanks for posting the code.

I would guess that this:
```
def on_button_clicked():
    button.setEnabled(False)

    # Capture an image array for openCV
    picam2.capture_array("main", signal_function=qpicamera2.signal_done)
    print(' Captured array') # A check to make sure the capture array happened

    # Extract the dims for openCV
    height, width, channels = img.shape  # Attribution error raised: Job object has no attribute 'shape'

def capture_done(job):
    picam2.wait(job)
    button.setEnabled(True)
```
needs to look like this:
```
def on_button_clicked():
    button.setEnabled(False)

    # Capture an image array for openCV
    picam2.capture_array("main", signal_function=qpicamera2.signal_done)

def capture_done(job):
    img = picam2.wait(job)
    print(' Captured array') # A check to make sure the capture array happened

    # Extract the dims for openCV
    height, width, channels = img.shape  # Attribution error raised: Job object has no attribute 'shape'

    button.setEnabled(True)
```
The reason is that once `capture_array` returns, the capture hasn't (necessarily) happened yet. You 
have to wait until control is returned to you in the function that you have connected to the Qt signal.

Simplecoding16, 2023-08-24:
Hi @davidplowman , this worked, but only in this format. When implemented in a class, I recieve a NameError: name 'job' is not defined. Have you seen this before?

davidplowman, 2023-08-24:
I'm afraid it's difficult for me to understand what "when implemented in a class" means, can you be more precise? Usually, it is helpful if a minimal (that is, as small as possible) code example can be submitted that demonstrates the problem.

davidplowman, 2023-08-24:
Thanks for the code.

To make it work I found I had to move `self.setup_ui()` to just below the line `self.qpicamera2 = ...`. Once I'd done this, it seemed to work as expected. Are you able to try that?