raspberrypi/picamera2 #337: [BUG] Exception: EGL_EXT_image_dma_buf_import missing

shaja-asm, 2022-10-03:
Board: Raspberry Pi 3B+
Camera: Raspberry Pi NoIR Camera V2
Everything is up to date.
Code: [https://github.com/raspberrypi/picamera2/blob/d4b22e80e5c23676eafe06120cfe0df71aad0fc1/examples/capture_png.py](url)

```
#!/usr/bin/python3

# Capture a PNG while still running in the preview mode.

import time

from picamera2 import Picamera2, Preview

picam2 = Picamera2()
picam2.start_preview(Preview.QTGL)

preview_config = picam2.create_preview_configuration(main={"size": (800, 600)})
picam2.configure(preview_config)

picam2.start()
time.sleep(2)

picam2.capture_file("test.png")
```

When I tried to run capture_png.py from the examples and I got the above error. Here is the console output.

```
[1:47:15.317425717] [7130]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3949-ed591e70-dirty (2022-10-03T10:13:27+05:30)
[1:47:15.729729289] [7133]  WARN RPI raspberrypi.cpp:1263 Mismatch between Unicam and CamHelper for embedded data usage!
[1:47:15.733908484] [7133]  INFO RPI raspberrypi.cpp:1379 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
libEGL warning: DRI2: failed to authenticate
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3/dist-packages/picamera2/previews/qt_previews.py", line 22, in thread_func
    self.qpicamera2 = self.make_picamera2_widget(picam2, width=self.width, height=self.height, transform=self.transform)
  File "/usr/lib/python3/dist-packages/picamera2/previews/qt_previews.py", line 75, in make_picamera2_widget
    return QGlPicamera2(picam2, width=self.width, height=self.height, transform=self.transform)
  File "/usr/lib/python3/dist-packages/picamera2/previews/q_gl_picamera2.py", line 101, in __init__
    self.egl = EglState()
  File "/usr/lib/python3/dist-packages/picamera2/previews/q_gl_picamera2.py", line 32, in __init__
    self.choose_config()
  File "/usr/lib/python3/dist-packages/picamera2/previews/q_gl_picamera2.py", line 50, in choose_config
    check_egl_extensions(self.display, ["EGL_EXT_image_dma_buf_import"])
  File "/usr/lib/python3/dist-packages/picamera2/previews/gl_helpers.py", line 71, in check_egl_extensions
    raise Exception(ext + " missing")
Exception: EGL_EXT_image_dma_buf_import missing

```

**Additional context**

- I'm able to capture camera output using qcam app `sudo qcam`
- I'm unable to run basic libcamera functions either. Here is the camera bug report for the `camera-bug-report -t 5 -o bug.txt -c "libcamera-hello -v"`
 [bug.txt](https://github.com/raspberrypi/picamera2/files/9695594/bug.txt)
- `libcamera-hello -t 0 --qt-preview` gives a preview window. But `libcamera-hello -t 0` gives the following error
```
libEGL warning: DRI2: failed to authenticate
Made X/EGL preview window
[2:14:44.519406501] [7770]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3949-ed591e70-dirty (2022-10-03T10:13:27+05:30)
[2:14:44.738688278] [7785]  WARN RPI raspberrypi.cpp:1263 Mismatch between Unicam and CamHelper for embedded data usage!
[2:14:44.745319937] [7785]  INFO RPI raspberrypi.cpp:1379 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
[2:14:44.750219008] [7770]  INFO Camera camera.cpp:1026 configuring streams: (0) 1640x1232-YUV420
[2:14:44.753027108] [7785]  INFO RPI raspberrypi.cpp:766 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to import fd 21
Aborted
```

davidplowman, 2022-10-03:
Hi, there are a couple of things to check first.

1. Have you enabled Glamor? Please check the [Picamera2 manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf), sections 2.3 and 3.2.
2. Are you logged in over ssh using X forwarding? If so, the GL-accelerated preview implementation will not work. Please check section 3.4 of the manual.

Thanks!

shaja-asm, 2022-10-03:
1) Yes. Glamor is enabled.
2) No. I'm not logging over ssh. I have a separate monitor connected to Raspberry Pi.

davidplowman, 2022-10-03:
Do you have a spare SD card on which you can download the latest image using the Raspberry Pi imager tool? Let it apply all its updates when it first boots and then enable Glamor (and reboot again). Can you confirm that this is working correctly? If it does, then then problem becomes understanding how your image is different. Thanks!

shaja-asm, 2022-10-03:
I've been struggling with this issue for more than a week. My current OS is based on the image I flashed on last Friday using Pi Imager tool. I don't have a spare SD card but I can re-flash my current SD card since I don't have any important data on it. It's still fresh.
ImageDetails:
Raspberry Pi OS (64-bit)
Release date: September 22nd 2022
System: 64-bit
Kernel version: 5.15
Debian version: 11 (bullseye)
Size: 783MB

```
piuser@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

piuser@raspberrypi:~ $ uname -a
Linux raspberrypi 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux
```

davidplowman, 2022-10-03:
I've just been round the loop again. I've installed the latest 64-bit OS using the Raspberry Pi imager tool, booted my Pi 3B and let it perform its updates. Then I enabled Glamor and rebooted. Finally I cloned the picamera2 repository and ran the picamera/examples/capture_png.py example and it worked perfectly. I am logged into the Pi directly with a screen, keyboard and mouse.

Is there anything different that you're doing? For the avoidance of doubt here are my software versions:
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux
pi@raspberrypi:~ $ vcgencmd version
Aug 26 2022 14:04:36 
Copyright (c) 2012 Broadcom
version 102f1e848393c2112206fadffaaf86db04e98326 (clean) (release) (start)
```
This certainly looks like yours. Is there anything else you've installed or altered (perhaps in /boot/config.txt, for example)?

shaja-asm, 2022-10-04:
I have installed libcamera from [https://libcamera.org/getting-started.html](url) before cloning picamera2. It was mentioned as a requirement in picamera2 manual [https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf](url) Chapter 2.1. Is that can be a reason for the issue?

davidplowman, 2022-10-04:
Hi again, those links in your post appear to be broken. Nonetheless you should _not_ be installing libcamera yourself, it is already installed in the image. Can you please try again and follow the same instructions that I did, which really involved nothing other than installing the image as normal and enabling Glamor. Thanks.

shaja-asm, 2022-10-06:
I re-flashed my sd card. But libcamera was not there. I was getting command not found message. Therefore I built libcamera myself. I also increased the GPU memory to 256 and set he GL driver to `GL (Full KMS)` in advanced options of `raspi-config`. Now it's working fine. Thank you very much for the support.

closed 2022-10-06 by shaja-asm

dhaval-khatri1996, 2022-10-14:
Facing the same issue. Raspberry pi 4 Bullseye 32 bit.

dhaval-khatri1996, 2022-10-16:
@davidplowman @shaja-asm, I am facing the same issue

davidplowman, 2022-10-17:
@dhaval-khatri1996 Hi, please you you report

* Exactly what kind of Pi you have.
* What camera are you using?
* Exactly what OS you are running - full, Lite, something else? If you have not tried the latest image as reported by the Raspberry Pi Imager tool, then please do so.
* Have you made any modifications to your /boot/config.txt file?
* Are you using the standard versions of libcamera and Picamera2 installed with our OS image or have you installed them from anywhere else?
* Are you running with X windows? Or from the system console?
* Are you logged into your Pi directly with a screen and keyboard?
* Or are you logging in using ssh? If you are using ssh, are you using X forwarding?
* Or are you using VNC?

Thanks!

davidplowman, 2022-10-17:
@shaja-asm Hi again, glad to hear it's working. I just wanted to add that you might want to try running _without_ setting GPU mem to 256. libcamera and Picamera2 do not use GPU mem, so all this does is reserve 256MB of memory... that no one will ever use and will therefore be completely wasted. So your system should simply run better if you remove this line from your /boot/config.txt.

dhaval-khatri1996, 2022-10-17:
@davidplowman 

- Raspberry pi 4 B
- wave share ov9281-110
- Bullseye full version latest 
- Yes, camera_auto_detect=1, dtoverlay=ov9281
- Standard version
- Tried both from terminal using putty/ssh and uisng xrdp with remote desktop connection

dhaval-khatri1996, 2022-10-17:
@davidplowman  I think this discussion might point out the issue, please take a look [here](https://github.com/kbingham/libcamera/discussions/56)