raspberrypi/picamera2 #341: [BUG] python3-libcamera not available in venv

mnbf9rca, 2022-10-04:
**Describe the bug**
When running from a venv, libcamera is not available. It works fine when not in venv.

**To Reproduce**
1. fresh Raspbian 32bit bullseye installation, after running both `apt-get dist-upgrade` and `apt-get upgrade`
2. create a venv e.g. `python3 -m venv venv`
3. activate `source ./venv/bin/activate`
4. install build dependencies: `sudo apt-get install -y libpcap-dev python3-libcamera`
5. upgrade pip to v22: `sudo python3 -m pip install --upgrade pip` 
6. install picamera2 in venv: `pip install picamera2`
7. copy the `mjpeg_server.py` demo to local machine: `wget https://raw.githubusercontent.com/raspberrypi/picamera2/main/examples/mjpeg_server.py`
8. run the server: `python3 mjpeg_server.py`

**Expected behaviour**
Streaming server should start

**Console Output, Screenshots**
```python
(venv) pi@mariner:~ $ python3 mjpeg_server.py
Traceback (most recent call last):
  File "/home/pi/mjpeg_server.py", line 13, in <module>
    from picamera2 import Picamera2
  File "/home/pi/venv/lib/python3.9/site-packages/picamera2/__init__.py", line 1, in <module>
    from .configuration import CameraConfiguration, StreamConfiguration
  File "/home/pi/venv/lib/python3.9/site-packages/picamera2/configuration.py", line 1, in <module>
    from .controls import Controls
  File "/home/pi/venv/lib/python3.9/site-packages/picamera2/controls.py", line 4, in <module>
    from libcamera import ControlType, Size, Rectangle
ModuleNotFoundError: No module named 'libcamera'
```

**Hardware :**
Camera: original Raspberry PI NoIR camera

PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

**Additional context**
apt-get reports the library is already installed:
```bash
(venv) pi@mariner:~ $ sudo apt-get install python3-libcamera
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-libcamera is already the newest version (0~git20220826+3fad116f-1).
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

davidplowman, 2022-10-05:
@chrisuk Hi Chris, just wondering if by any change you've ever used "venv"s? (I must confess that I know nothing about them...)

davidplowman, 2022-10-05:
Actually I've just done a web search. It seems you need to use the ``--system-site-packages`` option when you create the venv. Please see https://docs.python.org/3/library/venv.html.

chrisruk, 2022-10-05:
Sorry was afk, yep that's what I've used before, otherwise you'd have to install libcamera within the venv, which might be tricky as I don't think there's a PyPi version.

davidplowman, 2022-10-05:
@mnbf9rca I think `--system-site-packages` sounds like the answer. Does that resolve this for you? Thanks!

mnbf9rca, 2022-10-05:
Yes! I've never even noticed that option before. Thanks so much. Perhaps worth adding to the documentation?

closed 2022-10-05 by mnbf9rca

DennisSchulmeister, 2023-02-02:
Hello,

Just stumbling upon the same problem. While `--system-site-packages` works around the problem, this not a good solution. Because the very purpose of python environments is to be isolated from the system-installed packages. The real solution would in deed be a pypi version of the libcamera module pulled in by picamera2 as a regular dependency.

KR, Dennis

eya1903, 2024-03-11:
> Actually I've just done a web search. It seems you need to use the `--system-site-packages` option when you create the venv. Please see https://docs.python.org/3/library/venv.html.

I confirm, it works guys. Thank you saved me !

Wences1024, 2024-06-23:
I had the same issue D: !!!!  and I double confirm, it works by creating a new environment with "--system-site-packages". I am working on a project using openCV with the Raspberry camera.
These were the steps I followed:

#Create a virtual environment
python3 -m venv --system-site-packages new_env

#Activate the new environment
source new_env/bin/activate

#Update pip
pip install --upgrade pip

#Install mediapipe
pip install mediapipe

#Install OpenCV
pip install opencv-python

#Check if you have all the libraries available
pip list

You must see "picamera2" in the list when using "pip list"