raspberrypi/picamera2 #344: hi can anyone help me with finding the frame rate or the duration of a single frame in picamera2 I checked the documentation it says to use "FrameDuration" command but I dont get the exact format of using this. does anyone know?[HOW-TO]

faeq123, 2022-10-08:
**Please only ask one question per issue!**

**Describe what it is that you want to accomplish**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the request here.

Tim-Brown-NZ, 2022-10-08:
The metadata contains the "FrameDuration" in microseconds, so for 1/8 of a second the FrameDuration is 125,000.  
The exposure time in the attributes can be set to a value between 0 and 999999 so if you set this to 125,000 and then take a image, the image (or perhaps the next one) should have a FrameDuration of 125,000.

faeq123, 2022-10-08:
Ok but I have to use the picamera to detect drowsiness so if eye blinks exceed 3 seconds I need to alert. So for that if I know the frame duration or fps of the video I can set the threshold frames accordingly. So I need to get the actual frame duration of the video any idea how to do that?

Tim-Brown-NZ, 2022-10-08:
Sorry, I've used video very little.  The only thing I can think of is, what happens if you make an explicit capture_metadata() call inside your frame reading loop, do you get a sensible response?  Otherwise you might have to wait for David to come to work on Monday in the UK for an answer.

sandyol55, 2022-10-10:
The default video configuration will set a framerate of 30fps with the FrameDurationLimits of 33333,33333 (min and max microseconds per frame, so fixed at that one setting).

But you can force other durations within the limits of your particular sensor with for example:-

`config = picam2.create_video_configuration(controls={"FrameDurationLimits": (40000, 40000)})`

for 25fps,

or 

`config = picam2.create_video_configuration(controls={"FrameDurationLimits": (100000, 100000)})`

for 10 fps which would give the opportunity for longer exposure times per frame if in a low light ambient.

Taken from   **4.2.3.** _Configurations and runtime camera controls_ section of the manual.

faeq123, 2022-10-10:
Ok. thank you 😊

davidplowman, 2022-10-18:
Hi, thanks to everyone for providing such good answers here!

Yes, once the camera is running, `picam2.capture_metadata()["FrameDuration"]` is a good place to start. You also have `picam2.capture_metadata()["SensorTimestamp"]` which gives the time the frame was delivered measured in nanoseconds (!) since the system booted.

If there's anything else please let us know, otherwise I'll look at closing this issue shortly. Thanks!

davidplowman, 2022-10-21:
Please feel free to open up a new issue if anything else comes up. Thanks!

closed 2022-10-21 by davidplowman