raspberrypi/picamera2 #347: [SUGGESTION] Option to show fps, exposure, etc in preview window title, similar to libcamera-apps

vincentneo, 2022-10-09:
In libcamera-apps, preview windows have it's window title with live updating info, probably it would be nice to have a convenient way of having the same thing for QTGL previews here as well?

[image]
(Image from https://www.raspberrypi.com/documentation/accessories/camera.html)

davidplowman, 2022-10-17:
Hi, yes I think I wondered about that but maybe it would be a useful feature. In the meantime you could try this kind of thing as a workaround:
```
from picamera2 import Picamera2
from picamera2.previews import QtGlPreview

def callback(request):
    preview.qpicamera2.setWindowTitle(f"{request.get_metadata()['ExposureTime']}")

picam2 = Picamera2()
preview = QtGlPreview()
picam2.pre_callback = callback
picam2.start_preview(preview)
picam2.start()
```

davidplowman, 2022-10-21:
I've made some changes here: https://github.com/raspberrypi/picamera2/pull/369

It's not the same as libcamera-apps, but is quite flexible and easy to use.

davidplowman, 2022-10-24:
That PR has been merged now, and will become available in the next release. Alternatively, you can fetch the "next" branch and try it early. So I think I'll close this one now, and thanks again for the suggestion!

closed 2022-10-24 by davidplowman