raspberrypi/picamera2 #350: [SUGGESTION] Any plans to implement the picamera "annotate" functionality?

jaysonlarose, 2022-10-11:
The picamera library had support for "baking in" text to the output image via firmware using the [annotate_text](https://picamera.readthedocs.io/en/release-1.13/api_camera.html#picamera.PiCamera.annotate_text) property. Are there any plans to reimplement this ability in picamera2?

I use this functionality — as well as a little [brute force and ignorance](http://www.catb.org/jargon/html/B/brute-force-and-ignorance.html) — to apply timestamps with 1/100th second precision to video. I'd basically push the updated timestamp text to the camera module 100 times per second as part of my main event loop, and let the chips fall where they may as far as which frames received what timestamp.

I tried to reimplement the same thing using a technique similar to [this](https://github.com/raspberrypi/picamera2/blob/main/examples/timestamped_video.py) last night on the same Raspberry Pi 3 Model B, and CPU usage exceeded 100% and capture at 25+ frames per second was no longer possible.

One caveat with the above statement: I had to implement the code that writes the timestamp text to the current video frame by rendering via freetype and blitting via numpy, because raspberry pi os lite bullseye's gtk-3 is currently broken. Until that's fixed I won't know how performance of cv2's text rendering compares, but I'm concerned about the performance, as my example code running at 1280x720@30fps produced 5 minutes 5 seconds worth of frames during 10 minutes worth of recording and consumed 110% CPU while doing it; in contrast, the old annotate method worked fine at 1600x1200@30fps, consuming 15% cpu while also serving up the video stream using TLS encryption.

Here's the example code I cobbled together that uses freetype while the gtk3 packages are broken:

```python
#!/usr/bin/python3
import time

# cv2 disabled because:
#$ date
#Tue 11 Oct 2022 10:43:46 AM PDT

#$ printf "%s\n" "$( cat /proc/device-tree/model )"
#-bash: warning: command substitution: ignored null byte in input
#Raspberry Pi 3 Model B Rev 1.2

#$ lsb_release -a
#No LSB modules are available.
#Distributor ID: Raspbian
#Description:    Raspbian GNU/Linux 11 (bullseye)
#Release:        11
#Codename:       bullseye

#$ sudo apt install python3-opencv
#Reading package lists... Done
#Building dependency tree... Done
#Reading state information... Done
#Some packages could not be installed. This may mean that you have
#requested an impossible situation or if you are using the unstable
#distribution that some required packages have not yet been created
#or been moved out of Incoming.
#The following information may help to resolve the situation:
#
#The following packages have unmet dependencies:
# libgtk-3-0 : Depends: libwayland-client0 (>= 1.20.0) but 1.18.0-2~exp1.1 is to be installed
#E: Unable to correct problems, you have held broken packages.

#import cv2

import os

# NOTE: point this to a .ttf file that actually exists on your system!
FONT_FILE = os.path.join(os.environ['HOME'], ".local/share/fonts/alarm_clock_7mod.ttf")
FONT_SIZE = 16

from picamera2 import MappedArray, Picamera2
from picamera2.encoders import H264Encoder
from picamera2.outputs import FileOutput

# (blit() and paste() lifted from https://stackoverflow.com/questions/28676187/numpy-blit-copy-part-of-an-array-to-another-one-with-a-different-size)

import numpy as np

def blit(a, b, offsets=(0,), as_shapes=False):
    """
    Computes the slices of the overlapping regions of arrays <a> and <b>. If offsets are specified,
    <b> will be shifted by these offsets before computing the overlap.

    Example:
          50
       ┌──────┐
       │      │
     65│  ┌───┼────┐
       │  │   │    │50
[... 157 lines cut ...]
    #origin = (0, 30)
    #font = cv2.FONT_HERSHEY_SIMPLEX
    #scale = 1
    #thickness = 2

    face = freetype.Face(FONT_FILE)
    face.set_char_size(FONT_SIZE * 64)

    def apply_timestamp(request):
        timestamp_text = format_timestamp(now_tzaware(), precision=2)
        canvas = render_numpy(face, timestamp_text)
        timestamp = time.strftime("%Y-%m-%d %X")
        with MappedArray(request, "main") as m:
            #cv2.putText(m.array, timestamp, origin, font, scale, colour, thickness)
            paste(m.array, canvas, (10, 10), copy=False)

    picam2.pre_callback = apply_timestamp

    encoder = H264Encoder(10000000)

    picam2.start_recording(encoder, "test.h264")
    time.sleep(600)
    picam2.stop_recording()
```
I assume a Raspberry Pi 4 is performant enough to run this in real-time, but the Raspberry Pi 3 is not, so I'm concerned about the claim that the legacy interface is going to be removed while the new library isn't capable of doing the same thing.

davidplowman, 2022-10-17:
Hi, I've never been very happy with the Python libraries that I've found for rendering text onto images, though the least bad option seems to be OpenCV, which has always worked fine for me. If you're having trouble with this, then please report it on one of the forums, obviously it will never get fixed otherwise! Can you say if there is anything "unusual" about your configuration? Does it work on a clean install of the latest Bullseye image with no other modifications?

Another thing I contemplated was using PIL. This will render fonts easily enough but I could never get it to work "in place", so I had to copy out the piece of image where the text will go, write the text, and then copy it back. Might that work better?

jaysonlarose, 2022-10-18:
I didn't bother reporting it, because it seemed to be a pretty widespread issue: <https://forums.raspberrypi.com/viewtopic.php?t=340631>. I was trying to install on a brand new installation of Bullseye. Anyways, I gave things another go last night, this package problem is apparently sorted out, because I was able to get python3-opencv installed without a hitch this time around.

cv2.putText would appear to be a LOT more performant than my freetype + numpy blitting method. Processing a 1600x1200@30fps Raspberry Pi Camera 2 feed on my Raspberry Pi 3 model B resulted in an average CPU usage (as reported by `top`) of about 60% using cv2.putText.  Disabling the `camera.pre_callback` hook resulted in an average CPU usage of about 40%.  Nowhere near the 15% performance I got using the old legacy module and picamera, but the only things this raspberry pi needs to do is serve up video and process GPS data for an NTP server, so that's fine.

closed 2022-10-18 by jaysonlarose