raspberrypi/picamera2 #351: [BUG] capture_MJPEG no timestamp

tf4ong, 2022-10-12:
Hi,

I tried to capture_MJPEG example script for video, and added pts='timestamp.txt' in the start_recording similar to h264.

But the timestamp.txt was empty.

chrisruk, 2022-10-13:
Thanks for reporting this, will work on this soon.

In the meantime if you add the pts parameter in this example - https://github.com/raspberrypi/picamera2/blob/main/examples/capture_mjpeg_v4l2.py
you should get a timestamp.txt file.

chrisruk, 2022-10-13:
Thanks for pointing that out! That appears to be a bug too, it shouldn't be h264 data.

davidplowman, 2022-10-19:
I remember ages ago there was a problem where the mjpeg hardware encoder wasn't (always) getting included in the firmware builds, and it would fall back to h264. Might this be the same thing?

chrisruk, 2022-10-19:
Ah yeah it could well be that.  I checked that I am still setting V4L2_PIX_FMT_MJPEG.

When using 'file' on the output it says -
 test.mjpeg: JVT NAL sequence, H.264 video @ L 40

chrisruk, 2022-10-19:
Just found something interesting @davidplowman 

If I add start_x=1, to /boot/config.txt I do get -
test.mjpeg: JPEG image data, baseline, precision 8, 1280x720, components 3

I think that might of been the behaviour we saw before

davidplowman, 2022-10-19:
Yes, `start_x=1` will be the workaround. But I thought we'd sorted this out... evidently not. I'll have another look. Anyway, it's not a Picamera2 issue, it'll require an OS/firmware update, so no need to hold up our bugfix 0.3.5 release I think.

You can check the state of your image using `v4l2-ctl -d /dev/video11 --list-formats`. If that doesn't list MJPG then you'll end up with h264.

chrisruk, 2022-10-19:
Thanks, that's a useful command to know.

Added a pull request regarding timestamps for MJPEG here - https://github.com/raspberrypi/picamera2/pull/365

davidplowman, 2022-10-19:
Indeed there is still a problem enabling the MJPEG encoder in non-"start_x" builds. Should be fixed now, but will take a few days to appear via `sudo rpi-update` (usual caveats apply about that, of course).

davidplowman, 2022-10-27:
This should be available through `sudo rpi-update` now if you feel like trying it. As I said, it's the bleeding edge firmware so don't try it on a system where there's stuff you can't afford to lose!

closed 2022-11-05 by tf4ong