raspberrypi/picamera2 #353: [INFO] stack_processed.py

Bra1nsen, 2022-10-15:
Thanks for providing so many useful examples, like the stack_processed.py! 

```
#!/usr/bin/python3

import cv2
import numpy as np
from picamera2 import Picamera2

exposure_time = 60000  # put your own numbers here
num_frames = 6

# We must tweak the tuning to disable the non-linear gamma transform. Load the
# tuning file for the sensor that you have attached.
tuning = Picamera2.load_tuning_file("imx477.json")
contrast_algo = Picamera2.find_tuning_algo(tuning, "rpi.contrast")
gamma_curve = contrast_algo["gamma_curve"]
contrast_algo["ce_enable"] = 0
contrast_algo["gamma_curve"] = [0, 0, 65535, 65535]

# Create a gamma lookup table to apply at the end.
gamma_x = np.array(gamma_curve[::2], dtype=float) * 255 / 65535
gamma_y = np.array(gamma_curve[1::2], dtype=float) * 255 / 65535
gamma_lut = np.interp(range(num_frames * 255 + 1), gamma_x, gamma_y, right=255).astype(np.uint8)

picam2 = Picamera2(tuning=tuning)
config = picam2.create_still_configuration({"format": "RGB888"}, buffer_count=2)
picam2.configure(config)
images = []
picam2.set_controls({"ExposureTime": exposure_time // num_frames, "AnalogueGain": 1.0})
picam2.start()

for i in range(num_frames):
    images.append(picam2.capture_array())

# Add the images up, apply the gamma transform and we're done.
accumulated = images.pop(0).astype(np.uint16)
for image in images:
    accumulated += image
accumulated = gamma_lut[accumulated]

cv2.imwrite("accumulated.jpg", accumulated)
```

i) How can it be that multiple exposures are merged, if one sets only one exposure time? And if there are different exposure frames, how constant is this program in terms of exposure times? For example, if you set 10 frames, how random/arbitrary are these 10 frames in terms of capture timing?

ii) How far can you max out the frames? What would be a reasonable value and why?

iii) Adding up the frames creates a high dynamic range image (hdr), where does the compression of the dynamic range take place (tone mapping)?

iv) How to shorten the code  so that no gamma transformation is added?

_Best regards
Paul_

davidplowman, 2022-10-17:
Before talking more about this, it's worth noting that this example uses processed images. That means the bottom 4 bits of the 12-bit raw image are basically lost because you get only 8 bits of output. You would have to decide whether this is too much of a problem. The other technique would be to add raw images (that's the stack_raw.py example), but of course you end up with a high dynamic range raw Bayer image, not something you could view so easily.

> i) How can it be that multiple exposures are merged, if one sets only one exposure time? And if there are different exposure frames, how constant is this program in terms of exposure times? For example, if you set 10 frames, how random/arbitrary are these 10 frames in terms of capture timing?

If you capture just one image then you get 8 bits of dynamic range (it's a processed, not a raw, image). If I capture two images of the same exposure and add them I get 9 bits of dynamic range, if I add 4 images I get 10 bits of range, and so on.

> 
> ii) How far can you max out the frames? What would be a reasonable value and why?

What do you mean by "max out"? Do you mean how many frames? I don't think there's a good answer, it probably depends on the circumstances and trying out what works best.

> 
> iii) Adding up the frames creates a high dynamic range image (hdr), where does the compression of the dynamic range take place (tone mapping)?

This technique gives you higher dynamic range and the signal remains linear, but it does not do any tone mapping. Applying tone mapping will make the signal non-linear, and you'd have to decide what kind of tone mapping you want.

> 
> iv) How to shorten the code so that no gamma transformation is added?

You can just delete anything to do with the gamma_lut, probably something like this (untested):
```
#!/usr/bin/python3

import cv2
import numpy as np
from picamera2 import Picamera2

exposure_time = 60000  # put your own numbers here
num_frames = 6

# We must tweak the tuning to disable the non-linear gamma transform. Load the
# tuning file for the sensor that you have attached.
tuning = Picamera2.load_tuning_file("imx477.json")
contrast_algo = Picamera2.find_tuning_algo(tuning, "rpi.contrast")
contrast_algo["ce_enable"] = 0
contrast_algo["gamma_curve"] = [0, 0, 65535, 65535]

picam2 = Picamera2(tuning=tuning)
config = picam2.create_still_configuration({"format": "RGB888"}, buffer_count=2)
picam2.configure(config)
images = []
picam2.set_controls({"ExposureTime": exposure_time // num_frames, "AnalogueGain": 1.0})
picam2.start()

for i in range(num_frames):
    images.append(picam2.capture_array())

# Add the images up, apply the gamma transform and we're done.
accumulated = images.pop(0).astype(np.uint16)
for image in images:
    accumulated += image

cv2.imwrite("accumulated.jpg", accumulated)```

Bra1nsen, 2022-10-18:
Hello David, thanks for your detailed answer.

My first question would be how exactly does the 12bit raw data become an 8bit image.  Is the transformation/compression/downscaling "clean"?

Then, what is the point of adding images with the same exposure time, I don't understand that?

Normally it only makes sense to merge/merge/add images from different exposure times, since the scene brightness is much greater than the camera can capture with one image at all. 

happy greetings Paul

davidplowman, 2022-10-18:
Hi again
> My first question would be how exactly does the 12bit raw data become an 8bit image. Is the transformation/compression/downscaling "clean"?

There's nothing fancy happening here. The values just get bit-shifted down by 4 places, so you are definitely losing some "detail". For "normal" images it's less of a problem because the gamma transform boosts the low values before everything gets reduced to 8 bits, but here we aren't doing a gamma transform. I don't think there are any easy answers - you'd just have to try it and see if it works well enough or not.

> Then, what is the point of adding images with the same exposure time, I don't understand that?

Consider an image taken with a 20ms exposure, and compare it with the result of taking two images, both at 10ms, and then added together (assuming no movement between them). Also assume that the signals are linear (so no gamma has been applied). Let's call this the "HDR image" (for the sake of argument).

Pixels that are _unsaturated_ in the 20ms image should have the exact same values in the HDR image (up to noise). In each 10ms image they will have half the level of the 20ms image, but adding two of them brings the level back up again. So we have the exact same image everywhere where the 20ms image is not saturated. Now let us consider saturated pixels.

Suppose that a pixel in the 20ms image "should" have had the value 300, but has saturated and we get 255. But in the 10ms images, that pixel will have the value 300/2 = 150. We add them, and the HDR image will have the correct value of 300. So we've extended the dynamic range by an extra bit, and only when the values reach 511 are the pixels saturated now.

Does that make sense?

> 
> Normally it only makes sense to merge/merge/add images from different exposure times, since the scene brightness is much greater than the camera can capture with one image at all.

That's true, mostly. Often it makes sense to capture a short exposure and a long exposure, you can scale up the short exposure and then add/combine the images, and this is a very common technique. But as I explained above, you can also capture many short exposure images and add them all together - sometimes this technique is more convenient.

> 
> happy greetings Paul

Bra1nsen, 2022-10-19:
Ok I just looked at the binary to decimal conversion again and I understand it now. It really hurts me to lose those 4 bits of precision. At least for now.

Ok yes very interesting! The conflict of goals then consists in the fact that with short exposure times often completely unsaturated pixels can be added, nothing would improve...

with added gamma:
[image]
no gamma added:
[image]

Bra1nsen, 2022-10-19:
different question.

```
	with picamera2.Picamera2(tuning=tuning) as camera: 
		config = camera.create_still_configuration(main={"size": (1014, 760),"format": "RGB888"}, raw={"format": "SRGGB12", "size" : (2032, 1520)})
		camera.configure(config)
		
		for exp in exposures:
			camera.set_controls({"ExposureTime": exp, "AnalogueGain": 1, "ColourGains": (1.0,1.0)})
			camera.start()
			metadata = camera.capture_file(f"{exp}.tga")
			lux = metadata["Lux"]
			sens_temp = metadata["SensorTemperature"]
			camera.stop()
			print("Exposure:", exp, "lux:", lux, " sensor temp:", sens_temp)
```

Why isnt the lux value increasing further with longer exposure time?

Exposure: 50 lux: 400.0  sensor temp: 24.0
Exposure: 100 lux: 18250.900390625  sensor temp: 24.0
Exposure: 150 lux: 17686.6484375  sensor temp: 24.0
Exposure: 250 lux: 17249.421875  sensor temp: 24.0
Exposure: 400 lux: 17114.546875  sensor temp: 25.0
Exposure: 650 lux: 17190.9453125  sensor temp: 25.0
Exposure: 1050 lux: 17171.302734375  sensor temp: 25.0
Exposure: 1700 lux: 17134.34765625  sensor temp: 25.0
Exposure: 2750 lux: 15190.162109375  sensor temp: 25.0

closed 2022-10-19 by Bra1nsen