raspberrypi/picamera2 #354: [BUG] OSError: ioctl(VIDIOC_STREAMON) failure : 22, Invalid argument

dhaval-khatri1996, 2022-10-16:
openCv python version 4.5.5.
Camera module waveshare OV9281-110
I am working with raspberry-pi model 4-B with latest bullseye OS.

I am able to capture image using **libcamera-still -o test.png** command but when I try to read an image with opencv it returns None.
The output for cam.isOpened() is true but cam.grab() and cam.read() return false and (false,None) respectively.

Code: 
```
import pygame
import pygame.camera
from pygame.locals import *

pygame.init()
pygame.camera.init()
camlist = pygame.camera.list_cameras()
if camlist:
    cam = pygame.camera.Camera(camlist[0])
    cam.start()
    image = cam.get_image()
    print(image)

```

Output:
```
OSError: ioctl(VIDIOC_STREAMON) failure : 22, Invalid argument
```
camera info
```

~ $ v4l2-ctl --device /dev/video0 --all

Driver Info:
	Driver name      : unicam
	Card type        : unicam
	Bus info         : platform:fe801000.csi
	Driver version   : 5.15.56
	Capabilities     : 0xa5a00001
		Video Capture
		Metadata Capture
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x25200001
		Video Capture
		Read/Write
		Streaming
		Extended Pix Format
Media Driver Info:
	Driver name      : unicam
	Model            : unicam
	Serial           : 
	Bus info         : platform:fe801000.csi
	Media version    : 5.15.56
	Hardware revision: 0x00000000 (0)
	Driver version   : 5.15.56
Interface Info:
	ID               : 0x03000005
	Type             : V4L Video
Entity Info:
	ID               : 0x00000003 (3)
	Name             : unicam-image
	Function         : V4L2 I/O
	Flags         : default
	Pad 0x01000004   : 0: Sink
	  Link 0x02000007: from remote pad 0x1000002 of entity 'ov9281 10-0060': Data, Enabled, Immutable
Priority: 2
Video input : 0 (unicam-image: ok)
Format Video Capture:
	Width/Height      : 640/480
	Pixel Format      : 'BGR3' (24-bit BGR 8-8-8)
	Field             : None
	Bytes per Line    : 1920
	Size Image        : 921600
	Colorspace        : sRGB
	Transfer Function : sRGB
	YCbCr/HSV Encoding: ITU-R 601
	Quantization      : Full Range
	Flags  
```

chrisruk, 2022-10-17:
Hi,

I'm afraid it looks like you're not using the picamera2 library, but a camera library
that's part of pygame.

I'm not sure what the issue is unfortunately.

davidplowman, 2022-10-17:
Hi, yes it looks like you're using pygame's camera library, not Picamera2. I expect pygame only supports "simple" cameras, not embedded cameras with separate image signal processors, so my guess is that it can't work like this. Sorry not to be more help.

dhaval-khatri1996, 2022-10-17:
I am really sorry.

closed 2022-10-17 by dhaval-khatri1996