raspberrypi/picamera2 #355: [OTHER]

expertgrammer, 2022-10-17:
im trying to use raspberry pi camera 1.3 on zero but getting this issue [image] in config.txt i have dtoverlay=ov5647 
pls help me

davidplowman, 2022-10-17:
Hi, thanks for the question. Could you please report:

* Exactly what kind of Raspberry Pi you are using (a Raspberry Pi Zero W?)
* Exactly what OS image you have installed. The latest Bullseye? Raspberry Pi Full OS or Lite?
* It looks like you're using X Windows, but I can't quite tell if it's the usual installation. Can you say what it is?
* Have you made any changes to your /boot/config.txt and if so, what?
* Are you able to try downloading a fresh copy of the latest Bullseye image? Does this work? The only thing you should need to do is let it apply its updates and then use raspi-config to re-enable Glamor.

Thanks.

expertgrammer, 2022-10-17:
yea im using raspberry pi zero w, i updated everything, i think im using bulseye , yea i changed /boot/config.txt by changing dtoverlay=ov5647 .
i forgot what command it was but it returning suported=1 , detected=0 

also first time u plugged cameras cable upside down then it got kinda hot i think that got burned then i changed that cable to another, when im turning on raspberry pi camera red light working so it means camera is working

expertgrammer, 2022-10-17:
im new for raspberry pi and linux so pls help me in details if you can thx ❤️

expertgrammer, 2022-10-17:
im using now this cable

[image]

i think this got burned

[image]

davidplowman, 2022-10-17:
Hi, I'm still not really clear exactly what image you are running and what has happened. Are you able to reinstall the latest 32-bit Raspberry Pi OS Lite image using the Raspberry Pi imager tool? Once it's booted, perform any necessary updates (`sudo apt update` and then `sudo apt upgrade`) but don't touch or change anything else. Reboot again and then run `libcamera-hello`. Does this work?

In the instructions above I've avoided installing an image with X Windows. I'd be very wary about using X Windows on a Pi Zero. It runs slowly at the best of times, and trying to get camera preview images through the X display stack is really very expensive, and I would try hard to avoid it.

expertgrammer, 2022-10-17:
sorry i dont know what is x windows i set up this raspberry pi with image installer i think i installed full not lite, can you tell me the commands that can answer your questions , if i still cant answer your questions just tell me few ways to fix this problems ill try all of them

expertgrammer, 2022-10-17:
now ill try to explain more details 
when in /boot/config.txt  dtoverlay=vc4-kms-v3d im getting this [image]
and if im trying to enable legacy camera in raspi-config this is returning this [image]

...

otherwise if im changing dtoverlay=ov5647 in /boot/config.txt
im getying those errors i showed you before [image]
and also vcgencmd get_camera returning

suported=1, detected=0 , libcamera interfaces=1

davidplowman, 2022-10-17:
I would start by installing the latest 32-bit Raspberry Pi OS Lite. This will give you an image without X Windows, but on a Pi Zero X Windows is going to be very slow and the camera preview will not look very good. Once the image has booted up for the first time, run
```
sudo apt update
sudo apt upgrade
```
and reboot again. Touch nothing else. Log in and run `libcamera-hello`. This should just work.

expertgrammer, 2022-10-17:
i installed raspberry pi os like 7-9 months ago but i updated everything so that should be the reason of this issue ?

tvoverbeek, 2022-10-17:
@expertgrammer As @davidplowman wrote, running the full desktop OS on a Pi Zero is not a good idea (unless it its a Pi Zero 2).
picamera2 and libcamera apps will not work if you have the legacy camera stack enabled, as you obviously have.
Easiest way to get rid of it is to do a clean install of Raspberry Pi Os Lite followed by `sudo apt update`, `sudo apt -y upgrade` as @davidplowman suggested

davidplowman, 2022-10-24:
Hi, is there anything else for us to investigate here? If not I'll consider closing it in a few days. Thanks!

davidplowman, 2022-10-31:
I'll close this one for now, but please open a new issue if anything else comes up. Thank.

closed 2022-10-31 by davidplowman